lvgl/lvgl #5201: add a element `id` into lv_obj_t

denisdemaisbr, 2024-01-05:
### Introduce the problem

this a simple attribute that made easyway to identify object and avoid userdata.

lv_obj_t * btn = lv_btn_create(lv_scr_act());
btn->id = 100;

### Proposal

add a element id into lv_obj_t struct

XuNeo, 2024-01-05:
It's already added. See

https://github.com/lvgl/lvgl/blob/4f69d10b22ed3bbbe10e32bb5a60a5df9ac759ef/src/core/lv_obj.h#L229-L231

denisdemaisbr, 2024-01-05:
thanks!

I'm working project lv_port_linux_frame_buffer

 $ git branch
* (HEAD detached at 49c59f461)

typedef struct _lv_obj_t {
    const lv_obj_class_t * class_p;
    struct _lv_obj_t * parent;
    _lv_obj_spec_attr_t * spec_attr;
    _lv_obj_style_t * styles;
#if LV_USE_USER_DATA
    void * user_data;
#endif
    lv_area_t coords;
    lv_obj_flag_t flags;
    lv_state_t state;
    uint16_t layout_inv : 1;
    uint16_t scr_layout_inv : 1;
    uint16_t skip_trans : 1;
    uint16_t style_cnt  : 6;
    uint16_t h_layout   : 1;
    uint16_t w_layout   : 1;
} lv_obj_t;

closed 2024-01-05 by denisdemaisbr