lvgl/lvgl #5202: Possible bug in lv_calendar, prevents year entry of 2024

wreyford, 2024-01-05:
### LVGL version

8.3

### What happened?

I am unable to set a date from 2024 in the calendar.
It will not go beyond 2023

If you set the date to 2024 01 05, and the display date to 2024 01 it will set it to 1901 01
When scrolling up in the dropdown, it will end at 2023.
You cannot set a date for 2024
This is happening in my production code, and can easily be replicated in the online simulator.

[image]

[image]

[image]

### How to reproduce?

To replicate the problem easily, go to the 8.3 docs, then -> lv_calendar ->simulator.
In the simulator you can paste this code
```
##### startup script #####

#!/opt/bin/lv_micropython -i

import lvgl as lv

import display_driver

##### main script #####

def event_handler(evt):
    code = evt.get_code()

    if code == lv.EVENT.VALUE_CHANGED:
        source = evt.get_current_target()
        date = lv.calendar_date_t()
        if source.get_pressed_date(date) == lv.RES.OK:
            calendar.set_today_date(date.year, date.month, date.day)
            print("Clicked date: %02d.%02d.%02d"%(date.day, date.month, date.year))

calendar = lv.calendar(lv.scr_act())
calendar.set_size(200, 200)
calendar.align(lv.ALIGN.CENTER, 0, 20)
calendar.add_event_cb(event_handler, lv.EVENT.ALL, None)

calendar.set_today_date(2024, 1, 5)
calendar.set_showed_date(2024, 1)

# Highlight a few days
highlighted_days=[
    lv.calendar_date_t({'year':2024, 'month':1, 'day':1}),
    lv.calendar_date_t({'year':2023, 'month':12, 'day':25}),
    lv.calendar_date_t({'year':2024, 'month':1, 'day':22})
]

calendar.set_highlighted_dates(highlighted_days, len(highlighted_days))

lv.calendar_header_dropdown(calendar)
```

PGNetHun, 2024-01-06:
Hello!

Yes, it's indeed a bug:
https://github.com/lvgl/lvgl/blob/release/v8.3/src/extra/widgets/calendar/lv_calendar_header_dropdown.c#L43

There are more problems with the implementation, so it needs further change.
https://github.com/lvgl/lvgl/blob/release/v8.3/src/extra/widgets/calendar/lv_calendar_header_dropdown.c#L123

It needs to be fixed in v9, too:
https://github.com/lvgl/lvgl/blob/master/src/widgets/calendar/lv_calendar_header_dropdown.c#L46

wreyford, 2024-01-06:
OK, this is a bug, and will result in **maintenance problems** to keep updating the years in static const char * year_list = { "2023\...}
See code in calnedar-> lv_calendar_header_dropdown. c (Version 8.3)
Lines 42 to 51.

There is also a hardcoded 2023 dates lower down - lines 123 and135 that will also result in maintenance problems.

```
/**********************
 *  STATIC VARIABLES
 **********************/
const lv_obj_class_t lv_calendar_header_dropdown_class = {
    .base_class = &lv_obj_class,
    .width_def = LV_PCT(100),
    .height_def = LV_SIZE_CONTENT,
    .constructor_cb = my_constructor
};

static const char * month_list = "01\n02\n03\n04\n05\n06\n07\n08\n09\n10\n11\n12";
static const char * year_list = {
    "2023\n2022\n2021\n"
    "2020\n2019\n2018\n2017\n2016\n2015\n2014\n2013\n2012\n2011\n2010\n2009\n2008\n2007\n2006\n2005\n2004\n2003\n2002\n2001\n"
    "2000\n1999\n1998\n1997\n1996\n1995\n1994\n1993\n1992\n1991\n1990\n1989\n1988\n1987\n1986\n1985\n1984\n1983\n1982\n1981\n"
    "1980\n1979\n1978\n1977\n1976\n1975\n1974\n1973\n1972\n1971\n1970\n1969\n1968\n1967\n1966\n1965\n1964\n1963\n1962\n1961\n"
    "1960\n1959\n1958\n1957\n1956\n1955\n1954\n1953\n1952\n1951\n1950\n1949\n1948\n1947\n1946\n1945\n1944\n1943\n1942\n1941\n"
    "1940\n1939\n1938\n1937\n1936\n1935\n1934\n1933\n1932\n1931\n1930\n1929\n1928\n1927\n1926\n1925\n1924\n1923\n1922\n1921\n"
    "1920\n1919\n1918\n1917\n1916\n1915\n1914\n1913\n1912\n1911\n1910\n1909\n1908\n1907\n1906\n1905\n1904\n1903\n1902\n1901"
};

/**********************
 *      MACROS
 **********************/
```

and 

```
static void year_event_cb(lv_event_t * e)
{
    lv_obj_t * dropdown = lv_event_get_target(e);
    lv_obj_t * calendar = lv_event_get_user_data(e);

    uint16_t sel = lv_dropdown_get_selected(dropdown);

    const lv_calendar_date_t * d;
    d = lv_calendar_get_showed_date(calendar);
    lv_calendar_date_t newd = *d;
    newd.year = 2023 - sel;

    lv_calendar_set_showed_date(calendar, newd.year, newd.month);
}

static void value_changed_event_cb(lv_event_t * e)
{
    lv_obj_t * header = lv_event_get_target(e);
    lv_obj_t * calendar = lv_obj_get_parent(header);
    const lv_calendar_date_t * cur_date = lv_calendar_get_showed_date(calendar);

    lv_obj_t * year_dd = lv_obj_get_child(header, 0);
    lv_dropdown_set_selected(year_dd, 2023 - cur_date->year);

    lv_obj_t * month_dd = lv_obj_get_child(header, 1);
    lv_dropdown_set_selected(month_dd, cur_date->month - 1);
}
```

Please can you fix this, as all calendars must be broken now in 2024, possibly without hardcoding??
Thx

C47D, 2024-01-06:
@kisvegabor Adding the possibility for the user to set a custom `year_list` would solve this?

PGNetHun, 2024-01-07:
I thought on adding a new function: 
`lv_calendar_set_year_range(lv_obj_t * obj, uint32_t start, uint32_t end)`

When the calendar object initializes, then a default range will be set (-100 to +10 years).

kisvegabor, 2024-01-07:
My idea was to make the default year list in the range of 2100..1900 and a `set_start_year` could just point to a different element of the array as a start address. 
```c
&year_list[5 * 10]; //Means 2100 - 10
```

However adding a `set_year_list` is probably the simplest and most flexible solution, so if someone is interested in sending a PR, it would be highly appreciated.

wreyford, 2024-01-07:
If I may add, we are working on memory constrained devices, so a static list from 2100..1900 is ok, but unnecessary memory usage. Adding the option to set the date range allows the end user all the flexibility they could ask for. Thx for a wonderful product.

kisvegabor, 2024-01-07:
> Adding the option to set the date range allows the end user all the flexibility they could ask for. Thx for a wonderful product.

If we had a `set_year_list` function, the default year list could be shorter too as it will act only as an example. E.g. 2025..1990.

wreyford, 2024-01-07:
That in my humble opinion is an excellent workable solution. Good luck with v9. 0 too. You must be very busy. Sent from my phone-------- Original message --------From: Gabor Kiss-Vamosi ***@***.***>Date: Sun, 07 Jan 2024, 12:27To: lvgl/lvgl ***@***.***>Cc: wreyford ***@***.***>, Author ***@***.***>Subject: Re: [lvgl/lvgl] Possible bug in lv_calendar, prevents year entry of 2024 (Issue #5202)

Adding the option to set the date range allows the end user all the flexibility they could ask for. Thx for a wonderful product.

If we had a set_year_list function, the default year list could be shorter too as it will act only as an example. E.g. 2025..1990.

C47D, 2024-01-08:
I'm working on a draft @kisvegabor, I guess this needs to be under the new development process?

lhdjply, 2024-01-08:
I think we can handle it this way
```c
static char * year_list;
void lv_calendar_set_year_range(lv_obj_t * obj, uint32_t start, uint32_t end)
{
    char *year_list_temp;
    uint32_t year_diff;
    year_diff = end - start;
    if(end < start) {
        LV_LOG_ERROR("The end must be larger than the start");
    }
    if(year_list != NULL) {
        lv_free(year_list);
    }
    year_list = lv_malloc((year_diff + 1) * 5);
    year_list_temp=year_list;
    for(uint32_t i = 0; i < year_diff; i++) {
        lv_snprintf(year_list_temp, 5, "%04d\n", start + year_diff);
        year_list_temp = year_list_temp + 4;
    }
    lv_snprintf(year_list_temp, 4, "%04d", start + year_diff);
    year_list_temp[(year_diff + 1) * 5 - 1] = '\0';
}
```

kisvegabor, 2024-01-08:
I think letting the user to explicitly set a year list is better as this the list can be stored in flash too.

@C47D We are in big hurry now to release v9 next Monday, so we can just work with a PR as usual. We will start the shaping/pitching after the release. Thank you for asking anyway, highly appreciated :hugs:

lhdjply, 2024-01-08:
> I think letting the user to explicitly set a year list is better as this the list can be stored in flash too.

I feel that setting an explicit year is too cumbersome. If the year is set too large, manually input a particularly long string. Moreover, it is not very flexible and cannot be dynamically modified. What do you think?

kisvegabor, 2024-01-08:
Thank you, let's continue in #5227

closed 2024-01-08 by kisvegabor

wreyford, 2024-03-06:
Hi, its been a while. 
I use the LVGL 8.3.11 component from the ESP-IDF Component register for building my apps.
I thus cannot edit the fixes into this build.
I have tried to Convert from ESP-IDF V5.2 LVLG8.3.11 to LVGL9.0 but that too is a massive task and my progress has been very slow with multiple hurdles.
So how do I get the calendar to work, using the Component Register [https://components.espressif.com/components/lvgl/lvgl/versions/8.3.11?language=en](url)

wreyford, 2024-03-11:
Thank you. Much appreciated.

PR #5227 (2024-01-11): feature(calendar): Add interface to set a custom year list
Help us review this PR! Anyone can [approve it or request changes](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/reviewing-changes-in-pull-requests/approving-a-pull-request-with-required-reviews).

### Description of the feature or fix

Adds an interface to set a custom year list to the calendar widget, fixes #5202 

### Checkpoints
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tes [... cut]