lvgl/lvgl #5203: v9 github codespace fails to build

ttump, 2024-01-05:
### LVGL version

v9 current github

### What happened?

Followed to 3 clicks vscode github codespace procedure, build fails with multiple errors of the same type

/usr/local/emsdk/upstream/emscripten/cache/sysroot/include/bits/alltypes.h:182:15: error: Expected ,, instead got: int16_t
typedef short int16_t;
              ^
make[2]: *** [lvgl/CMakeFiles/lvgl.dir/build.make:1155: lvgl/CMakeFiles/lvgl.dir/src/draw/sw/blend/neon/lv_blend_neon.S.o] Error 1

C47D, 2024-01-06:
@kisvegabor Sounds like the same issue we had at eclipse, is it?

kisvegabor, 2024-01-07:
It should be fixed by #5209.

To test it just remove the `#include <stdint.h>` at the beginning of `lv_conf.h`

closed 2024-01-07 by FASTSHIFT

ttump, 2024-01-07:
That works fine, thank you. Good luck with the release of v9

kisvegabor, 2024-01-08:
Thank you :blush:

PR #5209 (2024-01-07): fix(codespace): remce the extra include from lv_conf.h
Help us review this PR! Anyone can [approve it or request changes](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/reviewing-changes-in-pull-requests/approving-a-pull-request-with-required-reviews).

### Description of the feature or fix

fixes #5203

### Checkpoints
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicab [... cut]