lvgl/lvgl #5205: Update C-array images in demos/examples

XuNeo, 2024-01-05:
### Introduce the problem

Current C-array image resources don't have `data_size` filed set, thus caused issues mentioned in below PR, where it's used to invalidate cache. It could also be used to make draw buffer stride alignment etc.

https://github.com/lvgl/lvgl/pull/5200#discussion_r1443001120

### Proposal

Ideally all PNGs should be put into a directly, simply re-run image converter script can update them all.

kisvegabor, 2024-01-07:
We can move the PNGs to a single folder but the result images shall be stored in their own demo folder. Would it be possible to make the converter script recursively traverse the demos and examples folders and convert all images starting with `lv_demo_image_*` or `lv_example_image_*`? It also not clear how to select the output color format. Maybe it can be indicated in the name as well. E.g.
```c
lv_demo_image_..._argb8888.png; //convert to argb8888
lv_demo_image_..._rgb.png; //convert to LV_COLOR_FORMAT_NATIVE
lv_demo_image_..._argb.png; //convert to LV_COLOR_FORMAT_NATIVE_ALPHA
```

I need to mention that it will be a larger work to unify the names of all images but I can help with that.

XuNeo, 2024-01-25:
Not stale.

 Needed by https://github.com/lvgl/lvgl/pull/5471

kisvegabor, 2024-02-12:
Not stale.

closed 2024-03-14 by lvgl-bot

nicusorcitu, 2024-03-19:
Can we reopen this issue? It has to be fixed.

XuNeo, 2024-03-19:
It should be already added in https://github.com/lvgl/lvgl/pull/5608

nicusorcitu, 2024-03-19:
OK, I see. This ticket is only about data_size field.
What am I looking for is the ticket where it mention that the images needs to be updated with the stride alignment set to 64 to satisfy GPU needs.

E.G. all the images will need to be padded until stride is aligned.

nicusorcitu, 2024-03-19:
As they are right now upstream, all the assets (C array files) can not be processed by the GPU.

nicusorcitu, 2024-03-20:
Recomposing the image is inefficient.
You allocate another buffer where you copy the source image line by line and add the stride padding. It is resource consuming (CPU and memory).

My approach is that all image files (c array files) to be already aligned to a stride (we can chose 64 bytes to comply with any GPU) so that those images can be processed right away.

The clients are aware of the stride alignment requirements and they should pick images to satisfy this need.
@cristian-stoica @anaGrad

XuNeo, 2024-03-20:
> https://github.com/lvgl/lvgl/blob/master/scripts/LVGLImage.py

Yes, this is the script to generate images meet GPU, you need to specify the align parameter in command line.

> My suggestion is that all upstream demos/examples to use this kind of assets
I haven't done this because for software renderer, alignment is not needed. For performance evaluation, we internally create real scene for them using correct aligned images.

nicusorcitu, 2024-03-20:
We have as well internal images to test the GPU. But we want to have the possibility to run the benchmark out of the box. It should be doable.
We will create soon the PR with the new C array files updated.

nicusorcitu, 2024-03-21:
https://github.com/lvgl/lvgl/pull/5925