lvgl/lvgl #5206: newest version break! lv_global.disp_refresh->buf_act is nullptr!!!!

xinsuinizhuan, 2024-01-06:
### LVGL version

v9.0.0

### What happened?

when i run the lv_demo_widgets ,break! 2022.01.06
[image]

### How to reproduce?

newest, run the lv_demo_widgets 
[image]

kisvegabor, 2024-01-06:
Hi,

Have you updated LVGL to the latest version in the project?

Please confirm if you are using this project: https://github.com/lvgl/lv_port_pc_visual_studio If so @MouriNaruto could you take a look at this issue?

xinsuinizhuan, 2024-01-06:
> Hi,
> 
> Have you updated LVGL to the latest version in the project?
> 
> Please confirm if you are using this project: https://github.com/lvgl/lv_port_pc_visual_studio If so @MouriNaruto could you take a look at this issue?

the  latest version, 2024.01.04 version is ok， but the version of 2024.01.05 start break.

PGNetHun, 2024-01-06:
Maybe this is the problem:
1 line is missing from the new code:
https://github.com/lvgl/lvgl/commit/304a3c35736178ce19e36f8cbd0dcd197a232a58#diff-d5e5e7b4b434f724deeead3491cc6d37ecc20b0db8f61600d528bcdd73353ebfL397

[image]

**BUT:**
XuNeo has alread a PR to fix this:
https://github.com/lvgl/lvgl/pull/5204/files#diff-d5e5e7b4b434f724deeead3491cc6d37ecc20b0db8f61600d528bcdd73353ebfR396

kisvegabor, 2024-01-06:
And do you use the our visual studio project as it is?

MouriNaruto, 2024-01-06:
[image]

I have found the same issue when I using the latest LVGL implementation.

Kenji Mouri

MouriNaruto, 2024-01-06:
It looks https://github.com/lvgl/lvgl/commit/304a3c35736178ce19e36f8cbd0dcd197a232a58 break the Windows platform implementation.

[image]

I think LVGL should check the pointer of disp_refr->buf_act.

Windows platform won't set the buffer at the beginning because the Window size needs to be determined after the sizing fixing. (For example, in simulator mode, we need to ensure the content size is the same as the user set for that with considering the zoom. Also, I don't want to create the buffer we actually unused.)

Kenji Mouri

MouriNaruto, 2024-01-06:
[image]

I tried to do a simple fix and looks OK. I think maybe LVGL guys have a better way to do that.

Kenji Mouri

kisvegabor, 2024-01-07:
Thank you for looking into it. Please review #5210.

PR #5210 (2024-01-07): fix(refr): update checking for a valid draw buffer
Help us review this PR! Anyone can [approve it or request changes](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/reviewing-changes-in-pull-requests/approving-a-pull-request-with-required-reviews).

### Description of the feature or fix

Fixes #5206

### Checkpoints
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
 [... cut]
closed 2024-01-07 by FASTSHIFT