lvgl/lvgl #5211: Touch screen input events missed

stwirth, 2024-01-07:
### LVGL version

v8.3

### What happened?

Sometimes very short taps on the touch screen are not recognized by lvgl.
I'm using evdev as input device and I can see the events coming in when watching `evtest` output. However, the lv_obj does not receive events.
When the touch is longer (like if you would press a physical button), everything works as expected.
This happens only 50% of the time on very quick tapping.

Here's a video of three taps, left shows debugging output from my program in lvgl event callbacks, right side shows output of `evtest`. You can see that the first two taps are being recognized by lvgl but the last one isn't (no output on left side).

https://github.com/lvgl/lvgl/assets/1481786/da605b47-512d-43c3-8bf9-10c6f13279cd

stwirth, 2024-01-08:
It works for me. Here's a GIF version:
[image]

kisvegabor, 2024-01-08:
Strange, now the video works for me too. 

To be honest it doesn't tell me much :slightly_frowning_face: 
Can you print what happens in [evdev_read](https://github.com/lvgl/lv_drivers/blob/master/indev/evdev.c#L173)?

stwirth, 2024-02-09:
I finally got back to this.

The issue seems to happen when both the touch and the release events are in the event queue and evdev_read handles both in the same while loop.

I added some logging to investigate this:
```diff
@@ -107,9 +107,11 @@ bool evdev_set_file(char* dev_name)
  */
 void evdev_read(lv_indev_drv_t * drv, lv_indev_data_t * data)
 {
+    LV_LOG_WARN("evdev_read");
     struct input_event in;
 
     while(read(evdev_fd, &in, sizeof(struct input_event)) > 0) {
+        LV_LOG_WARN("evdev_read: type=%d, code=%d, value=%d", in.type, in.code, in.value);
         if(in.type == EV_REL) {
             if(in.code == REL_X)
                                #if EVDEV_SWAP_AXES
```

Here is a working example: left is the output of evtest, showing all events, right is the output of a test app with the added logging in evdev_read:

[image]
You can see that the execution enters evdev_read and processes the touch (type=1, code=330, value=1).
Then, the next time evdev_read is called, the release event (type=1, code=330, value=0) is processed.
All my registered callbacks are being called (pressed, pressing, released, clicked).

Here is an example of lvgl events not being emitted:
[image]
The output of evtest on the left side looks the same as above but no lvgl event is created. evdev_read processes both the touch and the release event in one while loop. Seems some return or break is missing.

stwirth, 2024-02-09:
As far as I see this problem also exists on `master`:
https://github.com/lvgl/lvgl/blob/cfd14cdacd682977ce6ceb681a1bc357c6723f1b/src/drivers/evdev/lv_evdev.c#L117-L133
When both a press and a release event can be read from the file descriptor, then the press event is not registered (forwarded) by `_evdev_read`.

stwirth, 2024-02-09:
@kisvegabor re-open?

kisvegabor, 2024-02-11:
If there are multiple events to read this `while` loop really overwrites the previously read events.

The most simple solution would be to refactor `_evdev_read` like this:
- Try to read an event. If there is no event, return
- If an event was read, process it and set `data->continue_reading = true;` to make LVGL check for events again
- This way there is no `while` loop, but only one event is read and processed at a time.

Could you try it out?

stwirth, 2024-02-12:
That sounds good, I'll try it.

stwirth, 2024-02-12:
I just tried removing the while loop, processing each individual event with `continue_reading = true` and this does not work. I get really strange dragging behavior.

If I understand https://www.kernel.org/doc/Documentation/input/event-codes.txt correctly, then the application should not process each and every incoming event directly but instead wait for EV_SYN events.
So for example this sequence:
```
Event: time 1707752573.099950, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 1707752573.099950, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 128
Event: time 1707752573.099950, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 237
Event: time 1707752573.099950, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1707752573.099950, -------------- SYN_REPORT ------------
Event: time 1707752573.110449, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1707752573.110449, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1707752573.110449, -------------- SYN_REPORT ------------
```
should be interpreted as "press on position (128, 237)", followed by a "release".
With `evdev_read` processing each individual event as you suggested, it would be interpreted as "press on wherever the last position was", followed by "move X to 128", followed by "move Y to 237", followed by "release".

stwirth, 2024-02-12:
I changed the (original v8.3) code to only forward the position and touch state changes when a SYN_REPORT event comes in:
```patch
@@ -109,7 +109,12 @@ void evdev_read(lv_indev_drv_t * drv, lv_indev_data_t * data)
 {
     struct input_event in;
 
+    bool do_report = false;
     while(read(evdev_fd, &in, sizeof(struct input_event)) > 0) {
+        if(in.type == EV_SYN && in.code == SYN_REPORT) {
+            do_report = true;
+            break;
+        }
         if(in.type == EV_REL) {
             if(in.code == REL_X)
                                #if EVDEV_SWAP_AXES
@@ -214,7 +219,7 @@ void evdev_read(lv_indev_drv_t * drv, lv_indev_data_t * data)
         data->state = evdev_button;
         return;
     }
-    if(drv->type != LV_INDEV_TYPE_POINTER)
+    if(drv->type != LV_INDEV_TYPE_POINTER || !do_report)
         return ;
     /*Store the collected data*/
```

With that even the very short tap events get processed correctly (which was the original issue). But now dragging lags behind a lot...

stwirth, 2024-02-13:
I though that my last change would mean that less events are processed but it is actually the other way around. With the original code, between two calls of `evdev_read`, input events are queued up, the while loop flushes the queue and only one event needs to be processed by lvgl.
Forcing the processing of the event through breaking the while loop leaves some events in the queue unprocessed and unfortunately the speed of my system is not fast enough to keep the queue small. During a drag event, more events come in than the event processing can handle and the queue grows, resulting in a lag.

stwirth, 2024-02-14:
In the current implementation, the processing rate of events is somewhat independent of the rate at which events are coming in. The while loop guarantees that all evens from the queue are flushed, resulting in a more responsive UI. However, a button press event is pretty important and should not be overwritten by a following button release event.
I'll experiment a bit more with forcing the processing of certain events.

stwirth, 2024-02-14:
The problem of missing touch events is fixed with this change:
```diff
@@ -109,7 +109,12 @@ void evdev_read(lv_indev_drv_t * drv, lv_indev_data_t * data)
 {
     struct input_event in;
 
+    bool got_button_event = false;
     while(read(evdev_fd, &in, sizeof(struct input_event)) > 0) {
+        // if there was a button event, we force processing on the next syn event
+        // so that it does not get overwritten by the next button event
+        if(in.type == EV_SYN && in.code == SYN_REPORT && got_button_event)
+            break;
         if(in.type == EV_REL) {
             if(in.code == REL_X)
                                #if EVDEV_SWAP_AXES
@@ -160,6 +165,7 @@ void evdev_read(lv_indev_drv_t * drv, lv_indev_data_t * data)
                     evdev_button = LV_INDEV_STATE_REL;
                 else if(in.value == 1)
                     evdev_button = LV_INDEV_STATE_PR;
+                got_button_event = true;
             } else if(drv->type == LV_INDEV_TYPE_KEYPAD) {
 #if USE_XKB
                 data->key = xkb_process_key(in.code, in.value != 0);
```
The reason to wait for the next SYN_REPORT is that new touches come in in this sequence on my system:
```
Event: time 1707942154.030935, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 1
Event: time 1707942154.030935, type 3 (EV_ABS), code 57 (ABS_MT_TRACKING_ID), value 0
Event: time 1707942154.030935, type 3 (EV_ABS), code 53 (ABS_MT_POSITION_X), value 232
Event: time 1707942154.030935, type 3 (EV_ABS), code 54 (ABS_MT_POSITION_Y), value 259
Event: time 1707942154.030935, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1707942154.030935, -------------- SYN_REPORT ------------
Event: time 1707942154.041030, type 1 (EV_KEY), code 330 (BTN_TOUCH), value 0
Event: time 1707942154.041030, ++++++++++++++ SYN_MT_REPORT ++++++++++++
Event: time 1707942154.041030, -------------- SYN_REPORT ------------
```
So if you schedule the processing right after the BTN_TOUCH event you would initiate the touch wherever the last buffered position was. Forcing the processing after the SYN_REPORT ensures* that both X and Y positions are updated before the touch is passed on to LVGL.

*) it's actually not _ensured_, it could still happen that the queue is empty after [BTN_TOUCH, ABS_MT_POSITION_X] which would then result in LVGL processing the touch with the previously buffered Y value. To fix this the queue would need to be flushed in packets and only full packets should be passed on to LVGL. That in turn would enforce the need for the existence of SYN_REPORT events.

@kisvegabor should I make a PR for the patch above?

kisvegabor, 2024-02-15:
Thank you for the investigation! I haven't tested it yet, but please send a PR and I will try it out on my end too.

closed 2024-03-28 by kisvegabor