lvgl/lvgl #5212: lv_chart: No event  in crowded mode

euphi, 2024-01-07:
### Introduce the problem

If a line chart is drawn in crowded mode (at least as many points as pixels), there is no event triggered.

If not in crowded mode, LV_EVENT_DRAW_PART_BEGIN and LV_EVENT_DRAW_PART_END are triggered before and after drawing the line.

(Version: 8.3.x)

### Proposal

Events shall be consistent.

kisvegabor, 2024-01-08:
It's intentional as sending event in crowded mode too could mean a bottle neck.

euphi, 2024-01-08:
It's a bit arbitrary, but ok. 

Personally I don't need it  - I just stumbled upon this while having a look if it is possible to have a callback to provide series data (instead of external array).

However I saw this is not foreseen and understand that it is more performant to (re-)generate or update the external array whenever it is necessary.

closed 2024-01-08 by euphi