lvgl/lvgl #5214: lv_image_set_src: failed to get image info: E:/lvgl/git/lvgl-master/examples/libs/bmp/example_32bit.bmp lv_image.c:175

xinsuinizhuan, 2024-01-08:
### LVGL version

v9.0.0

### What happened?

the lv_example_bmp_1.c example,
/**
 * Open a BMP file from a file
 */
void lv_example_bmp_1(void)
{
	lv_obj_t* img = lv_image_create(lv_screen_active());
	/* Assuming a File system is attached to letter 'A'
	 * E.g. set LV_USE_FS_STDIO 'A' in lv_conf.h */
	lv_image_set_src(img, "E:/lvgl/git/lvgl-master/examples/libs/bmp/example_32bit.bmp");
	lv_obj_center(img);

}

[Warn]  (7889.578, +7889578)     lv_image_set_src: failed to get image info: E:/lvgl/git/lvgl-master/examples/libs/bmp/example_32bit.bmp lv_image.c:175
[Warn]  (7889.578, +0)   _lv_display_refr_timer: No draw buffer lv_refr.c:349

### How to reproduce?

run the lv_example_bmp_1.c example, in widows

kisvegabor, 2024-01-08:
`E:` is not your drive on Window but a File system identifier for LVGL. See [here](https://docs.lvgl.io/master/overview/fs.html).

xinsuinizhuan, 2024-01-09:
> `E:` is not your drive on Window but a File system identifier for LVGL. See [here](https://docs.lvgl.io/master/overview/fs.html).

I am still don't know how to do, i only want to load a image by gived image path? how to do, for example?
 i try to do below, but nod sucess:
lv_conf.h:
[image]
the run:
void lv_example_libjpeg_turbo_1(void)
{
	lv_obj_t* wp;

	wp = lv_image_create(lv_screen_active());
	/* Assuming a File system is attached to letter 'A'
	 * E.g. set LV_USE_FS_STDIO 'A' in lv_conf.h */
	lv_image_set_src(wp, "E:lvgl/examples/libs/libjpeg_turbo/flower.jpg");
	lv_obj_center(wp);
}
but still could not find image:
[image]

kisvegabor, 2024-01-10:
Have you installed and enabled turbo-jpeg? If not I suggest using TJPGD

closed 2024-02-02 by lvgl-bot