lvgl/lvgl #5215: how to use LV_DRAW_BUF_DEFINE?

xinsuinizhuan, 2024-01-08:
### LVGL version

v9.0.0

### What happened?

i run the lv_example_canvas_1, add the project to myself project, but could not recognise the LV_DRAW_BUF_DEFINE?
how to use the LV_DRAW_BUF_DEFINE, and where define he draw_buf_16bpp? I search all the file, but could not find the draw_buf_16bpp define
[image]

### How to reproduce?

void lv_example_canvas_1(void)
{
	lv_draw_rect_dsc_t rect_dsc;
	lv_draw_rect_dsc_init(&rect_dsc);
	rect_dsc.radius = 10;
	rect_dsc.bg_opa = LV_OPA_COVER;
	rect_dsc.bg_grad.dir = LV_GRAD_DIR_VER;
	rect_dsc.bg_grad.stops[0].color = lv_palette_main(LV_PALETTE_RED);
	rect_dsc.bg_grad.stops[0].opa = LV_OPA_100;
	rect_dsc.bg_grad.stops[1].color = lv_palette_main(LV_PALETTE_BLUE);
	rect_dsc.bg_grad.stops[1].opa = LV_OPA_50;
	rect_dsc.border_width = 2;
	rect_dsc.border_opa = LV_OPA_90;
	rect_dsc.border_color = lv_color_white();
	rect_dsc.shadow_width = 5;
	rect_dsc.shadow_offset_x = 5;
	rect_dsc.shadow_offset_y = 5;

	lv_draw_label_dsc_t label_dsc;
	lv_draw_label_dsc_init(&label_dsc);
	label_dsc.color = lv_palette_main(LV_PALETTE_ORANGE);
	label_dsc.text = "Some text on text canvas";
	/*Create a buffer for the canvas*/
	LV_DRAW_BUF_DEFINE(draw_buf_16bpp, CANVAS_WIDTH, CANVAS_HEIGHT, LV_COLOR_FORMAT_RGB565);

	lv_obj_t* canvas = lv_canvas_create(lv_screen_active());
	lv_canvas_set_draw_buf(canvas, &draw_buf_16bpp);
	lv_obj_center(canvas);
	lv_canvas_fill_bg(canvas, lv_palette_lighten(LV_PALETTE_GREY, 3), LV_OPA_COVER);

	lv_layer_t layer;
	lv_canvas_init_layer(canvas, &layer);

	lv_area_t coords_rect = { 30, 20, 100, 70 };
	lv_draw_rect(&layer, &rect_dsc, &coords_rect);

	lv_area_t coords_text = { 40, 80, 100, 120 };
	lv_draw_label(&layer, &label_dsc, &coords_text);

	lv_canvas_finish_layer(canvas, &layer);

	/*Test the rotation. It requires another buffer where the original image is stored.
	 *So use previous canvas as image and rotate it to the new canvas*/
	LV_DRAW_BUF_DEFINE(draw_buf_32bpp, CANVAS_WIDTH, CANVAS_HEIGHT, LV_COLOR_FORMAT_ARGB8888);
	/*Create a canvas and initialize its palette*/
	canvas = lv_canvas_create(lv_screen_active());
	lv_canvas_set_draw_buf(canvas, &draw_buf_32bpp);
	lv_canvas_fill_bg(canvas, lv_color_hex3(0xccc), LV_OPA_COVER);
	lv_obj_center(canvas);

	lv_canvas_fill_bg(canvas, lv_palette_lighten(LV_PALETTE_GREY, 1), LV_OPA_COVER);

	lv_canvas_init_layer(canvas, &layer);
	lv_image_dsc_t img;
	lv_draw_buf_to_image(&draw_buf_16bpp, &img);
	lv_draw_image_dsc_t img_dsc;
	lv_draw_image_dsc_init(&img_dsc);
	img_dsc.rotation = 120;
	img_dsc.src = &img;
	img_dsc.pivot.x = CANVAS_WIDTH / 2;
	img_dsc.pivot.y = CANVAS_HEIGHT / 2;

	lv_area_t coords_img = { 0, 0, CANVAS_WIDTH - 1, CANVAS_HEIGHT - 1 };
	lv_draw_image(&layer, &img_dsc, &coords_img);

	lv_canvas_finish_layer(canvas, &layer);
}

xinsuinizhuan, 2024-01-09:
> See `LV_DRAW_BUF_DEFINE` [here](https://github.com/lvgl/lvgl/blob/master/src/draw/lv_draw_buf.h#L51-L65). So it declares the `draw_buf_16bpp` variable.

Oh, i see. LV_DRAW_BUF_DEFINE declares the draw_buf_16bpp variable. But now it could not recognise the LV_DRAW_BUF_DEFINE, how should i do?

kisvegabor, 2024-01-09:
The built in `lv_example_canvas_1()` is still failing?

cc @XuNeo

XuNeo, 2024-01-09:
Just tested with lvgl simulator, `lv_example_canvas_1 ` works as expected.
I think the issue reported is related to IDE setup.

closed 2024-02-01 by lvgl-bot