lvgl/lvgl #5220: Why lvgl flush area is larger than the real area?

lanistor, 2024-01-08:
### Introduce the problem

Using `flush_cb` to paint in epaper, found that the flush area always larger than real object area.
Such as repaint a object with area `{x: 100, y: 100, width: 100, height: 100}`, the `flush_cb` will repaint the area `{x: 95, y: 95, width: 110, height: 110}`.

I don't know why, but this cause a problem on epaper: rendering ghost cannot be aligned with the edge of the element, but exceeds the element.

kisvegabor, 2024-01-08:
Is this an `lv_btn` with the default theme? If so, it has a slight shadow which increases its draw area. See the grey line here:
[image]

lanistor, 2024-01-08:
> Is this an `lv_btn` with the default theme? If so, it has a slight shadow which increases its draw area. See the grey line here: [...]

No, it's just a normal lv_obj_t, created by `lv_obj_create`, all the objects has the same behavior, i thought it's a rule of lvgl.

kisvegabor, 2024-01-08:
Hm, it shouldn't be the default behavior. 

Which LVGL version do you use?

What if you call `lv_obj_remove_style_all(obj)` right after creating that widget?

lanistor, 2024-01-09:
> Hm, it shouldn't be the default behavior.
> 
> Which LVGL version do you use?
> 
> What if you call `lv_obj_remove_style_all(obj)` right after creating that widget?

Using 8.x (tested 8.2, 8.3).

`lv_obj_remove_style_all` is useless.

lanistor, 2024-01-24:
This problem still exists, how should I solve it?

kisvegabor, 2024-01-24:
If you can use a debugger (or printf-s) could you check what happens in [_lv_inv_area](https://github.com/lvgl/lvgl/blob/master/src/core/lv_refr.c). This is the final functions to mark an area as invalidated. Does the input `area_p` parameter look good?

lanistor, 2024-01-28:
> If you can use a debugger (or printf-s) could you check what happens in [_lv_inv_area](https://github.com/lvgl/lvgl/blob/master/src/core/lv_refr.c). This is the final functions to mark an area as invalidated. Does the input `area_p` parameter look good?

The `area_p` of `_lv_inv_area` has problem already, i call `update_widget` to update an area, here is the log:
```cpp
W (88153) ####: update_widget; area: x1: 477; y1: 0; x2: 545; y2: 28;
W (88153) ####: _lv_inv_area area. x1: 472; y1: -5; x2: 550; y2: 33;
W (88153) ####: _lv_inv_area area. x1: 472; y1: -5; x2: 487; y2: 10;
...
W (88293) ####: epdiy_flush. area. x1: 472; y1: 0; x2: 550; y2: 33;
```

The `_lv_inv_area` log is on the top of the function: 
```cpp
void _lv_inv_area(lv_disp_t * disp, const lv_area_t * area_p)
{
    ESP_LOGW("####", "_lv_inv_area area. "
    "x1: %d; y1: %d; x2: %d; y2: %d;", area_p->x1, area_p->y1, area_p->x2, area_p->y2);
    if(!disp) disp = lv_disp_get_default();
    if(!disp) return;
    if(!lv_disp_is_invalidation_enabled(disp)) return;
...
}
```

The `update_widget` is like this:
```cpp
  lv_obj_clean(this->view);
  this->create_children();
```

kisvegabor, 2024-01-29:
Ah, I know what the problem is now. LVGL sets a slightly larger draw area for the labels. It's because some letters are overflowing their box and they would be clipped at the edges of the widget. 

You can do 2 things:
1. Wrap the labels in to an `lv_obj` which will clip them
2. Overwrite LVGL's internal setting for the larger clip area:

```c
static void event_cb(lv_event_t)
{
    int32_t * size = lv_event_get_param(e);
    *size = 0;
}

lv_obj_add_event_cb(label, event_cb, LV_EVENT_REFR_EXT_DRAW_SIZE, NULL);
``` 

Of course you can create a simple wrapper like `my_label_create` which creates a label and adds this event.

kisvegabor, 2024-01-30:
On v8 it work for me with this modification:
```c
static void label_init_event_cb(lv_event_t* e) {
    lv_coord_t* size = (int32_t*)lv_event_get_param(e); //Use lv_coord_t instead of int32_t
  *size         = 0;
}
```

lanistor, 2024-01-31:
> On v8 it work for me with this modification:
> 
> ```c
> static void label_init_event_cb(lv_event_t* e) {
>     lv_coord_t* size = (int32_t*)lv_event_get_param(e); //Use lv_coord_t instead of int32_t
>   *size         = 0;
> }
> ```

I don't know why. I update to the latest release/v8.3, still not work, event i add to both the label wrapper and the label, here is my code (the update_widget function will called period):
```cpp
static void label_init_event_cb(lv_event_t* e) {
  //Use lv_coord_t instead of int32_t
  lv_coord_t* size = (lv_coord_t*)lv_event_get_param(e);
  *size            = 0;
}

static lv_style_t temp_cont_style;

lv_obj_t* status_bar_battery_debug_widget::create_widget() {
  lv_style_init(&temp_cont_style);
  lv_style_set_pad_all(&temp_cont_style, 0);

  this->view = lv_obj_create(par);
  lv_obj_add_style(view, &temp_cont_style, LV_PART_MAIN);
  lv_obj_set_size(view, LV_SIZE_CONTENT, lv_pct(100));
  add_border(view, LV_BORDER_SIDE_FULL, color_make(0x99), 1); // wrapper to add border, nothing else
  lv_obj_add_event_cb(view, label_init_event_cb, LV_EVENT_REFR_EXT_DRAW_SIZE, NULL);
  return view;
}

void status_bar_battery_debug_widget::create_children() {
  auto label = lv_label_create(view);
  lv_obj_add_event_cb(label, label_init_event_cb, LV_EVENT_REFR_EXT_DRAW_SIZE, NULL);
  lv_label_set_text(
    label, string("battery: ")
             .append(to_string(debug_battery_mv))
             .append(" mv")
             .c_str());
}

void status_bar_battery_debug_widget::update_widget() {
  lv_obj_clean(this->view);
  this->create_children();
}
```
Here is the result video:
https://github.com/lanistor/assets/blob/master/lvgl/lvgl_label_wrapper.mp4

How to clip the label wrapper lv_obj?

kisvegabor, 2024-02-01:
Please check what happens in `lv_obj_pos.c` in `lv_obj_invalidate`. What is the [`ext_size`](https://github.com/lvgl/lvgl/blob/release/v8.3/src/core/lv_obj_pos.c#L857)?

lanistor, 2024-02-02:
> Please check what happens in `lv_obj_pos.c` in `lv_obj_invalidate`. What is the [`ext_size`](https://github.com/lvgl/lvgl/blob/release/v8.3/src/core/lv_obj_pos.c#L857)?

It's value is `4`,`5` or `6`, same as the extra size.

lanistor, 2024-02-02:
> Please check what happens in `lv_obj_pos.c` in `lv_obj_invalidate`. What is the [`ext_size`](https://github.com/lvgl/lvgl/blob/release/v8.3/src/core/lv_obj_pos.c#L857)?

It's not only the `label`, or the `ext_draw_size`, even i change to `lv_obj_t`, the problem still exists:
```cpp
static void _the_label_init_event_cb(lv_event_t* e) {
  //Use lv_coord_t instead of int32_t
  lv_coord_t* size = (lv_coord_t*)lv_event_get_param(e);
  *size            = 0;
  if(e->target->spec_attr) {
    e->target->spec_attr->ext_draw_size = 0;
  }
}

static lv_style_t temp_cont_style;

// 电池电压调试
lv_obj_t* status_bar_battery_debug_widget::create_widget() {
  lv_style_init(&temp_cont_style);
  lv_style_set_border_width(&temp_cont_style, 0);
  lv_style_set_pad_all(&temp_cont_style, 0);
  lv_style_set_radius(&temp_cont_style, 0);
  this->view = lv_obj_create(par);
  lv_obj_add_style(view, &temp_cont_style, LV_PART_MAIN);

  lv_obj_set_size(view, LV_SIZE_CONTENT, lv_pct(100));
  lv_obj_set_style_text_color(view, color_make(0x00), LV_PART_MAIN);
  lv_obj_add_event_cb(view, _the_label_init_event_cb, LV_EVENT_REFR_EXT_DRAW_SIZE, NULL);
  return view;
}

static int the_count = 0;

void status_bar_battery_debug_widget::create_children() {
  auto child = lv_obj_create(view);
  lv_obj_set_size(child, 100, lv_pct(100));
  lv_obj_add_style(child, &temp_cont_style, LV_PART_MAIN);
  lv_obj_add_event_cb(child, _the_label_init_event_cb, LV_EVENT_REFR_EXT_DRAW_SIZE, NULL);
  ++the_count;
  if (the_count % 2) {
    lv_obj_set_style_bg_color(child, color_make(0x00), LV_PART_MAIN);
  }
}

// lv_event.c
void lv_event_set_ext_draw_size(lv_event_t * e, lv_coord_t size)
{
    if(e->code == LV_EVENT_REFR_EXT_DRAW_SIZE) {
        lv_coord_t * cur_size = lv_event_get_param(e);
        // *cur_size = LV_MAX(*cur_size, size);
        *cur_size = 0;
    }
    else {
        LV_LOG_WARN("Not interpreted with this event code");
    }
}

// lv_obj_draw.c
lv_coord_t _lv_obj_get_ext_draw_size(const lv_obj_t * obj)
{
    return 0;
    // if(obj->spec_attr) return obj->spec_attr->ext_draw_size;
    // else return 0;
}
```

Here is the video:

https://github.com/lvgl/lvgl/assets/15553840/8866c89b-1b11-408d-8b54-785241800c49

kisvegabor, 2024-02-02:
So do you say not only the label but all widget has larger draw area?

Have you used a `display_drv.rounder_cb`?

lanistor, 2024-02-04:
> So do you say not only the label but all widget has larger draw area?
> 
> Have you used a `display_drv.rounder_cb`?

Yeah, all widget has larger draw area, i didn't use a `display_drv.rounder_cb`, i remove all `rounder_cb` called in `lvgl`, the problem still exists.

But, if the repaint is called by `lv_obj_set_style_bg_color`, it won't draw larger.

kisvegabor, 2024-02-07:
> But, if the repaint is called by lv_obj_set_style_bg_color, it won't draw larger.

Maybe we are missing something fundamental. What/How do change when you see the issue? I assumed you just call `lv_label_set_text`

lanistor, 2024-02-07:
> > But, if the repaint is called by lv_obj_set_style_bg_color, it won't draw larger.
> 
> Maybe we are missing something fundamental. What/How do change when you see the issue? I assumed you just call `lv_label_set_text`

This is the test code:

```cpp

static void _the_label_init_event_cb(lv_event_t* e) {
  //Use lv_coord_t instead of int32_t
  lv_coord_t* size = (lv_coord_t*)lv_event_get_param(e);
  *size            = 0;
  if (e->target->spec_attr) {
    e->target->spec_attr->ext_draw_size = 0;
  }
}

static lv_style_t temp_cont_style;

// 电池电压调试
lv_obj_t* status_bar_battery_debug_widget::create_widget() {
  lv_style_init(&temp_cont_style);
  lv_style_set_border_width(&temp_cont_style, 0);
  lv_style_set_pad_all(&temp_cont_style, 0);
  lv_style_set_radius(&temp_cont_style, 0);

  this->view = lv_obj_create(par);
  lv_obj_add_style(view, &temp_cont_style, LV_PART_MAIN);

  lv_obj_set_size(view, LV_SIZE_CONTENT, lv_pct(100));
  lv_obj_set_style_text_color(view, color_make(0x00), LV_PART_MAIN);
  return view;
}

static int the_count = 0;

void status_bar_battery_debug_widget::create_children() {
  auto child = lv_obj_create(view);
  lv_obj_set_size(child, 100, lv_pct(100));
  lv_obj_add_style(child, &temp_cont_style, LV_PART_MAIN);
  lv_obj_add_event_cb(child, _the_label_init_event_cb,
                      LV_EVENT_REFR_EXT_DRAW_SIZE, NULL);
  ++the_count;
  if (the_count % 2) {
    lv_obj_set_style_bg_color(child, color_make(0x00), LV_PART_MAIN);
  }
}
```

kisvegabor, 2024-02-07:
I see how you creat the widgets and it looks good, but how do you update the values?

lanistor, 2024-02-07:
> I see how you creat the widgets and it looks good, but how do you update the values?

The `update_widget` will called by period.
```cpp
void status_bar_battery_debug_widget::update_widget() {
  lv_obj_clean(this->view);
  this->create_children();
}
```

kisvegabor, 2024-02-07:
Ah, it's not ideal. You should call only lv_obj_set_style_bg_color. As you create the widgets some extra areas are invalidated. So instead of continuously creating/deleting widgets, you can call only the "set" functions.

lanistor, 2024-02-07:
> Ah, it's not ideal. You should call only lv_obj_set_style_bg_color. As you create the widgets some extra areas are invalidated. So instead of continuously creating/deleting widgets, you can call only the "set" functions.

The `create_children` function may be really complex, and may have lots of logical branch, so clean and rebuild it is a simple and reusable way.
If using this way to update views, will this problem be resolved? Or if you have another better way to organize `lv_obj_t` to combine?

kisvegabor, 2024-02-08:
Well, I see that recreating widgets can be tempting to handle complex logic, but this way you will always have this updating area issue. Imagine that there are 2 labels and only the first one needs to be updated. However you will delete and recreate both, so LVGL needs to redraw both of them.

If there is a larger state or mode change when new widgets are really required (I guess it's rare), you can delete/recreate the widgets. But just to update a label I recommend building up some extra logic, to see in which mode you are and call only `lv_label_set_text`.

Alternatively, you can disable LVGL's widget invalidation before widget deletion/creation by calling `lv_display_enable_invalidation(NULL, false)` and enable it after that. However in this case you need to take care of the invalidation which could be more complicated. You need to know which widget will change and invalidate the old area, and invalidate the area of the new widgets too.

lanistor, 2024-03-19:
> Well, I see that recreating widgets can be tempting to handle complex logic, but this way you will always have this updating area issue. Imagine that there are 2 labels and only the first one needs [...]

@kisvegabor  Even only call `lv_label_set_text` to update text, the area is still larger, even i add `LV_EVENT_REFR_EXT_DRAW_SIZE` or change `lv_event_set_ext_draw_size`, `_lv_obj_get_ext_draw_size` in lvgl's source code to clear `ext_draw_size`.

So i think this is not the `ext_draw_size`'s reason.

kisvegabor, 2024-03-20:
Just to be sure we are talking about the same thing, please try this:
```c

  lv_obj_t * label = lv_label_create(lv_screen_active());
  lv_obj_center(label);
  lv_label_set_text(label, "Hello");
  lv_refr_now(NULL);
  your_delay_ms(1000);
  
  lv_label_set_text(label, "This is longer");
  lv_refr_now(NULL);
  your_delay_ms(1000);
```

And send a video about the result.

lanistor, 2024-03-27:
> Just to be sure we are talking about the same thing, please try this:
>
> ```c
>   lv_obj_t * label = lv_label_create(lv_screen_active());
>   lv_obj_center(label);
>   lv_label_set_text(label, "Hello"); [...]

Thanks, here is the video:
https://github.com/lvgl/lvgl/assets/15553840/499e0bf3-f716-4af0-8104-807418fa8198

And here is the demo code:

```cpp
  lv_obj_t* label = lv_label_create(lv_scr_act());
  lv_obj_center(label);
  lv_label_set_text(label, "Hello");
  lv_refr_now(NULL);

  fl_lvgl_delay(
    [=](void* arg) {
      lv_label_set_text(label, "This is longer");
      lv_refr_now(NULL);
    },
    1000);

  while (1) {
    if (pdTRUE == xSemaphoreTake(xGuiSemaphore, portMAX_DELAY)) {
      lv_task_handler();
      xSemaphoreGive(xGuiSemaphore);
    }

    vTaskDelay(pdMS_TO_TICKS(10));
  }
```

kisvegabor, 2024-03-28:
Ah, I've found something: if I replace `lv_obj_center()` with `lv_obj_set_pos(label, 100, 200);` it's much better. This is what happens with `lv_obj_center`:
1. Let's say the width of `"Hello"` is 100px and when it's center aligned it has `x=200px`
2. When the text is changed to `"This is longer"`width will be 250px and it's area is invalidated. But it's position is not updated yet. So the 100-350 band is invalidated.
3. Finally the position is updated too and the label is shifted to the left to align it to the center. But the too large are is already invalidated.

With `set_pos` the x coordinate is always the same so it works better.

I don't have a really good solution as this mechanism is very deeply embedded into LVGL, however I have two workarounds:
1. use `lv_obj_set_pos` instead of `lv_obj_center` 
2. 
```c
  lv_obj_invalidate(label); //Invalidate the old area
  lv_display_enable_invalidation(NULL, false);
  lv_label_set_text(label, "This is longer");
  lv_obj_update_layout(label);
  lv_display_enable_invalidation(NULL, true);
  lv_obj_invalidate(label); //Invalidate the new area
```

lanistor, 2024-03-29:
> Ah, I've found something: if I replace `lv_obj_center()` with `lv_obj_set_pos(label, 100, 200);` it's much better. This is what happens with `lv_obj_center`:
> 
> 1. Let's say the width of `"Hello"` is 100px and when it's center aligned it has `x=200px`
> 2. When the text is changed to `"This is longer"`width will be 250px and it's area is invalidated. But it's position is not updated yet. So the 100-350 band is invalidated.
> 3. Finally the position is updated too and the label is shifted to the left to align it to the center. But the too large are is already invalidated.
> 
> With `set_pos` the x coordinate is always the same so it works better.
> 
> I don't have a really good solution as this mechanism is very deeply embedded into LVGL, however I have two workarounds:
> 
> 1. use `lv_obj_set_pos` instead of `lv_obj_center`
> 
> ```c
>   lv_obj_invalidate(label); //Invalidate the old area
>   lv_display_enable_invalidation(NULL, false);
>   lv_label_set_text(label, "This is longer");
>   lv_obj_update_layout(label);
>   lv_display_enable_invalidation(NULL, true);
>   lv_obj_invalidate(label); //Invalidate the new area
> ```

Here is the new video, is this expected:

https://github.com/lvgl/lvgl/assets/15553840/ca48dae2-990a-42e2-ab48-4386e401d57d

kisvegabor, 2024-03-29:
It looks good to me. Or am I missing something?

lanistor, 2024-03-29:
> @kisvegabor Even only call `lv_label_set_text` to update text, the area is still larger, even i add `LV_EVENT_REFR_EXT_DRAW_SIZE` or change `lv_event_set_ext_draw_size`, `_lv_obj_get_ext_draw_size` [...]

Will this code resolve the above problem?

lanistor, 2024-03-29:
> ### Introduce the problem
> Using `flush_cb` to paint in epaper, found that the flush area always larger than real object area. Such as repaint a object with area `{x: 100, y: 100, width: 100, height: 100}`, the `flush_cb` will repaint the area `{x: 95, y: 95, width: 110, height: 110}`.
> 
> I don't know why.
> 
> ### Proposal
> This cause a problem on epaper: rendering ghost cannot be aligned with the edge of the element, but exceeds the element. [image]

This is also the initial question.

kisvegabor, 2024-04-03:
Can you can minimal code to reproduce this as well?

lanistor, 2024-04-03:
> ```c++
> lv_obj_add_event_cb(child, _the_label_init_event_cb,
>                       LV_EVENT_REFR_EXT_DRAW_SIZE, NULL);
> ```

Here is the minimal code to reproduce,

```cpp
#include <stdbool.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#define USE_LV_LOG 1
#include "lv_conf.h"

#include "basic.h"
#include "fl_pm.h"
#include "fl_utils.h"
#include "gui_message.h"
#include "modules/app/index.h"
#include "modules/device/index.h"
#include <buzzer.h>
#include <time.h>

#define _GUI_TAG "GUI_TASK"

extern "C" {
void app_main();
}

static void gui_task(void*);  // UI初始化

static void lv_tick_task(void*);

lv_disp_drv_t disp_drv;

esp_timer_handle_t lvgl_tick_periodic_timer;

static void _the_label_init_event_cb(lv_event_t* e) {
  //Use lv_coord_t instead of int32_t
  lv_coord_t* size = (lv_coord_t*)lv_event_get_param(e);
  *size            = 0;
  if (e->target->spec_attr) {
    e->target->spec_attr->ext_draw_size = 0;
  }
}

void app_main() {
  gpio_install_isr_service(0);
  lvgl_driver_init();

  lv_init();

  xTaskCreatePinnedToCore(gui_task, "gui_task", 1024 * 16, NULL, 5,
                          get_gui_task_handler(), 0);
}

/* Creates a semaphore to handle concurrent call to lvgl stuff
 * If you wish to call *any* lvgl function from other threads/tasks
 * you should lock on the very same semaphore! */
SemaphoreHandle_t xGuiSemaphore;

static void gui_task(void* arg) {
  (void)arg;
  ESP_LOGI(_GUI_TAG, "gui_task started");
  xGuiSemaphore = xSemaphoreCreateMutex();

  int display_width  = epd_rotated_display_width();
  int display_height = epd_rotated_display_height();
[... 121 lines cut ...]
  lv_obj_set_style_border_width(view, 1, LV_PART_MAIN);
  lv_obj_add_event_cb(view, _the_label_init_event_cb,
                      LV_EVENT_REFR_EXT_DRAW_SIZE, NULL);

  lv_obj_t* label = lv_label_create(view);
  lv_obj_set_size(label, lv_pct(100), lv_pct(100));
  lv_label_set_text(label, "Hello");
  lv_obj_add_event_cb(label, _the_label_init_event_cb,
                      LV_EVENT_REFR_EXT_DRAW_SIZE, NULL);
  lv_refr_now(NULL);

  fl_lvgl_delay(
    [=](void* arg) {
      lv_obj_invalidate(label);  //Invalidate the old area
      lv_disp_enable_invalidation(NULL, false);
      lv_label_set_text(label, "This is longer");
      lv_obj_update_layout(label);
      lv_disp_enable_invalidation(NULL, true);
      lv_obj_invalidate(label);  //Invalidate the new area

      lv_refr_now(NULL);
    },
    2000);
}
```

kisvegabor, 2024-04-15:
I've looked into the example code but there are custom things in it and I can't compile it on my end. :slightly_frowning_face: 
What I would need is only a short snippet for the rectangle in question.

lanistor, 2024-05-27:
I found the effective code in lvgl, it's `lv_area_increase(area, 5, 5);`:
```cpp
void lv_obj_get_transformed_area(const lv_obj_t * obj, lv_area_t * area, bool recursive,
                                 bool inv)
{
    lv_point_t p[4] = {
        {area->x1, area->y1},
        {area->x1, area->y2},
        {area->x2, area->y1},
        {area->x2, area->y2},
    };

    lv_obj_transform_point(obj, &p[0], recursive, inv);
    lv_obj_transform_point(obj, &p[1], recursive, inv);
    lv_obj_transform_point(obj, &p[2], recursive, inv);
    lv_obj_transform_point(obj, &p[3], recursive, inv);

    area->x1 = LV_MIN4(p[0].x, p[1].x, p[2].x, p[3].x);
    area->x2 = LV_MAX4(p[0].x, p[1].x, p[2].x, p[3].x);
    area->y1 = LV_MIN4(p[0].y, p[1].y, p[2].y, p[3].y);
    area->y2 = LV_MAX4(p[0].y, p[1].y, p[2].y, p[3].y);
    lv_area_increase(area, 5, 5);
}
```

Can this be safely removed? Are there any negative effects of removing it?

kisvegabor, 2024-06-04:
If you don't use and widget transformations (`style_transform_*`) it can be safely removed. 

(In v9 it should be already resolved)

lanistor, 2024-06-04:
> If you don't use and widget transformations (`style_transform_*`) it can be safely removed.
> 
> (In v9 it should be already resolved)

Got it, thanks👍

closed 2024-06-04 by lanistor