lvgl/lvgl #5224: Draw line from array instead of 2 points

kisvegabor, 2024-01-08:
### Introduce the problem

Now we `lv_draw_line_dsc_t` has only 2 points which is a problem because:
- A GPU cannot draw all the lines at once
- The draw unit cannot handle the line joints

### Proposal

Make `lv_draw_line_dsc_t` accept an array of points. Notes:
- `lv_chart` needs to allocate a point array, which is saved in draw task and freed when drawing is ready. It temporarily increases the memory usage .
- We can consider keep the 2 points too, because in several cases  `e.g. lv_scale` only  one line segment is drawn so we can save the allocation.

An other option would be to have an array of fixed 8 points where we can pass maximum of 8 points at once. In case of the chart we can draw points like this:
1. Set the clip area to include point 0..6  (7 points) (the last point is just to have the right joint for point 6)
2. Draw the first 8 points (0..7) (the 7th will be clipped)
3. Set the clip area to include point 7..13  (7 points) (the last point is just to have the right joint for point 13)
4. Draw the point 7..14 (the 14th will be clipped)
5. and so on

kisvegabor, 2024-01-09:
I've realized that it should be a chart feature to draw vector lines instead of normal lines.

closed 2024-01-09 by kisvegabor

XuNeo, 2024-01-09:
Hi Gabor,

I'm thinking of include two points in `lv_draw_line_dsc_t ` by default, and allow to use variable length of array if more points are needed.

```c
struct _lv_draw_line_dsc_t {
  uint32_t point_count;
  lv_points_t points[2]; /*2 or more points array depending on `point_count` */
};
```

I think it's possible for GPUs to draw multiple lines in single run.

kisvegabor, 2024-01-10:
My main concern is the life cycle management of the `points` array. E.g. for the chart it needs to be allocated which might consume significant extra memory + it's cumbersome. Besides it doesn't really open the door for smooth curve drawing whereas vector drawing mode does.