lvgl/lvgl #5225: [Micropython] feat/multi-instance, canvas.init_layer() is freezing after Jan 4 commit

sam0910, 2024-01-08:
### LVGL version

v9.0.0

### What happened?

Compiling feat/multi-instance branch with [lvgl@e217674](https://github.com/lvgl/lvgl/commit/e217674295204785f3f61e299929f2943a97fdae) ([#5161](https://github.com/lvgl/lvgl/pull/5161)) ```chore: move lv_obj_t define to lv_types.h```
it was Jan 4, after that commit, canvas.init_layer() is causing the board freezing.
lv.image(), lv.label() is working normally, only happens on canvas.init_layer().

[lvgl@e217674295](https://github.com/lvgl/lvgl/commit/e217674295204785f3f61e299929f2943a97fdae) is working fine.

ESP32 + SPIRAM, ESP-IDF v4.4.6, GC9A01 display.

### How to reproduce?

```
cbuf = bytearray(100 * 100 * 4)
canvas = lv.canvas(lv.screen_active())
canvas.set_buffer(cbuf, 100, 100, lv.COLOR_FORMAT.NATIVE)
canvas.fill_bg(lv.color_hex(0xCCCCCC), lv.OPA.COVER)
canvas.center()

layer = lv.layer_t()
canvas.init_layer(layer)
```

kisvegabor, 2024-01-08:
@XuNeo Can it be related to the changes [here](https://github.com/lvgl/lvgl/pull/5132/files#diff-44e6c48dcb2ea880a01bbae515ecb86bba087c705413c45a37e7f9eccd43ab5b)?

XuNeo, 2024-01-09:
I cannot reproduce it using ports/unix simulator.
Board freezing could be due to an assert happened. Enable all logs and check where it reports error.

sam0910, 2024-01-09:
It is not happening on latest PR on lv_binding_micropython, [Update LVGL to v9 prerelease version #311](https://github.com/lvgl/lv_binding_micropython/pull/311)

It will be merged soon, It is solved!
Thank you so much for the concern, and I am very grateful to use latest v9 !

closed 2024-01-09 by sam0910