lvgl/lvgl #5226: lvgl_lock(-1) caused watchdog reset

zamek42, 2024-01-08:
### Introduce the problem

lock with -1 means wait forever for a resource. (lcd.c). Sometimes it happens and activates watchdog timer. 
 `
static void lvgl_port_task(void *arg)
{
    ESP_LOGI(TAG, "Starting LVGL task");
    ESP_ERROR_CHECK(esp_task_wdt_add(NULL));
    ESP_ERROR_CHECK(esp_task_wdt_status(NULL));
    uint32_t task_delay_ms = LVGL_TASK_MAX_DELAY_MS;
    while (1) {
        esp_task_wdt_reset();
        // Lock the mutex due to the LVGL APIs are not thread-safe
        if (lvgl_lock(-1)) {	
            task_delay_ms = lv_timer_handler();
            // Release the mutex
            lvgl_unlock();
        }
        if (task_delay_ms > LVGL_TASK_MAX_DELAY_MS) {
            task_delay_ms = LVGL_TASK_MAX_DELAY_MS;
        } else if (task_delay_ms < LVGL_TASK_MIN_DELAY_MS) {
            task_delay_ms = LVGL_TASK_MIN_DELAY_MS;
        }

        vTaskDelay(pdMS_TO_TICKS(task_delay_ms));
    }
}
`

### Proposal

I suggest to use exact ms for a locking and logging if cannot

kisvegabor, 2024-01-08:
`lvgl_lock` is not implemented by us. Is it coming from ESP-IDF?

Anyway, it can wait long  if you keep LVGL locked long in an other thread, which shouldn't be required. You should keep it locked only while you are calling some functions (e.g. to the a label's text).

zamek42, 2024-01-09:
it is not my code, I always lock it only until refreshing. This code is is in lcd.c.  
When a thread cannot get a lock infinite time, it means, something went wrong, but we don't have an information about it. We can detect this, by wdt reset. But, if we use couple of msesc intead of -1, we can detect it and we can log it.

kisvegabor, 2024-01-09:
I think `lcd.c` is created by Espressif, so I cannot comment much about the intended usage. :slightly_frowning_face:

closed 2024-02-01 by lvgl-bot