lvgl/lvgl #5235: Freetype failed when run LVGL test locally

XuNeo, 2024-01-09:
### LVGL version

master

### What happened?

CC @W-Mai 

Run the test using `./tests/main.py test`, it report a slightly different snapshot image.

lvgl/tests/src/test_cases/libs/test_freetype.c:172:test_freetype_bitmap_rendering_test:INFO:
Screenshot compare error
  - File: ref_imgs/libs/freetype_1.lp64.png
  - Mode: full refresh
  - At x:89, y:17.
  - Expected: 0x00DBDBDB
  - Actual:   0x00DCDCDC
  
[image]
[image]

XuNeo, 2024-01-09:
It also reports memory leak.

```bash
==2608794==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 112 byte(s) in 1 object(s) allocated from:
    #0 0x7fd7260a1808 in __interceptor_malloc ../../../../src/libsanitizer/asan/asan_malloc_linux.cc:144
    #1 0x5624876ea90d in lv_malloc_core /home/neo/projects/lvgl/lv_port_pc_eclipse/lvgl/src/stdlib/clib/lv_mem_core_clib.c:64
    #2 0x5624876eb006 in lv_malloc_zeroed /home/neo/projects/lvgl/lv_port_pc_eclipse/lvgl/src/stdlib/lv_mem.c:100
    #3 0x5624876b382d in lv_freetype_font_create /home/neo/projects/lvgl/lv_port_pc_eclipse/lvgl/src/libs/freetype/lv_freetype.c:152
    #4 0x562487647c46 in test_freetype_outline_rendering_test /home/neo/projects/lvgl/lv_port_pc_eclipse/lvgl/tests/src/test_cases/libs/test_freetype.c:468
    #5 0x562487648b22 in run_test /home/neo/projects/lvgl/lv_port_pc_eclipse/lvgl/tests/build_test_sysheap/test_freetype_Runner.c:70
    #6 0x562487648bc5 in main /home/neo/projects/lvgl/lv_port_pc_eclipse/lvgl/tests/build_test_sysheap/test_freetype_Runner.c:88
    #7 0x7fd72523a082 in __libc_start_main ../csu/libc-start.c:308

SUMMARY: AddressSanitizer: 112 byte(s) leaked in 1 allocation(s).
```

W-Mai, 2024-01-09:
mem leak fix is here https://github.com/lvgl/lvgl/pull/5238

Then the difference in font rendering may be caused by different freetype versions.

The highest on `ubuntu 20.04` is `freetype 2.10`, and the current ci uses `freetype 2.11`

kisvegabor, 2024-01-09:
Passing locally for me with `freetype 2.10.1`.

XuNeo, 2024-01-09:
That's really weird. I finally decide to upgrade to ubuntu 22.04. And the issue is immediately gone.

kisvegabor, 2024-01-09:
> I finally decide to upgrade to ubuntu 22.04.

Brave move :slightly_smiling_face:

kisvegabor, 2024-01-09:
I think it's basically impossible to figure out the root of the issue, so I would close this.

XuNeo, 2024-01-09:
Agree.

closed 2024-01-09 by XuNeo