lvgl/lvgl #5237: After scaling the image, there is a slight abnormality in the edges of the image.

IAMMX, 2024-01-09:
### LVGL version

v9.0.0

### What happened?

After scaling the image, there is a slight abnormality in the edges of the image.
[image]

### How to reproduce?

```c
LV_IMG_DECLARE(img_test);
void img_scale_test(void)
{
  lv_obj_t *logo1;
  logo1 = lv_image_create(lv_screen_active());
  lv_image_set_src(logo1, &img_test);
  lv_obj_align(logo1,LV_ALIGN_CENTER,-110,0);
  lv_image_set_scale(logo1, 256);

  lv_obj_t *logo2;
  logo2 = lv_image_create(lv_screen_active());
  lv_image_set_src(logo2, &img_test);
  lv_obj_align(logo2,LV_ALIGN_CENTER,110,0);
  lv_image_set_scale(logo2, 255);
}
```

kisvegabor, 2024-01-11:
Hi, 

thank you for the report. Please attach the test image too.

IAMMX, 2024-01-11:
[image]

IAMMX, 2024-01-11:
[image]

IAMMX, 2024-01-11:
[image]

IAMMX, 2024-01-26:
no stale

kisvegabor, 2024-01-26:
Should be related to https://github.com/lvgl/lvgl/issues/5447

IAMMX, 2024-01-26:
ok👌

lhdjply, 2024-02-06:
related to #5548 ,please test it

kisvegabor, 2024-02-22:
It turned out that it's hard to solve rounding error. It would require a large rework on the transformation system.

I close it for now.

closed 2024-02-22 by kisvegabor