lvgl/lvgl #5241: Use `LV_PCT` is not correct in `lv_obj_align_to`

IAMMX, 2024-01-09:
### LVGL version

v9.0.0

### What happened?

Use `LV_PCT` is not correct in `lv_obj_align_to`

### How to reproduce?

```c
void lv_pct_test(void)
{
  lv_obj_t *obj1,*obj2;

  obj1=lv_obj_create(lv_screen_active());
  lv_obj_set_style_bg_color(obj1,lv_palette_main(LV_PALETTE_RED),LV_PART_MAIN);
  lv_obj_align(obj1,LV_ALIGN_TOP_MID,0,0);

  obj2=lv_obj_create(lv_screen_active());
  lv_obj_set_style_bg_color(obj2,lv_palette_main(LV_PALETTE_BLUE),LV_PART_MAIN);
  lv_obj_align_to(obj2,obj1,LV_ALIGN_OUT_BOTTOM_MID,0,LV_PCT(1));
}
```

kisvegabor, 2024-01-09:
What would be the expected behavior in your opinion? Which object's which property's 1% shall be considered?

IAMMX, 2024-01-09:
> What would be the expected behavior in your opinion? Which object's which property's 1% shall be considered?

Both obj1 and obj2 are created on lv_screen_active. Once obj2 is moved down 1% based on obj1, it should normally be very close to obj1. But the actual result is that obj2 goes to the next page.

kisvegabor, 2024-01-09:
I mean 1% of 
- obj1
- obi2
- parent of obj1
- or parent of obj2

Normal `lv_obj_align` uses the parent size, in case of `lv_obj_align_to` obj1 and obj2 can be on different parents.

IAMMX, 2024-01-09:
I think it means 1% of  the obj size.But in the video, the 1% is so large.

kisvegabor, 2024-01-09:
Think more about it I agree to use `obj1` (base) for percentage calculation.

Adding this [here](https://github.com/lvgl/lvgl/blob/master/src/core/lv_obj_pos.c#L454) should fix it.  
```c
    if(LV_COORD_IS_PCT(x_ofs)) x_ofs = (lv_obj_get_width(base) * LV_COORD_GET_PCT(x_ofs)) / 100;
    if(LV_COORD_IS_PCT(y_ofs)) y_ofs = (lv_obj_get_height(base) * LV_COORD_GET_PCT(y_ofs)) / 100;
```

It's not clear for me if we should use the `content_width/height` (reduced by padding) or the normal width/height. As these alignment is usually used to position widgets next each other, I think normal width/height is a better choice.

IAMMX, 2024-01-10:
Oh，thank you

IAMMX, 2024-01-10:
let's continue in #5252

PR #5252 (2024-01-10): fix(obj): Use LV_PCT is not correct in lv_obj_align_to
Help us review this PR! Anyone can [approve it or request changes](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/reviewing-changes-in-pull-requests/approving-a-pull-request-with-required-reviews).

### Description of the feature or fix

fix #5241
fix(obj): Use LV_PCT is not correct in lv_obj_align_to

### Checkpoints
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.c [... cut]
closed 2024-01-10 by IAMMX