lvgl/lvgl #5243: Measure FReeRTOS idle time to see how much the GPU off-loads the CPU

kisvegabor, 2024-01-09:
### Introduce the problem

Now we measure the time spent in `lv_timer_handler` to calculate CPU usage. However if a GPU is used the main thread is yielded so the CPU could be free while the GPU is working.

### Proposal

I've tried several things:
- Idle hook: With 1000 Hz FreeRTOS tick rate it was called 20,000 times in 500 ms. So it's cannot be used as it's not called for each tick but for each yield (I guess).
-

GorgonMeducer, 2024-01-15:
In FreeRTOS, you can use `traceTASK_SWITCHED_OUT` and `traceTASK_SWITCHED_IN` to update performance counters.

kisvegabor, 2024-01-15:
Thank you! Finally I ended up using the task name because TaskNumber wasn't still working well.  See https://github.com/kisvegabor/lvgl_upstream/commit/e2782299f4eaa409fd05d3838e977f8fed5c56c2

This is a super hacky implementation and it uses lv_tick which has millisecond resolution. So it's just an initial test.

I've tried it out on the  Renesas RA6 board. 

When the 3 gauges were displayed LVGL measured 100% CPU usage, however `t_no_idle_sum` and `t_idle_sum` showed that the real CPU usage is only 33%

On the main screen of the widgets demo LVGL reported ~10% CPU usage. From `t_no_idle_sum` and `t_idle_sum` I calculated 4% CPU usage. 

It shows me that this needs to explored and an OS level CPU usage measurement needs to implemented. However seemingly this is beyond my current understanding of FreeRTOS.

GorgonMeducer, 2024-01-15:
My suggestion is that you do not use the idle task. 

Suppose we can get system timestamp via a function called `get_system_ticks()`, and we manage to get the raw cycle count used by a thread, then the CPU usage is:

"Raw Thread Cycle Count" / "Elapsed System Time" * 100%

pseudo-code:

```c

/*!
 * \brief measure the cpu usage for a given code segment and print out the
 *        result in percentage.
 * \param[in] __CNT generate result on every given iterations
 * \param[in] ... an optional code segement, in which we can read the measured
 *                result from __usage__ which is a float value.
 * \details Here is an example, 50% cpu time:
    E.g.
    \code
        while (1) {
            __cpu_time__(100) {
                delay_us(5000);
            }
            delay_us(5000);
        }
    \endcode
 */
#define __cpu_time__(__CNT, ...)                                                \
    static int64_t SAFE_NAME(s_lTimestamp) = 0, SAFE_NAME(s_lTotal) = 0;        \
    static uint32_t s_wLoopCounter = (__CNT);                                   \
    using(float __usage__ = 0, ({                                               \
    if (0 == s_wLoopCounter) {                                                  \
        __usage__ = (float)((double)SAFE_NAME(s_lTotal)                         \
                        / (double)(     get_system_ticks()                      \
                                  -     SAFE_NAME(s_lTimestamp)));              \
        __usage__ *= 100.0f;                                                    \
        SAFE_NAME(s_lTimestamp) = 0;                                            \
        SAFE_NAME(s_lTotal) = 0;                                                \
        if (__PLOOC_VA_NUM_ARGS(__VA_ARGS__) == 0) {                            \
            __perf_counter_printf__("CPU Usage %3.2f%%\r\n", (double)__usage__);                 \
        } else {                                                                \
            __VA_ARGS__                                                         \
        }                                                                       \
    }                                                                           \
    if (0 == SAFE_NAME(s_lTimestamp)) {                                         \
        SAFE_NAME(s_lTimestamp) = get_system_ticks();                           \
        s_wLoopCounter = (__CNT);                                               \
    }                                                                           \
    start_task_cycle_counter();}),                                              \
    ({SAFE_NAME(s_lTotal) += stop_task_cycle_counter();                         \
    s_wLoopCounter--;}))
```

kisvegabor, 2024-01-15:
How can it take into account if the function being measured yields for some time?

GorgonMeducer, 2024-01-15:
> How can it take into account if the function being measured yields for some time?

you can use traceTASK_SWITCHED_OUT and traceTASK_SWITCHED_IN to update performance counters.

kisvegabor, 2024-01-15:
Ah, got it! So the questions is either
1. We want to measure the whole system's CPU usage -> measure the idle task
2. Measure the CPU usage a single task -> use your method

GorgonMeducer, 2024-01-16:
In fact, when you use my method to measure the cpu Usage of idle task, you can easily get the system CPU Usage of the whole system.

kisvegabor, 2024-01-16:
I think main trick is to find a platform independent high resolution timer.  `lv_tick_get()` or `xTaskGetTickCount` are platform independent but the millisecond resolution is on the very limit.

GorgonMeducer, 2024-01-16:
All Cortex-M processors have Systick, which works at the same system frequency. 
Almost All Cortex-A processors provide a Generic Timer.
RISC-V processors provide 64bit timer for most of the case.

We can let OSAL to provide a high-resolution timer that generate > 10KHz ticks.

nicusorcitu, 2024-01-16:
Measuring the CPU idle task is simple, and will return the overall system idle time.

How can you certainly measure the CPU usage from the draw task? Part of the task is handled by CPU (GPU like driver setup) and part is handled by the GPU.
You can not simply use counter method between task "switch in" and "switch out".

nicusorcitu, 2024-01-17:
True, and VgLite really creates an additional task for queue command buffer.

So it seems that the only solution would be to just measure the overall system idle time in order to really say what is the CPU usage for all tasks.

kisvegabor, 2024-01-17:
@GorgonMeducer 
So when the thread is interrupted by an other thread the counter is paused?

GorgonMeducer, 2024-01-17:
> So when the thread is interrupted by an other thread the counter is paused?

YES. To implement this feature, we need the help of `traceTASK_SWITCHED_OUT` and `traceTASK_SWITCHED_IN`. 
Here:

- `traceTASK_SWITCHED_OUT` pauses the old thread's counter
- `traceTASK_SWITCHED_IN` starts the next thread's counter

nicusorcitu, 2024-01-17:
In the SDK code I have use the same traces to measure the IDLE time. I counted only the IDLE task so I could figure out the overall system CPU load.

> when you read the cycle-count in the LVGL thread, you can only get the result of this LVGL thread.
>
It means that each thread need to calculate its own CPU load.
But from lvgl thread we would not have access to the cycle counter for the thread that the GPU driver is creating internally.

I still think the easies way would be to the method 1.
> 1. We want to measure the whole system's CPU usage -> measure the idle task
>

kisvegabor, 2024-01-17:
I also believe that for benchmarking we need method 1).

For this probably we cannot offer a built solution as in case of FreeRTOS 
- it requires modifications in FreeRTOS config
- we needed a higher resolution timer (platform dependent)

I think the safest and simplest is to allow replacing `lv_timer_get_idle` [here](https://github.com/lvgl/lvgl/blob/master/src/others/sysmon/lv_sysmon.c#L155) with a custom function. As it's highly platform dependent we can use a `#define` instead of a callback.

nicusorcitu, 2024-01-17:
I agree with your solution to have an callback for lv_timer_get_idle. And this API shall return the idle time from the last call.

kisvegabor, 2024-01-18:
Sounds good, I'll take care of that!

kisvegabor, 2024-02-02:
I've successfully implemented in on Renesas RA8D1, so I consider the issue solved.

closed 2024-02-02 by kisvegabor

nicusorcitu, 2024-02-02:
Do I have to add any api to work on NXP products?
We use freertos as well.
________________________________
De la: Gabor Kiss-Vamosi ***@***.***>
Trimis: Friday, February 2, 2024 6:26:37 PM
Către: lvgl/lvgl ***@***.***>
Cc: Nicușor Cîțu ***@***.***>; Mention ***@***.***>
Subiect: [EXT] Re: [lvgl/lvgl] Measure FreeRTOS idle time to see how much the GPU off-loads the CPU (Issue #5243)

Caution: This is an external email. Please take care when clicking links or opening attachments. When in doubt, report the message using the 'Report this email' button

Closed #5243<https://github.com/lvgl/lvgl/issues/5243> as completed.

kisvegabor, 2024-02-05:
We cannot integrate it into LVGL because it required some changes in FreeRTOS header files too. So I added a generic solution: 
using [LV_SYSMON_GET_IDLE](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L741) you can set how LVGL gets the current idle time.

nicusorcitu, 2024-02-08:
What shall [LV_SYSMON_GET_IDLE](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L741) do? Return the idle time from the last 300 ms or return the idle time from the last call? What is the functioning behind?

Thanks, Nicu.
@cristian-stoica

nicusorcitu, 2024-02-08:
The idea is how would you call it in LVGL library:
1. If you call it from 10 places during let's say 500ms - then probably you expect to get the same value every time.
2. If you call it from a single places from time to time - then it makes sense to return the last idle time.

kisvegabor, 2024-02-08:
> If you call it from a single places from time to time - then it makes sense to return the last idle time.

That's the case. It's called only from [here](https://github.com/lvgl/lvgl/blob/master/src/others/sysmon/lv_sysmon.c#L184).

nicusorcitu, 2024-02-08:
Perfect, we have already our own implementation.
Thanks!

kisvegabor, 2024-02-08:
Cool!