lvgl/lvgl #5264: An implement porting toWindows and Linux that can be add in porting topic in docs to helps more

Staok, 2024-01-10:
### Introduce the problem

Here is a VsCode as the IDE, built using cmake and supporting both Windows and Linux implementations, which is very useful for quick browsing ui.

https://github.com/Staok/lvgl_port_win-linux_vscode

### Proposal

that can be add in porting topic in docs to helps more.
thanks.

kisvegabor, 2024-01-10:
Hey,

Thanks for sharing! Can it be integrated with https://github.com/lvgl/lv_port_pc_vscode?

Note that from LVGL v9 (will be released next Monday) we have integrated a lot of drivers into the LVGL repo. 
- Drivers: https://github.com/lvgl/lvgl/tree/master/src/dev
- Eclipse port as reference: https://github.com/lvgl/lv_port_pc_eclipse
- CHANGELOG: https://docs.lvgl.io/master/CHANGELOG.html

The documentation could be added to these places:
- Add Windows, and SDL docs: https://docs.lvgl.io/master/integration/driver/index.html
- A few words about the VSCode port: https://docs.lvgl.io/master/integration/ide/pc-simulator.html#select-an-ide
- README of https://github.com/lvgl/lv_port_pc_vscode

What do you think?

Staok, 2024-01-10:
@kisvegabor 
for the "The documentation could be added to these places" that what i`m talking about,thanks
but first give me some time so i can make some polish of my implement, add some en msg .etc

by the way, Drivers in https://github.com/lvgl/lvgl/tree/master/src/dev, so in the future which drivers code we prefer to use between "lv_drivers" and "lvgl", and more, I see there is only linux things in "https://github.com/lvgl/lvgl/tree/master/src/dev", what about windrv and mcu?

kisvegabor, 2024-01-11:
`lvgl` will be preferred. `lv_drivers` won't be updated to v9. 

I thought the widows drivers are already added. :smile: 
I asked it [here](https://github.com/lvgl/lvgl/issues/4926#issuecomment-1886143702).

MouriNaruto, 2024-01-13:
For more information about new Windows driver, please read https://github.com/lvgl/lvgl/issues/5293.

Kenji Mouri

Staok, 2024-01-15:
@MouriNaruto thanks and good job, I will support lvgl v9 and your window drivers code by sometime

closed 2024-02-07 by lvgl-bot