lvgl/lvgl #5273: Why do we have lv_draw_buf_t everywhere

XuNeo, 2024-01-11:
### Introduce the problem

I suppose most people have this concern, and it's raised to a real discussion after https://github.com/lvgl/lvgl/pull/5204 is merged.

Let's discuss here so more can join in.

A background.
Some GPUs have limitation on both target and source image stride and buffer start address. Specifically, for VG-Lite, the start address must be 64 bytes aligned, image width should be 16pixel aligned.

So we introduced `lv_draw_buf_t` similar to other GUI libs, so a raw buffer pointer now always have a header to CLEARLY state the meta info of buffer.

```c
typedef struct {
    lv_image_header_t header;
    uint32_t data_size;     /*Total buf size in bytes*/
    void * data;
    void * unaligned_data;  /*Unaligned address of `data`, used internally by lvgl*/
} lv_draw_buf_t;
```

We have seen the benefits already after migrating to this new interface in image decoder, internal display logic, and layer. It simplifies processes since you cannot use a raw pointer directly, the meta info is already needed, and it's in the draw buffer header.

Now it comes to the porting layer specifically for API
```c

/**
 * Set the buffers for a display
 * @param disp              pointer to a display
 * @param buf1              first buffer
 * @param buf2              second buffer (can be `NULL`)
 */
void lv_display_set_draw_buffers(lv_display_t * disp, lv_draw_buf_t * buf1, lv_draw_buf_t * buf2);

typedef void (*lv_display_flush_cb_t)(lv_display_t * disp, const lv_area_t * area, uint8_t * px_map);

```

For platforms with GPU limitation, it's a must to have draw buffer configured correctly. Thus the API has changes so.

For lvgl v8 backward `compatibility`(This API has changed already, so no real compatibility), `lv_conf.h` has been updated to do non alignment by default. https://github.com/lvgl/lvgl/pull/5272

cc @kisvegabor @FASTSHIFT @PGNetHun

XuNeo, 2024-01-11:
I'm going to revert the change https://github.com/lvgl/lvgl/pull/5204 right now before we have conclusion.

> Besides we are very close to the release and without a clear added value I consider this change a risk.
> 
> If you insist to have an API to set the draw buffers from lv_draw_buf_t we can have 2 functins:
> 
> void lv_display_set_buffers(lv_display_t * disp, void * buf1, void * buf2)
> void lv_display_set_draw_buffers(lv_display_t * disp, lv_draw_buf_t * draw_buf1, lv_draw_buf_t * draw_buf2)
> The naming is not so fortunate, so ideas are welcome.

kisvegabor, 2024-01-11:
My concern is that introducing `lv_draw_buf_t` in `lv_display_set_draw_buffers` will result in initialization like this:
```c
//Possible issue: 
//  - use the same color format as the display
//  - know the resolutions 
//  - know _LV_DRAW_BUF_SIZE
uint32_t buf_size = _LV_DRAW_BUF_SIZE(320, 24, LV_COLOR_FORMAT_RGB565);

void * buf = malloc_for_dma(buf_size);

//Possible issue: it must be static
static lv_draw_buf_t draw_buf;

//Possible issue: 
//  - repeat width, height, color format, buf_size correctly
//  - know what "stride" is, and what 0 means
lv_draw_buf_init(&draw_buf, 320, 24, LV_COLOR_FORMAT_RGB565, 0,  buf, buf_size);

lv_display_set_draw_buffers(display, &draw_buf, NULL);
```

instead of
```c
uint32_t buf_size = 32*1024; //Control the memory usage

void * buf = malloc_for_dma(buf_size);

//Possible issue: pass the same buf_size
lv_display_set_draw_buffers(display, buf, NULL, buf_size);  
```

Besides the complication and possible pitfalls I can't see what problem the use `lv_darw_buf_t` solves here. `lv_draw_buf_t` stores: 
- width and height: which changes on resolution change or in PARTIAL mode on the rendering of each part. In PARTIAL mode it's even misleading to create a buffer with a specific with and a height
- color format: also stored in `lv_display_t` and the color format of `lv_draw_buf_t` is changed when the the display's color format changes
- stride: `lv_draw_buf_width_to_stride` can always tell the stride from width.

Even though I don't agree with this change I don't want to act as knowing it better. I respect your and your team's opinion and expertise, so we should either
- agree that one solution is better then the other
- support both ways via `lv_display_set_draw_buffers` and `lv_display_set_draw_buffers_2` (via a better name)

> I'm going to revert the change https://github.com/lvgl/lvgl/pull/5204 right now before we have conclusion.

Thank you!

kisvegabor, 2024-01-11:
The flush callback with `lv_draw_buf_t` would look like this:
```c
void flush_cb(lv_display_t * display, lv_draw_buf_t * draw_buf, const lv_area_t * area);
```

Here I have these concerns:
1. `draw_buf` contains redundant data (color format from `display`, width and height from `area`)
2. `lv_draw buf_t` is not required for most of the users, and if someone needs it, it can be created from the available data using `lv_draw_buf_init()`
3. We are too close to the release to change it.

As a workaround I suggest  having a helper like `lv_draw_buf_init_from_flush_area(&draw_buf, display, area)` (not a good name).

XuNeo, 2024-01-11:
> We are too close to the release to change it.

Agreed. Let's not break anything from now on, and discuss it further when we actually need a draw buf in flush_cb.

XuNeo, 2024-01-26:
Let's keep this issue open for sometime to see if others have questions.

closed 2024-02-12 by XuNeo