lvgl/lvgl #5277: Bin Font Loading

fatihozen3, 2024-01-11:
### Introduce the problem

I use NXP Imx RT1170 series mcu and I want to use font as a bin format. How could that possible ?

I see there is a source file as lv_binfont_loader.c  but this file is not in the latest tag, it exists in master branch. Why the tag does not include this file ? Is it still in progress ?

I have downloaded offline lvgl font converter and have got bin font file with script. After that I have written this bin file into flash's spesific address and used below functions but I could not see anything in screen. I am open any advice.

lv_font_t my_font;
lv_binfont_load(&my_font,"F:/font.bin");
	static lv_style_t font_style;
	lv_style_init(&font_style);
	lv_style_set_text_font(&font_style, &my_font);

	lv_obj_t *label_zh = lv_label_create(ui->screen);
	lv_obj_align(label_zh, LV_ALIGN_BOTTOM_MID, 0, 0);
	lv_obj_add_style(label_zh, &font_style, 0);
	lv_label_set_text(label_zh, "china font");

kisvegabor, 2024-01-14:
Hi,

We will release LVGL v9 tomorrow (January 15) so I suggest v9 from the `master` branch.

All related tools are updated to work with LVGL v9. 

```c
lv_binfont_load(&my_font,"F:/font.bin");
```

If you have loaded the font to the flash, you need to enable [`MEMFS`](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L643-L646).

Please enable [Logging](https://docs.lvgl.io/master/porting/log.html) to see what's happening.

fatiihyildizz, 2024-01-18:
@kisvegabor Hello, Is there a way to use the font file as .bin without using lv_binfont_load?

kisvegabor, 2024-01-18:
Unfortunately no. LVGL can't read the binary font directly.

fatiihyildizz, 2024-01-18:
This is sad. So is there a new date for LVGL v9? @kisvegabor

kisvegabor, 2024-01-18:
> This is sad.

Note that it's possible, just not supported now. If you  are interested in working on that, please open a new issue to discuss the details. 

> So is there a new date for LVGL v9? 

Yes, January 22 (Monday).

hyansuper, 2024-01-18:
If font/img bin file is in flash, no compression, it seems unnecessary to read it into RAM, we just need memory mapping, make the flash area addressable.

But we will need the bin file in  similar format as builtin c array, this can be very useful for small RAM devices

fatiihyildizz, 2024-01-18:
Yes, we created the rawfs structure for flash operations. But from what I see, the lv_obj_set_style_text_font function seems not suitable for this structure. When I give the .bin Font address in Flash to this function as (lv_font_t *), it reads incorrectly. @hyansuper

hyansuper, 2024-01-18:
I tried to use bin address directly with lv_img_set_src, result in crash, then I realized the bin file generated by the convert script is in different format than c array

fatiihyildizz, 2024-01-18:
Yes, although it is not completely wrong, some values are read incorrectly and a crash occurs. So did you try something different to read the .bin file or did you prefer to read the font file as .c? @hyansuper

hyansuper, 2024-01-18:
Now I have to use .c
But I am thinking about making a c-array-like bin file, that can be downloaded to flash, then I construct the lv_font_t/lv_img_dsc_t struct and point it to the address, then I can change UI on the fly!
But that's a bit over engineering and out of my power😅

fatiihyildizz, 2024-01-18:
It's a method that requires a little more effort. 😅 Good luck!  @hyansuper

kisvegabor, 2024-01-18:
I think the easiest would be reuse the `binfont_loader` but for the large and simple array, instead of malloc-ing use a simple pointer. See
- [bitmap](https://github.com/lvgl/lvgl/blob/master/src/font/lv_binfont_loader.c#L396)
- [kern tables](https://github.com/lvgl/lvgl/blob/master/src/font/lv_binfont_loader.c#L637-L639)

hyansuper, 2024-01-20:
Thank you @kisvegabor , modify the `malloc`s in `binfont_loader` is the way to go.
currently I am working on image bin files, which is much easier.
but I see there's no data_size output https://github.com/lvgl/lvgl/blob/master/scripts/LVGLImage.py#L368
how do I infer lv_img_dsc_t.data_size?

```c
	lv_img_dsc_t img_dsc;
	memcpy(img_dsc, mmap_addr, sizeof(lv_image_header_t));
	img_dsc->data = mmap_addr + sizeof(lv_image_header_t);
        img_dsc->data_size = ???

	lv_img_set_src(img_obj, &img_dsc);

```

hyansuper, 2024-01-20:
> but I see there's no data_size output https://github.com/lvgl/lvgl/blob/master/scripts/LVGLImage.py#L368
how do I infer lv_img_dsc_t.data_size?

I simply changed how `LVGLImage.py` outputs bin format:
```python

    def _compress(self, raw_data: bytes) -> bytearray:
        if self.compress == CompressMethod.NONE:
            bin = bytearray()
            bin += uint32_t(len(raw_data)) # data_size
            bin += uint32_t(20) # data: relative offset
            bin += raw_data
            return bin
        ...
```
C code:
```
        lv_img_dsc_t img_dsc;
	memcpy(img_dsc, mmap_addr, sizeof(lv_img_dsc_t ));
	img_dsc->data += (unsigned int)mmap_addr; // add absolute address to relative offset

	lv_img_set_src(img_obj, &img_dsc);
```

As for bin font, I think the`binfont_loader`  is too complex, I will probably take reference of c array convert script https://github.com/lvgl/lv_font_conv/tree/master/lib/writers/lvgl 
and see how to generate similar bin file. 
general idea is to let the c code read memory mapped bin file as easily as possible, only relative offset need to be changed to absolute.

kisvegabor, 2024-01-22:
> but I see there's no data_size output https://github.com/lvgl/lvgl/blob/master/scripts/LVGLImage.py#L368
how do I infer lv_img_dsc_t.data_size?

cc @XuNeo 

> So I won't reuse binfont_loader ,I will probably take reference of c array convert script

Note that a font is more than just the bitmaps:
- kerning
- fast searching
- bounding box vs advance width
- base line offset

XuNeo, 2024-01-22:
> I simply changed how `LVGLImage.py` outputs bin format:

For .bin image, you can calculate the data size directly from `header->stride * header->h`, double check it with file size to make sure data is not corrupted.

hyansuper, 2024-01-23:
@XuNeo  I don't know how builtin bin image decoder works, does it decode bin file into a `malloc`'ed area in RAM? which is true for binfont_decoder, that takes up too much space in RAM.
If bin image/font resource is in flash in a format that resembles compiled c array, then I can directly read it without using RAM. 

Dont' buther, I solved it just by adding data_size in the bin output.
currently you need to treat bin resource as a file, so I guess you can find data_size using file length.

XuNeo, 2024-01-23:
So you can generate the  image in C-array format directly right? No need to resemble it afterward.

hyansuper, 2024-01-23:
@fatihozen3 
[This fork of lv_font_conv](https://github.com/hyansuper/lv_font_conv) seems to work, at least on my ESP32 chip ( little endian, 4 bytes aligned）
You put `--format cbin` in the [convert command](https://github.com/hyansuper/lv_font_conv?tab=readme-ov-file#cli-params), put the bin file into flash. there's an [example](https://github.com/hyansuper/lv_font_conv/blob/master/cbin_example) of how to use it.

closed 2024-02-15 by lvgl-bot

hyansuper, 2025-08-30:
Great!

For fonts, you still need to malloc many structs and change their relative offsets to absolute address.

I just came up with a better idea: given the mapped address of asset partition, we can modify relative offsets to absolute address once and for all when you do OTA. An example is the esp_encrypted_img component, it has a callback that decodes bin file before its written to flash

78, 2025-08-30:
> Great!
>
> For fonts, you still need to malloc many structs and change their relative offsets to absolute address.
>
> I just came up with a better idea: given the mapped address of asset partition, we [...]

You idea works like a DLL (dynamic-link library) loader.

I think the current implementation is good enough for me to use in the project [xiaozhi-esp32](https://github.com/78/xiaozhi-esp32). Here is my test result:

Font glyphs: 6650
Font cmaps: 27
Font load time: < 1ms
RAM used: < 1KB

It's very fast and consumes only hundreds of bytes RAM, compared to the bin loader used in the current LVGL, which has a 1700ms load time and uses almost 2MB of RAM.