lvgl/lvgl #5279: Unify font create APIs

kisvegabor, 2024-01-11:
### Introduce the problem

From https://github.com/lvgl/lvgl/pull/5196#issuecomment-1878663717
> TinyTTF and lv_binfont_load I have changed the create API like this:
```c
lv_result_t lv_tiny_ttf_create_file(lv_font_t * font, const char * path, int32_t font_size);
```
> This way if a font is used in many styles, it can be easily reinitialized with different size or other settings without updating the LVGL styles. If a new font is allocated in the create function all styles needs to be updated manually. Therefore I recommend updating the API of FreeType like this as well.

(We should do this still before v9.)

### Proposal

Based on https://github.com/lvgl/lvgl/pull/5196#issuecomment-1879586307
```c
lv_freetype_init_font_from_file(lv_font_t * font, ...);

//Helper
lv_font_t * lv_freetype_create_font_from_file(...)
{
  lv_font_t * f = lv_malloc(sizeof(lv_font_t));
  lv_freetype_init_font_from_file(f, ...);
  return f;
}

void lv_freetype_delete_font(lv_font_t * f)
{
   //freetype specific destroy
   lv_free(f)
}
```

cc @W-Mai @FASTSHIFT

FASTSHIFT, 2024-01-14:
I prefer to keep only the lv_font_t * lv_freetype_font_create() API, because this way the font structure and the data structure it describes have the same life cycle, and the font deletion operation is relatively clear.
you mentioned:
```c
static lv_font_t font;
lv_freetype_init_font_from_file(&font, "XXX");
```
This way of writing is more flexible. Can you give an example?

kisvegabor, 2024-01-14:
From [here](https://github.com/lvgl/lvgl/pull/5196#issuecomment-1878663717):
> This way if a font is used in many styles, it can be easily reinitialized with different size or other settings without updating the LVGL styles. If a new font is allocated in the create function [...]

`font_init(&font)` should be superior, as you can do the same with it as what `lv_font_t * font_create()` does. I understand that you value the simplicity of `font_create`, but I think for flexibility `font_init` needs to be added too.

FASTSHIFT, 2024-01-15:
OK, Is it enough to revert this submission?
https://github.com/lvgl/lvgl/commit/90a7e3e

FASTSHIFT, 2024-01-15:
Fixed: https://github.com/lvgl/lvgl/pull/5327
After looking at the example and test, there should be no need to modify it.

kisvegabor, 2024-01-15:
Let's continue in the PRs.

closed 2024-01-15 by kisvegabor