lvgl/lvgl #5293: The new rewritten Windows backend for LVGL v9 Series

MouriNaruto, 2024-01-12:
### Introduce the problem

Hi, I'm Kenji Mouri, the maintainer of LVGL Windows backend.

At least in https://github.com/lvgl/lv_port_pc_visual_studio/issues/53, I started to do that.

Finally, finally, I have nearly finished this. I'm writing the comments for new APIs which will be introduced for this.

I will push them to mainline as soon as possible. So, I'm working hard on that.

### Proposal

For making people who care about that don't wait for so long. I will show the API definitions I finished comment writing.

### Context APIs for Windows backend

Private APIs for LVGL Windows backend implementation. LVGL users shouldn't use that because the maintainer doesn't promise the application binary interface compatibility for this API.

https://github.com/lvgl/lv_port_pc_visual_studio/blob/24cc50e83b089f119c081835ad3278fdafec4b7b/LvglWindowsSimulator/lv_windows_context.h
https://github.com/lvgl/lv_port_pc_visual_studio/blob/24cc50e83b089f119c081835ad3278fdafec4b7b/LvglWindowsSimulator/lv_windows_context.c

### Interop APIs for Windows backend

Help LVGL users to interop LVGL Windows backend with standard Windows API and other Windows applications.

https://github.com/lvgl/lv_port_pc_visual_studio/blob/24cc50e83b089f119c081835ad3278fdafec4b7b/LvglWindowsSimulator/lv_windows_interop.h
https://github.com/lvgl/lv_port_pc_visual_studio/blob/24cc50e83b089f119c081835ad3278fdafec4b7b/LvglWindowsSimulator/lv_windows_interop.c

### LVGL indev APIs for Windows backend

Working on writing comments for API definitions.

### LVGL display APIs for Windows backend

Working on writing comments for API definitions.

MouriNaruto, 2024-01-12:
@kisvegabor 

Here is the commented definition of lv_windows_create_display.

```
    /**
     * @brief Create a LVGL display object.
     * @param title The window title of LVGL display.
     * @param hor_res The horizontal resolution value of LVGL display.
     * @param ver_res The vertical resolution value of LVGL display.
     * @param zoom_level The zoom level value. Base value is 100 a.k.a 100%.
     * @param allow_dpi_override Allow DPI override if true, or follow the
     *                           Windows DPI scaling setting dynamically.
     * @param simulator_mode Create simulator mode display if true, or create
     *                       application mode display.
     * @return The created LVGL display object.
    */
    lv_display_t* lv_windows_create_display(
        const wchar_t* title,
        int32_t hor_res,
        int32_t ver_res,
        int32_t zoom_level,
        bool allow_dpi_override,
        bool simulator_mode);
```

Due to the title, zoom level, dpi mode and simulator mode all need to determined at the display creation. So, I have not use the dedicated functions. Maybe in the future revisions of Windows backend. Dynamic changing support for these will be introduced. (We don't need to have a title changing function because Windows has the API for that. It's time for interop API like lv_windows_get_display_window_handle.)

Here are the similarity for simulator mode and application mode.

- Support pointer, keypad and encoder devices in the HAL level.
- Support Windows touch input in the HAL level.
- Support Windows IME input in the HAL level.
- Support Per-monitor DPI Aware (both V1 and V2) in the HAL level.

Here are the differences for simulator mode and application mode.

#### Simulator Mode

- Designed for LVGL simulation scenario.
- Keep the resolution all time for trying best to simulate UI layout which will see in their production devices.
- When Windows DPI scaling setting is changed, Windows backend will stretch the display content.

#### Application Mode

- Designed for Windows desktop application development scenario.
- Have the Window resizing support.
- When Windows DPI scaling setting is changed, the LVGL display DPI value will also be changed.

Kenji Mouri

kisvegabor, 2024-01-12:
All clear and approved by me! :blush:

PGNetHun, 2024-01-12:
Thank you for Windows integration!

Could you please add also documentation later in a separate PRwhen you have time?
Just put all the info you mentioned here + example code, so other devs will know how to start their project with Windows “port”.

I think the right place would be under “Integration” section:
https://docs.lvgl.io/master/integration/index.html

MouriNaruto, 2024-01-12:
> Thank you for Windows integration!
>
> Could you please add also documentation later in a separate PR, when you have time? Just put all the info you mentioned here + example code, so other devs will [...]

OK. I will try to write some documents for that.

Kenji Mouri

MouriNaruto, 2024-01-13:
@PGNetHun 

The document is ready.

https://github.com/MouriNaruto/lvgl/blob/windows_backend_v9/docs/integration/driver/windows.rst

Kenji Mouri

kisvegabor, 2024-01-14:
#5313 is merged, I close this issue. 

Thank you @MouriNaruto! :rocket:

closed 2024-01-14 by kisvegabor