lvgl/lvgl #5296: Calendar can't show highlighted_dates and today_date

IAMMX, 2024-01-12:
### LVGL version

v9.0.0

### What happened?

[image]
[image]

kisvegabor, 2024-01-12:
I'll check it at latest on Monday.

C47D, 2024-01-13:
I went back to 00825c1ff to check if the latest change of calendar caused the issue and I see the same issue
[image]

I will try to take a look at it, I'm not very familiar with the calendar widget.

closed 2024-01-14 by FASTSHIFT

IAMMX, 2024-01-14:
Now it is correct ,thanks