lvgl/lvgl #5302: Why is lv_obj called lv_obj?

ggoraa, 2024-01-12:
### Introduce the problem

Knowing that in the documentation all objects are usually called "widgets" (even the top-level group for all objects is called Widgets)

kisvegabor, 2024-01-14:
I agree with you and I've proposed to rename it to `lv_widget` for v9 but as feedback basically all people said that the large API change would cause bigger trouble than it solves :smile:

ggoraa, 2024-01-15:
That's _is_ quite interesting 🧐 
In that case, can I name the widget class `lvgl::Widget` in the C++ binding, as a kind of "start" to the name migration? 😅

XuNeo, 2024-01-15:
> And also, isn't it possible to rename `lv_obj` to `lv_widget`, place a `typedef lv_widget lv_obj` for backwards compatibility, mark `lv_obj` deprecated, and then nuke it a major version after?)

The main problem is all those APIs/types/variables started with `lv_obj_` prefix. Those are massive.

ggoraa, 2024-01-15:
Oh, yes, right, I forgot about those...
That would definitely make the refactor much larger and more impactful

kisvegabor, 2024-01-15:
Later, we can investigate how can we make a gradual translation without breaking.

ufbycd, 2024-01-21:
> That's _is_ quite interesting 🧐 In that case, can I name the widget class `lvgl::Widget` in the C++ binding, as a kind of kickstart to the name migration? 😅
> 
> And also, isn't it possible to rename `lv_obj` to `lv_widget`, place a `typedef lv_widget lv_obj` for backwards compatibility, mark `lv_obj` deprecated, and then nuke it a major version after?)

We can define some macros for backward compatibility. For example:
```C
#define lv_obj_t lv_widget_t
#define lv_obj_set_x lv_widget_set_x
```

closed 2024-02-14 by lvgl-bot