lvgl/lvgl #5303: The amount of memory for the drawing buffer has increased.

ngc6589, 2024-01-12:
### LVGL version

v9-dev

### What happened?

The amount of memory for the drawing buffer has increased.

Previous LVGL drawing buffer
```
static lv_color_t buf1[240 * 320 / 10];
static lv_color_t buf2[240 * 320 / 10];
```
```
  .bss.buf1 0x2000cba8 0x5a00
  .bss.buf2 0x200125a8 0x5a00
```
Total 45k

```
static lv_draw_buf_t buf1[240 * 320 / 10];
static lv_draw_buf_t buf2[240 * 320 / 10];
```

```
.bss._ZL4buf1 0x2000d934 0x2d000
.bss._ZL4buf2 0x2003a934 0x2d000
```
Total 360k

The 264k RAM of Raspberry Pi Pico (RP2040) is no longer enough.
I hope it can be avoided in the future.

XuNeo, 2024-01-13:
Could you show the code so I can see where the misleading comes from?

The issue is that you should not change the type of the buffer. Should keep using lv_color32_t.

ngc6589, 2024-01-13:
I fixed this after the argument of lv_display_set_draw_buffers in src/display/display.c was changed a few days ago. The test code is below.

main.cpp
```
#include <stdio.h>
#include "pico/stdlib.h"
#include "hardware/spi.h"
#include "hardware/dma.h"
#include "hardware/pio.h"
#include "ili9341.h"
#include "xpt2046.h"

static lv_draw_buf_t buf1[240 * 320 / 10];
static lv_draw_buf_t buf2[240 * 320 / 10];
static lv_obj_t *screen;

void lv_example_button_1();
static void event_handler(lv_event_t * e);

int main()
{
    stdio_init_all();

    ili9341_Init(LCD_INV_PORTRAIT);
    xpt2046_Init(TP_INV_PORTRAIT);

    lv_init();

    lv_display_t *disp = lv_display_create(240, 320);
    lv_display_set_flush_cb(disp, lcd_Flash_CB);
    lv_display_set_draw_buffers(disp, buf1, buf2);
    lv_display_set_render_mode(disp, LV_DISPLAY_RENDER_MODE_PARTIAL);
    screen = lv_obj_create(NULL);

    lv_indev_t *indev = lv_indev_create();
    lv_indev_set_type(indev, LV_INDEV_TYPE_POINTER);
    lv_indev_set_read_cb(indev, xpt2046_read_cb);

    lv_example_button_1();

    while (1)
    {
        sleep_ms(5);
        lv_timer_handler();
        lv_tick_inc(5);
    }
}

static void event_handler(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);

    if(code == LV_EVENT_CLICKED) {
        LV_LOG_USER("Clicked");
    }
}

void lv_example_button_1(void)
{
    lv_obj_t * label;

    lv_obj_t * btn1 = lv_button_create(screen);
    lv_obj_add_event_cb(btn1, event_handler, LV_EVENT_ALL, NULL);
    lv_obj_align(btn1, LV_ALIGN_CENTER, 0, -40);
    lv_obj_remove_flag(btn1, LV_OBJ_FLAG_PRESS_LOCK);

    label = lv_label_create(btn1);
    lv_label_set_text(label, "Button");
    lv_obj_center(label);
}
```

The modifications to lv_conf.h are as follows.
```
#define LV_USE_DEV_VERSION
#define LV_COLOR_DEPTH 16
#define LV_MEM_SIZE (48* 1024U)    -> 256 to 48  

```

ngc6589, 2024-01-13:
Document Porting Page is
lv_disp_set_draw_buffers()

However, after changing to lv_display_set_buffers() it started working.
```
static uint32_t buf1[240 * 320 / 10];
static uint32_t buf2[240 * 320 / 10];
:
:
    lv_display_t *disp = lv_display_create(240, 320);
    lv_display_set_flush_cb(disp, lcd_Flash_CB);
    lv_display_set_buffers(disp, buf1, buf2, sizeof(buf1), LV_DISPLAY_RENDER_MODE_PARTIAL);
    //lv_display_set_render_mode(disp, LV_DISPLAY_RENDER_MODE_PARTIAL);
    screen = lv_obj_create(NULL);
```

XuNeo, 2024-01-13:
Thank you for feedback. I have updated the doc. :)

PGNetHun, 2024-01-13:
@ngc6589 

Please set also the color format. 
`LV_COLOR_DEPTH` will not be used in the future, instead the display stores the color format, and the pixel size is calculated from it. The documentation is being update continously to match to code.

In previous LVGL versions we gave an array of pixels to the display to use as buffer (in your original code an array of `lv_color_t`, then `uint32_t`), but since a recent PR we need to pass a **struct** `lv_draw_buf_t`, which contains width, height, color format, stride AND a pointer to the pixel buffer).

This `lv_draw_buf_t` struct can be created with the function `lv_draw_buf_create(uint32_t w, uint32_t h, lv_color_format_t cf, uint32_t stride)`, which allocates the pixel buffer also, and if the device is not restricted to use a specific memory area to store screen buffer (pixel array), then we can use this `lv_draw_buf_create()` function. 

For the special case when we are restricted to use (or for better performance want to use) specific memory area for screen buffer, there is also a second function: `lv_display_set_buffers()`, in this case we have to allocate the buffer in that restricted memory area, and pass to the function.
For example ESP32 has DMA-capable memory area, and for better performance we use that to store (partial) screen data so SPI controller can transfer the pixels to the display using DMA.

In your case using `lv_draw_buf_create()` would look like:

```
static lv_draw_buf_t *draw_buf1;
static lv_draw_buf_t *draw_buf2;

int main()
{
  ....
  lv_init();
  lv_display_t *disp = lv_display_create(240, 320);
  lv_display_set_flush_cb(disp, lcd_Flash_CB);
 
  draw_buf1 = lv_draw_buf_create(240, 32, LV_COLOR_FORMAT_ARGB8888, LV_STRIDE_AUTO);
  draw_buf2 = lv_draw_buf_create(240, 32, LV_COLOR_FORMAT_ARGB8888, LV_STRIDE_AUTO);
  lv_display_set_draw_buffers(display, draw_buf1, draw_buf2);
  ...
}
```

ngc6589, 2024-01-13:
@PGNetHun

I understood the latest coding examples.
It worked in my environment too.

thank you.

PGNetHun, 2024-01-13:
@ngc6589  or @XuNeo ,
could you please close this issue as resolved?

XuNeo, 2024-01-13:
It will be closed automatically when the linked PR is merged.

PR #5304 (2024-01-14): fix(doc): recommend to use lv_display_set_buffers instead

Help us review this PR! Anyone can [approve it or request changes](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/reviewing-changes-in-pull-requests/approving-a-pull-request-with-required-reviews).

### Description of the feature or fix

Fix doc inconsistent porting guide of `lv_disp_set_buffers` usage.

Fix #5303

### Checkpoints
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests] [... cut]
closed 2024-01-14 by kisvegabor