lvgl/lvgl #5305: LVGL 9: DRAM segment data does not fit when compiling against M5Stack Core2 (ESP32)

shiny-mira, 2024-01-13:
### LVGL version

9.0.0-dev+sha.ee8de1e

### What happened?

When trying to compile a LVGL 9.0 project that is pretty much empty for the M5Stack Core2, the following link error occurs:

```
Processing m5stack-core2 (platform: espressif32; board: m5stack-core2; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/m5stack-core2.html
PLATFORM: Espressif 32 (6.5.0) > M5Stack Core2
HARDWARE: ESP32 240MHz, 4.31MB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20014.231204 (2.0.14)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- M5Unified @ 0.1.12
|-- lvgl @ 9.0.0-dev+sha.ee8de1e
Building in release mode
Linking .pio/build/m5stack-core2/firmware.elf
/Users/m/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/m5stack-core2/firmware.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
/Users/m/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/Users/m/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: DRAM segment data does not fit.
/Users/m/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `dram0_0_seg' overflowed by 163088 bytes
collect2: error: ld returned 1 exit status
*** [.pio/build/m5stack-core2/firmware.elf] Error 1
```

With 8.3, you could compile/run a project without any problems.

Full (platformio) based source-code at https://github.com/shiny-mira/lvgl-base-platformio, snippet below.

### How to reproduce?

```
#include <Arduino.h>
#include <M5Unified.h>

#define LV_CONF_INCLUDE_SIMPLE
#include <lvgl.h>

#define LV_HOR_RES_MAX 320
#define LV_VER_RES_MAX 240

void setup(){
  M5.begin();
  lv_init();

  static lv_display_t *display = lv_display_create(LV_HOR_RES_MAX, LV_VER_RES_MAX);
  static lv_indev_t *indev = lv_indev_create();
}

void loop() {

  lv_task_handler();
  vTaskDelay(1);
}
```

and-elf, 2024-01-14:
Is this related to #5304 ?

shiny-mira, 2024-01-14:
It does not seem to be - just trying to link the library seems to be enough. In the example given above, there is no call to `lv_disp_set_buffers` or `lv_disp_set_draw_buffers`; there also is no allocation for a draw buffer.

and-elf, 2024-01-14:
Maybe lv_display_create needs to be patched to work with v9? How is it implemented?

shiny-mira, 2024-01-14:
`lv_display_create` is part of the LVGL v9 api: https://github.com/lvgl/lvgl/blob/ee8de1e1a0d7da36cbfd840c299abfe3ac027719/src/display/lv_display.h#L94

Also - the problem is during the linking step, not execution/compilation. It seems to be due to static memory use of the library itself.

kisvegabor, 2024-01-14:
It should be fixed by #5308. Please confirm it.

shiny-mira, 2024-01-14:
This indeed does fix the issue - thank you very much.

kisvegabor, 2024-01-14:
Great, thank you for the feedback!

PR #5308 (2024-01-14): fix(conf): lower the default RAM usage from 256k to 64k
Help us review this PR! Anyone can [approve it or request changes](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/reviewing-changes-in-pull-requests/approving-a-pull-request-with-required-reviews).

### Description of the feature or fix

fixes #5305

### Checkpoints
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applica [... cut]
closed 2024-01-14 by FASTSHIFT