lvgl/lvgl #5307: Build error while enabling file system

caichao, 2024-01-14:
### Introduce the problem

I am trying to enable FatFs file system while  using LVGL libs.

I clone the official repository here: https://blog.lvgl.io/2018-01-03/linux_fb#other-notes

I can make the right output. When I enable the file system by enabling the following defines

[image]

[image]

I still cannot make it right. The follow errors occur:

[image]

I just search over the source file and then enable WIN32 in the Makefile, try building again:
[image]

New errors occur:
[image]

Can anybody tell me how to fix these problems?
Thanks in advance.

caichao, 2024-01-14:
By the way, I can build bin file if do not activate file system and see the demo under build/bin folder.

kisvegabor, 2024-01-14:
`DT_DIR` in not defined in C)) but a GCC extension. See [here](https://stackoverflow.com/questions/46694350/cant-compare-dirent-d-type-to-dt-dir).

Can you try `LV_USE_FS_POSIX`?

caichao, 2024-01-15:
Not ok. post the following error:
[image]

Thanks for your response

caichao, 2024-01-15:
I feel confused. When I first change to LV_USE_FS_POSIX, I get the aforementioned error. It seems that ffmpeg is not installed. However, when I make clean and make again. The same problem happens again. 

[image]

caichao, 2024-01-16:
> `DT_DIR` in not defined in C99 but a GCC extension. See [here](https://stackoverflow.com/questions/46694350/cant-compare-dirent-d-type-to-dt-dir).
> 
> Can you try `LV_USE_FS_POSIX`?

Problem solved when add the macro "_DEFAULT_SOURCE"

caichao, 2024-01-16:
Thanks for the help

kisvegabor, 2024-01-16:
Great!

closed 2024-01-16 by kisvegabor