lvgl/lvgl #5329: How to make full use GPU command list

XuNeo, 2024-01-15:
### Introduce the problem

Let's discuss the command list optimization here.

Background:
GPU supports command list to store all drawing commands to a list and flush at once for better performance.
Now with the new draw unit implementation, how could we make better use of this feature.

We can take VG-Lite as an example.

@FASTSHIFT Please join the discussion here since you have better knowledge on it than me.

### Proposal

Code examples, may not work:
```c
static int32_t draw_dispatch(lv_draw_unit_t * draw_unit, lv_layer_t * layer)
{
    lv_draw_vg_lite_unit_t * draw_vg_lite_unit = (lv_draw_vg_lite_unit_t *)draw_unit;

    /* Return immediately if it's busy with draw task. */
    if(draw_vg_lite_unit->task_act) {
        return 0;
    }

    while(1) {
        /* Try to get an ready to draw. */
        lv_draw_task_t * t = lv_draw_get_next_available_task(layer, NULL, VG_LITE_DRAW_UNIT_ID);

        /* Return 0 is no selection, some tasks can be supported by other units. */
        if(!t || t->preferred_draw_unit_id != VG_LITE_DRAW_UNIT_ID) {
            return -1;
        }

        void * buf = lv_draw_layer_alloc_buf(layer);
        if(!buf) {
            return -1;
        }

        /* Return if target buffer format is not supported. */
        if(!lv_vg_lite_is_dest_cf_supported(layer->draw_buf->header.cf)) {
            return -1;
        }

        t->state = LV_DRAW_TASK_STATE_IN_PROGRESS;
        draw_vg_lite_unit->base_unit.target_layer = layer;
        draw_vg_lite_unit->base_unit.clip_area = &t->clip_area;
        draw_vg_lite_unit->task_act = t;

        draw_execute(draw_vg_lite_unit);

        draw_vg_lite_unit->task_act->state = LV_DRAW_TASK_STATE_READY;
        draw_vg_lite_unit->task_act = NULL;
    }

    /* The draw unit is free now. Request a new dispatching as it can get a new task. */
    lv_draw_dispatch_request();

    return 1;
}

kisvegabor, 2024-01-15:
```c
        /* Return 0 is no selection, some tasks can be supported by other units. */
        if(!t || t->preferred_draw_unit_id != VG_LITE_DRAW_UNIT_ID) {
            return -1;
        }
```
Can replaced by `if(!t) {...}`. `VG_LITE_DRAW_UNIT_ID` will always match if you have requested for it.

```c
lv_draw_layer_alloc_buf
```
Can be move outside of the loop.

What does `draw_execute` exactly does? Adds the draw_task to the command list or also wait for completions? 

cc @nicusorcitu, @jeremy-baker, @GorgonMeducer

nicusorcitu, 2024-01-15:
Probably the code has been taken from the NXP vglite dispatcher, I can answer to few questions:
1. 
> Can replaced by if(!t) {...}. VG_LITE_DRAW_UNIT_ID will always match if you have requested for it.
>
This is not true, the `lv_draw_get_next_available_task(layer, NULL, DRAW_UNIT_ID_VGLITE)` will return a task with UNIT_ID_ANY or UNIT_ID_VGLITE.

Specifically to this topic, in NXP vglite code I add the `LV_USE_VGLITE_DRAW_ASYNC` who will read the GPU idle state and if busy, will queue one more task to the list. If idle will submit the task list.

kisvegabor, 2024-01-15:
You are right, however `UNIT_ID_ANY` is not used now. We have no concept right now to make draw task to be taken by any units.

nicusorcitu, 2024-01-15:
Some of the tasks are being passed to the dispatcher without having the evaluation call. Hence it will end up without no CPU or GPU UNIT_ID, will end up with default 0. I do not recall which task was that, I have to test it again.

XuNeo, 2024-01-15:
I have questions about resource management.
For example, if an image is going to be rendered, the image is decoded and GPU command is assembled around it. 
If it's not going to be drawn immediately, when should we closed the image decoder, to free the image resources?

W-Mai, 2024-01-15:
> I'm not sure what the current solution for this is, but in my previous cache implementation I added a counter to see if an image is still used (maybe by multiple draw units simultaneously ) so it can't be removed. cc @W-Mai
@kisvegabor 
Don't worry, this mechanism is still well preserved. Each entry of cache will have a reference counting mechanism. 

1. Every time you acquire, you will get a result of a reference count +1. 
2. When you click the release, the reference count will be reduced by 1. 
3. If the resource is dropped, and the current reference count is not 0, then the entry will be marked as being dropped. 
4. At this time, if the resource is released again, it will be completely released. 
5. Of course, if the resource has not been dropped before, it will be considered to be kicked out when the cache is full.

FASTSHIFT, 2024-01-24:
I have solved the above problem (in PR: https://github.com/lvgl/lvgl/pull/5398) and have verified it on the hardware.
It can effectively use the cmd buffer that comes with the gpu driver to improve rendering efficiency, and there is no need to enable a separate rendering thread.

The following picture is the trace information of a launcher UI (32bpp 466x466 resolution, full screen black fill + 12x image + 2x arc) scene. The total refresh time of the UI is only about 9.5ms:
[image]

Regarding the solution to cache release, I designed a list to save the decoder to be closed. After submitting the image to the GPU, push the decoder dsc information into the list. After waiting for the GPU finish, traverse the list to close the decoder uniformly and clear the list. See the detailed code:  https://github.com/lvgl/lvgl/blob/master/src/draw/vg_lite/lv_vg_lite_utils.c#L1076

kisvegabor, 2024-01-24:
Amazing!
What is the rendering time (now 9.5 ms) if you send only 1 command at a time and wait for competition? 

Without a thread for VG-Lite can the CPU do something else while VG-Lite is working?

nicusorcitu, 2024-01-24:
VG-Lite driver already have an internal thread for executing the command buffer. Hence, I am also curious what is the performance difference with a VGLite draw thread versus without. In NXP code I have all the time a new VGLite draw thread.

FASTSHIFT, 2024-01-25:
> What is the rendering time (now 9.5 ms) if you send only 1 command at a time and wait for competition?

This results in roughly a doubling of performance. Because of the need to start and stop the GPU frequently and wait for interrupt signals, the process becomes longer.

> Without a thread for VG-Lite can the CPU do something else while VG-Lite is working?

Yes, after calling vg_lite_flush, the GPU will start reading the cmd list in the background to start rendering, and will send an interrupt signal to notify the CPU that the rendering is complete. While the GPU is rendering, the CPU can continue to populate the drawing instructions on another cmd list.

FASTSHIFT, 2024-01-25:
> VG-Lite driver already have an internal thread for executing the command buffer. Hence, I am also curious what is the performance difference with a VGLite draw thread versus without. In NXP code I have all the time a new VGLite draw thread.

With a single core, a single thread eliminates the overhead of context switching and is generally more efficient.

nicusorcitu, 2024-01-26:
You can always have that if you set LV_USE_OS to none.
https://github.com/lvgl/lvgl/blob/590363f7429cd0e128f8c030f22006188bac7981/src/draw/nxp/vglite/lv_draw_vglite.c#L113

Just to be sure, I am trying to put my comments in hope that it will be a decision taken to unify the existing two vglite implementation.
And we want to make sure that the performance will be the best.

nicusorcitu, 2024-02-07:
@cristian-stoica

closed 2024-03-03 by lvgl-bot