lvgl/lvgl #5337: NXP fixes for lvgl v9

nicusorcitu, 2024-01-15:
### LVGL version

v9

### What happened?

NXP code updates

### How to reproduce?

[0001-fix-nxp-Make-use-of-lv_draw_buf_t.patch](https://github.com/lvgl/lvgl/files/13940302/0001-fix-nxp-Make-use-of-lv_draw_buf_t.patch)
[0002-fix-nxp-Fixed-wrongly-assert-message.patch](https://github.com/lvgl/lvgl/files/13940304/0002-fix-nxp-Fixed-wrongly-assert-message.patch)

nicusorcitu, 2024-01-15:
[0001-fix-nxp-Make-use-of-lv_draw_buf_t.patch](https://github.com/lvgl/lvgl/files/13940311/0001-fix-nxp-Make-use-of-lv_draw_buf_t.patch)
[0002-fix-nxp-Fixed-wrongly-assert-message.patch](https://github.com/lvgl/lvgl/files/13940313/0002-fix-nxp-Fixed-wrongly-assert-message.patch)

nicusorcitu, 2024-01-15:
Hi Gabor, please include these patches in master branch. I will keep the ticket open until the v9 release is officially done. Thanks.

XuNeo, 2024-01-15:
Thank you Gabor and @nicusorcitu . 
It's a mistake from our side that forget to update those changes in NXP files when refactoring `layer` module.

kisvegabor, 2024-01-30:
As the patches are added, I close this issue