lvgl/lvgl #5343: v9.0.0 `LV_STYLE_TRANSFORM_SCALE` not defined

s-hadinger, 2024-01-15:
### LVGL version

V9.0.0

### What happened?

In `lv_api_map.h`: line 246:

`#define LV_STYLE_TRANSFORM_ZOOM       LV_STYLE_TRANSFORM_SCALE`

`LV_STYLE_TRANSFORM_SCALE` is not defined, there must be a typo somewhere.

https://github.com/lvgl/lvgl/blob/fb0fc86895331bac9f5ce514ef6324342e8ed3c6/src/lv_api_map.h#L246C1-L246C63

### How to reproduce?

Compilation issue found by Berry mapping

s-hadinger, 2024-01-15:
I suppose it should be replaced with:
```C
#define LV_STYLE_TRANSFORM_ZOOM_X       LV_STYLE_TRANSFORM_SCALE_X
#define LV_STYLE_TRANSFORM_ZOOM_Y       LV_STYLE_TRANSFORM_SCALE_Y
```

kisvegabor, 2024-01-17:
In v8 there was no ZOOM_X and ZOOM_Y only ZOOM. Therefore `LV_STYLE_TRANSFORM_SCALE_X/y` has no v8 counterpart. I think these rather should be removed.

s-hadinger, 2024-01-17:
So let's remove. It breaks the Berry mapping since `LV_STYLE_TRANSFORM_SCALE` is not defined.

kisvegabor, 2024-01-18:
I'm afraid many other things need to be updated for v9 too. :sweat_smile:

s-hadinger, 2024-01-18:
The good news is that it's the only constant that broke the mapping engine.

closed 2024-01-18 by kisvegabor

kisvegabor, 2024-01-19:
Oh, very nice! Btw, can you give give some feedback about how often lvgl used with Tasmota?

s-hadinger, 2024-01-22:
Not so many users yet, but we're seeing a slowly increasing number of questions about LVGL in Tasmota. Actually many prefer to use HASPmota (our own flavor of OpenHASP) based on LVGL, but easier to use quickly.

PR #5367 (2024-01-18): chore(api_map): remove unused api and repalce error name api
Help us review this PR! Anyone can [approve it or request changes](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/reviewing-changes-in-pull-requests/approving-a-pull-request-with-required-reviews).

### Description of the feature or fix

fix #5343 
chore(api_map): remove unused api and repalce error name api

### Checkpoints
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](htt [... cut]