lvgl/lvgl #5349: Calendar can't select year 2024

maxpromer, 2024-01-16:
### LVGL version

V8.3.11

### What happened?

Can't select year 2024 because in year_list limit to 2023. In docs you recommend use `lv_calendar_header_dropdown_set_year_list(calendar, years_list)` for set new year list but i don't see this function in `lv_calendar_header_dropdown.h`

### How to reproduce?

```
static const char * year_list = {
    "2023\n2022\n2021\n"
    "2020\n2019\n2018\n2017\n2016\n2015\n2014\n2013\n2012\n2011\n2010\n2009\n2008\n2007\n2006\n2005\n2004\n2003\n2002\n2001\n"
    "2000\n1999\n1998\n1997\n1996\n1995\n1994\n1993\n1992\n1991\n1990\n1989\n1988\n1987\n1986\n1985\n1984\n1983\n1982\n1981\n"
    "1980\n1979\n1978\n1977\n1976\n1975\n1974\n1973\n1972\n1971\n1970\n1969\n1968\n1967\n1966\n1965\n1964\n1963\n1962\n1961\n"
    "1960\n1959\n1958\n1957\n1956\n1955\n1954\n1953\n1952\n1951\n1950\n1949\n1948\n1947\n1946\n1945\n1944\n1943\n1942\n1941\n"
    "1940\n1939\n1938\n1937\n1936\n1935\n1934\n1933\n1932\n1931\n1930\n1929\n1928\n1927\n1926\n1925\n1924\n1923\n1922\n1921\n"
    "1920\n1919\n1918\n1917\n1916\n1915\n1914\n1913\n1912\n1911\n1910\n1909\n1908\n1907\n1906\n1905\n1904\n1903\n1902\n1901"
};
```

IAMMX, 2024-01-16:
https://github.com/lvgl/lvgl/blob/9d0942a11758664b58a1723d10858780cb254fc6/src/extra/widgets/calendar/lv_calendar_header_dropdown.h#L39-L46

C47D, 2024-01-16:
You could also use the `release/v8.3` branch, we recently updated the default year list.

closed 2024-02-01 by XuNeo