lvgl/lvgl #5351: Double and triple click and other gestures

kisvegabor, 2024-01-16:
### Introduce the problem

> This is feature is the pioneer of the [new shaping-pitching](https://docs.lvgl.io/master/CONTRIBUTING.html) project management process. Please, leave your feedback about how effective you find this method, so that we can improve it. 
In this issue we can discuss what we want or don't want to add to this feature. Based on that we will create a [pitch](https://github.com/lvgl/lvgl/blob/pitches/__pitch_template.md) and someone can start the development considering the set boundaries.

Now LVGL can detect long press and long press repeat events, but it lacks the ability to detect double and triple clicks.

We can also consider how to handle more complex events from multi touching.

### Proposal

Double and triple click seems straight forward, it can be implemented in `lv_indev.c` similarly to long press. 

For multi touch I wonder if we can rely on the touch controllers and handle it from the drivers level by setting something in `lv_indev_data_t`. E.g.

```c
data.state = LV_INDEV_STATE_PINCH_OUT;
data.point.x = 100;
data.point.y = 200;
```

C47D, 2024-01-19:
Can we split this into 2 or 3 features? One being double tap, other being triple tap and then gesture detection?

I think doing so will make the feature more approachable.

kisvegabor, 2024-01-19:
I think double and triple click are quite the same, but gesture really can be a different topic.

C47D, 2024-01-19:
Cool, I'll study indev to try implement the double and triple click.

I think that for the gesture support we need to have two approaches :
- Software implementation
- Assisted by touch controller

kisvegabor, 2024-01-22:
> Some questions after checking lv_indev

Great questions! 
> Which kind of input devices will support double and triple clicks?

I think pointer and encoder when clicked.

> Can double and triple click being enabled at the same time?

Yes! 

> If so, can they have different click/release time configurations? User defined values?

I think so. Similar to [these](https://github.com/lvgl/lvgl/blob/master/src/indev/lv_indev_private.h#L67-L70) (BTW, we don't have API adjust these now)

If you wish feel free to start creating a Pull Request for a pitch based on [this template](https://github.com/lvgl/lvgl/blob/pitches/__pitch_template.md).

kisvegabor, 2024-02-06:
Not stale

C47D, 2024-02-06:
I'm still trying to figure out how to get the double and triple click feature working.

C47D, 2024-02-09:
This is how I think it should be easy to implement this.

## 'Multiclick' detection support for LVGL

Once the indev has detected an Release -> Press transition it will start a dedicated timer with period X (to be defined, for experimentation I'm using a period of 500ms).

### While the timer is running (using `lv_timer_run` flag):

1. Count the number of releases, we care about releases because the UI should only send click events on release.
2. If the number of releases reaches 2, then the indev will send the `LV_EVENT_DOUBLE_CLICK` event to the object.
3. If the number of releases reaches 3, then the indev will send the `LV_EVENT_TRIPLE_CLICK` event to the object.

### Timer elapses
Indev will reset the release counter.

@kisvegabor What do you think?

mhaberler, 2024-02-12:
I was looking for this feature and found the [AceButton](https://github.com/bxparks/AceButton?tab=readme-ov-file) project

maybe some inspiration could be gleaned from there

XuNeo, 2024-02-12:
You can also refer to this piece of code to detect double, triple click.
https://github.com/lvgl/lvgl/pull/4949#issuecomment-1843337904

C47D, 2024-02-13:
> I was looking for this feature and found the [AceButton](https://github.com/bxparks/AceButton?tab=readme-ov-file) project
> 
> maybe some inspiration could be gleaned from there

Thanks for the link @mhaberler , it has very interesting information, such as [Distinguishing Clicked and DoubleClicked](https://github.com/bxparks/AceButton?tab=readme-ov-file#ClickedAndDoubleClicked). Maybe this is worth considering @kisvegabor

kisvegabor, 2024-02-15:
AceButton is really interesting and it confirms that it's difficult to distinguish these kind of events. 

I have an idea! What if we don't add new events for double and triple click but we just count the clicks (within a timeout) and provide an `lv_indev_get_click_count` functions. The advantages I see:
- It's very generic
- Doesn't add new event codes
- Adds more flexibility as e.g. `PRESSED`, `RELEASED`, `CLICKED`, `PRESSING`, `LONG_PRESSED`, etc can be processed differently depending on the click count. E.g. do something on the second press or on the third release.

I think the only disadvantage is the actual logic is hidden inside the event handler and not visible when the event is attached. E.g. `lv_obj_add_event_cb(obj, cb, LV_EVENT_PRESSED, NULL)` can process presses with any index. 

What do you think?

C47D, 2024-02-16:
Sounds reasonable, I've been getting familiar with the indev module and did a bit of adding comments, renaming variables to make it clearer, I'll send the request with those changes and then the click counting proposal.

kisvegabor, 2024-03-05:
@C47D What is the current status of this update?

C47D, 2024-03-11:
Hi @kisvegabor, I have the previous implementation in my Pc, but didn't pushed it, I'll restart it because I already created the double and triple click events.

kisvegabor, 2024-03-12:
Thank you!

We had a discussion with @XuNeo and concluded that `lv_indev_get_click_count` is really flexible and good, but adding only `LV_EVENT_DOUBLE_CLICKED` and `LV_EVENT_DOUBLE_CLICKED` for convenience would be also useful.

kisvegabor, 2024-04-03:
Continue in #6020

closed 2024-04-03 by kisvegabor