lvgl/lvgl #5352: Improve the `lv_span` widget

kisvegabor, 2024-01-16:
### Introduce the problem

> This is feature is the pioneer of the [new shaping-pitching](https://docs.lvgl.io/master/CONTRIBUTING.html) project management process. Please, leave your feedback about how effective you find this method, so that we can improve it. 
In this issue we can discuss what we want or don't want to add to this feature. Based on that we will create a [pitch](https://github.com/lvgl/lvgl/blob/pitches/__pitch_template.md) and someone can start the development considering the set boundaries.

In LVGL v9 inline text recoloring was removed. (E.g. `"hello #ff0000 red#"`). Instead the span widget can be used which is way more flexible but
- it's slower
- requires a some boilerplate
- minor features like supporting new-line  is missing

### Proposal

Examine how to simplify the API of the span widget, improve it's speed (if really needed), identify the missing features.

As reference this is how we can create this

[image]

```c
lv_obj_t * spans = lv_spangroup_create(lv_screen_active());
lv_obj_set_style_text_align(spans, LV_TEXT_ALIGN_CENTER, 0);
lv_obj_set_width(spans, 100);
lv_spangroup_set_mode(spans, LV_SPAN_MODE_BREAK);
lv_obj_center(spans);

lv_span_t * span;
span = lv_spangroup_new_span(spans);
lv_span_set_text(span, "Hello, I'm a ");

span = lv_spangroup_new_span(spans);   /*lv_spangroup_add_span(spangroup, text)?*/
lv_span_set_text(span, "big");
lv_style_set_text_font(&span->style,  &lv_font_montserrat_40);  /*span->style: internal shouldn't be exposed*/
lv_style_set_text_color(&span->style,  lv_color_hex3(0xf00)); /*can't use an existing style as poiner*/

span = lv_spangroup_new_span(spans);
lv_span_set_text(span, "text.");
```

To make it really convenient, I would expect something like:
```c
lv_obj_t * spans = lv_spangroup_create(lv_screen_active());
lv_obj_set_style_text_align(spans, LV_TEXT_ALIGN_CENTER, 0);
lv_obj_set_width(spans, 100); /*LV_SPAN_MODE_BREAK, LV_SIZE_CONTENT would mean LV_SPAN_MODE_EXPAND*/
lv_obj_center(spans);

lv_spangroup_add_span(spans, "Hello, I'm a");

lv_spangroup_add_span(spans, "big");
lv_spengroup_set_text_font(spans, NULL, &lv_font_montserrat_40); /*NULL means the last span*/
lv_spengroup_set_text_color(spans, NULL, lv_color_hex3(0xf00));

lv_spangroup_add_span(spans, " text.");
```

terry0012, 2024-01-17:
For `lv_span_set_text(span, "big");`
How to differentiate between static (`lv_span_set_text_static`) and dynamic text (`lv_span_set_text`)?
or deprecate the static interface ?

terry0012, 2024-01-17:
To improve performance, should we cache the width and height for each span ?
For now, only spangroup have cache for width and height.

C47D, 2024-01-17:
I would like to propose aliasing `NULL` for something more readable, example `LV_SPAN_USE_LAST_SPAN`, so instead of this (and having a comment explaining what `NULL` means):

```c
lv_spangroup_add_span(spans, "big");
lv_spengroup_set_text_font(spans, NULL, &lv_font_montserrat_40); /*NULL means the last span*/
lv_spengroup_set_text_color(spans, NULL, lv_color_hex3(0xf00));
```

We could do:
```c
typedef LV_SPAN_USE_LAST_SPAN  (NULL)

/* ... */

lv_spangroup_add_span(spans, "big");
lv_spengroup_set_text_font(spans, LV_SPAN_USE_LAST_SPAN, &lv_font_montserrat_40);
lv_spengroup_set_text_color(spans, LV_SPAN_USE_LAST_SPAN, lv_color_hex3(0xf00));
```

kisvegabor, 2024-01-17:
@terry0012 
> For lv_span_set_text(span, "big");
How to differentiate between static (lv_span_set_text_static) and dynamic text (lv_span_set_text)?
or deprecate the static interface ?

We can follow the same conventions as `lv_label` has:
- `set_text`: malloc the text
- `set_text_static`: save the pointer
- `set_text_fmt`: process as "printf" formatted  

> To improve performance, should we cache the width and height for each span ?

Mi initial idea is to draw multiple letters at once if possible. (Now it draws the text letter by letter). Trying to combine e.g. 16 or 32 letters if they fit into the same line and has the same style could mean a great speed up.

@C47D 
> I would like to propose aliasing NULL for something more readable, example LV_SPAN_USE_LAST_SPAN, so instead of this (and having a comment explaining what NULL means):

Good idea, but I'd call it `LV_SPAN_LAST` for less typing.

C47D, 2024-01-17:
Sounds good, maybe `LV_SPAN_USE_PREVIOUS`, I tend to be lengthy when it comes to naming lol

kisvegabor, 2024-01-17:
What is the added value of having `USE`? I think `LV_SPAN_LAST/PREVIOUS` already indicates that we will select to last span to do something with it.

C47D, 2024-01-18:
I've read the code snippet again and I think `LV_SPAN_LAST/PREVIOUS` is ok.

hyansuper, 2024-01-28:
`lv_span_set_text_fmt()` no such function??

XuNeo, 2024-01-29:
I haven't used lv_span(shame for me). I'm thinking can we make `lv_span_t` a normal `lv_obj_t` which is child of `lv_spangroup_class`. `lv_span_t` could have base class of `lv_label_class`. The styles can be set in a normal obj's way.

kisvegabor, 2024-01-30:
E.g. here the second (green: "good good study..." ) span wraps into 2 lines. As normal objs have an explicit bounding box, I think describing spans with normal objs is hard.

[image]

XuNeo, 2024-01-31:
Can it be achieved by `lv_label` widget with an additional property to specify where the first letter starts? Sure it needs magic layout to place them correctly.

anheiqq, 2024-02-02:
lv_span_set_long_mode() no such function?

Staok, 2024-02-07:
are we have chance to use span like html, it is also convenient I think:

lv_spangroup_add_span(spans, "<p>" "normal text, " "</p>" "<font color=\"#FF6F00\">" "colorred text." "</font>");

kisvegabor, 2024-02-07:
> Can it be achieved by lv_label widget with an additional property to specify where the first letter starts? Sure it needs magic layout to place them correctly.

I was thinking about it and it would be great to use the label widget as you have described. Or even more directly by adding an `inline` layout. Adding a "first-line-indent" would make it possible, except for one thing:
```html
<div style="width:120px; background-color:#faa">
<span>Hello world, I'm trying to show you</span> 
<span style="font-size:40px">this</span>. 
</div>
``` 
[image]

Here the line space of the first label changes in the last line. Theoretically the line height can be different only in the first and last line as the previous and next `span` can affect only these. So it doesn't need to be a very generic feature.

With the inline layout we could do this:
```
lv_obj_set_layout(cont, LV_LAYOUT_INLINE);

lv_obj_t * child;
child = lv_label_create(cont);
lv_label_set_text(child, "Hello");

child = lv_label_create(cont);
lv_label_set_text(child, "World");
lv_obj_set_style_text_font(child, &font1);

child = lv_image_create(cont)
lv_image_set_src(child, &my_img);

child = lv_image_create(cont)
lv_image_set_src(child, &my_img);

child = lv_dropdown_create(cont)
``` 

To be honest I don't fully understand how the inline layout would be different from the ROW flex layout.

What do you think?   

> are we have a chance to use span like html, it is also convenient I think:

Internally convert a HTML like text to spans is also a great idea! Maybe it's the beginning of something greater: an XML/HTML parser.

---

Staok, 2024-02-07:
@kisvegabor  if there is a light open source XML/HTML parser, that could be use as the frontend of span, but as a option in lv_conf.h, the original functions of span is remain. For the idea of "html", I used to develop qt and it`s convenient to add a html to a text area, even can add a link! I need it so I am here...ha

kisvegabor, 2024-02-08:
Sounds good.

At first probably we should treat these separately. 
- Let's have a flexible and fast span widget
- Add a HTML parser interface to it

kisvegabor, 2024-02-24:
Not stale

kisvegabor, 2024-03-05:
From #5784: add `lv_span_get_style`.

kisvegabor, 2024-04-03:
Let's continue in #6019.

closed 2024-04-03 by kisvegabor