lvgl/lvgl #5379: `lv_array` auto resize capability

XuNeo, 2024-01-18:
### Introduce the problem

It was raised in https://github.com/lvgl/lvgl/pull/5300.
For adding elements to array, it's clearly now that `push_back` will automatically resize as need. Same for `array_concat`.

Now we need to decide how to shrink the array. We can define a threshold, so after removing elements the array should shrink when empty space is larger than threshold .

Let's discuss it here.

cc @PGNetHun

kisvegabor, 2024-01-18:
What functions do we have remove elements from the array? I've found and `lv_array_remove`, `lv_array_erase`. Are there anything else?

PGNetHun, 2024-01-18:
Well, I think those code lines needs to be checked that call these functions: whether the callers repeatedly remove and add new items, or just remove occasionally.
I doubt that in current LVGL codebase there are many calls to remove and then adding items to same array continously.

XuNeo, 2024-02-02:
Not stale.

kisvegabor, 2024-02-19:
Not stale

XuNeo, 2024-03-07:
I'm thinking of adding SIZE_POLICY_DEFAULT, and SIZE_POLICY_DOUBLE. Not in hurry to implement though.

PGNetHun, 2024-03-11:
> > I doubt that in current LVGL codebase there are many calls (use cases) to remove and then adding items to same array continously.
> 
> I agree with @PGNetHun. Probably it doesn't worth the extra config option as (at least I believe) we don't add items to the arrays on performance critical places.

Great, then we could close this ticket.
I think there are other places that could be optimized, not this one.

closed 2024-03-12 by XuNeo