lvgl/lvgl #5382: Adding Arm2D support for Renesas RA8D1

kisvegabor, 2024-01-18:
### Introduce the problem

Renesas RA8D1 has Cortex-M85 so it should be able utilize Arm2D Helium support. 

I've tried to integrate

GorgonMeducer, 2024-01-22:
It is already supported in current version of LVGL9.

XuNeo, 2024-01-22:
Hi @GorgonMeducer ,

Just want to confirm if the `Please use gnu11` is required by `lvgl` or arm-2d? LVGL should support gnu99 but not standard c99, there's an issue for lvgl to track it.

kisvegabor, 2024-01-22:
I'm sorry, I just saw that somehow the half of my initial comment was lost. In the lost part I explained that I copy pasted code  from the Library folder but got many compiler errors.

---

I'm using E2 Studio from Renesas which is an Eclipse based IDE. I think CMSIS pack is not  supported there as in MDK.

GorgonMeducer, 2024-01-22:
> I'm using E2 Studio from Renesas which is an Eclipse based IDE. I think CMSIS pack is not supported there as in MDK.

You can add arm-2d manually as I mentioned in the previous answer.
Renesas RA tool can generate MDK projects.

You can also use the template here:

https://github.com/GorgonMeducer/EK-RA8D1

> Just want to confirm if the Please use gnu11 is required by lvgl or arm-2d? 

Arm-2D.

GorgonMeducer, 2024-01-22:
> I'm sorry, I just saw that somehow the half of my initial comment was lost. In the lost part I explained that I copy pasted code from the Library folder but got many compiler errors.

What kind of errors?

If it is about `arm_2d_cfg.h`, you can copy one from the `Library/Include/template/arm_2d_cfg.h` and place it in your application folder, and put a search path to this header file.

kisvegabor, 2024-01-23:
Thank you, Gabriel. I've tried again, and got a lot of errors similar to these:
[image]

I've added the CMSIS-DSP like this:
[image]

cc @jeremy-baker @renesasman

GorgonMeducer, 2024-01-23:
Would you have time to have a look at this? @christophe0606 @FabKlein ?

FabKlein, 2024-01-23:
if using GCC, can you please kindly add the 
```-flax-vector-conversions```
compile option ?

kisvegabor, 2024-01-24:
Thank you! `-fms-extensions` and `-flax-vector-conversions` really helped and it complied one step further. Now it with the errors below. I'm using gcc now, but I'm not sure what is the best way to change it to LLVM or AC6.

I've uploaded my E2 Studio project [here](https://github.com/lvgl/lv_renesas/tree/arm-2d).

```
Building file: ../ra/arm/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_xor_u8.c
/tmp/ccSZVsM9.s: Assembler messages:
/tmp/ccSZVsM9.s:158660: Error: garbage following instruction -- `vand q1,q1,s24'
/tmp/ccSZVsM9.s:158662: Error: garbage following instruction -- `vand q2,q2,s20'
/tmp/ccSZVsM9.s:158666: Error: garbage following instruction -- `vand q3,q3,s16'
/tmp/ccSZVsM9.s:158669: Error: lo register required -- `vstrh.32 q2,[ip],#8'
/tmp/ccSZVsM9.s:158840: Error: garbage following instruction -- `vand q1,q1,s20'
/tmp/ccSZVsM9.s:158842: Error: garbage following instruction -- `vand q2,q2,s16'
/tmp/ccSZVsM9.s:158844: Error: garbage following instruction -- `vand q3,q3,s20'
/tmp/ccSZVsM9.s:159328: Error: garbage following instruction -- `vand q1,q0,s24'
/tmp/ccSZVsM9.s:159336: Error: garbage following instruction -- `vand q0,q3,s20'
/tmp/ccSZVsM9.s:159340: Error: MVE vector register expected -- `vrmulh.u16 q2,q2,s16'
/tmp/ccSZVsM9.s:159341: Error: garbage following instruction -- `vand q1,q0,s24'
/tmp/ccSZVsM9.s:159342: Error: lo register required -- `vstrb.16 q2,[ip],#8'
/tmp/ccSZVsM9.s:159509: Error: garbage following instruction -- `vand q2,q0,s16'
/tmp/ccSZVsM9.s:159514: Error: garbage following instruction -- `vand q2,q0,s20'
/tmp/ccSZVsM9.s:159517: Error: garbage following instruction -- `vand q2,q0,s16'
/tmp/ccSZVsM9.s:159964: Error: cannot honor width suffix -- `add.n r0,#2'
/tmp/ccSZVsM9.s:159965: Error: cannot honor width suffix -- `add.n ip,#2'
/tmp/ccSZVsM9.s:160157: Error: cannot honor width suffix -- `add.n r0,#4'
/tmp/ccSZVsM9.s:160158: Error: cannot honor width suffix -- `add.n ip,#4'
/tmp/ccSZVsM9.s:160309: Error: lo register required -- `vldrb.u16 q0,[ip]'
/tmp/ccSZVsM9.s:160316: Error: lo register required -- `vldrb.u16 q2,[ip,#8]'
/tmp/ccSZVsM9.s:160318: Error: bad element type for instruction -- `vstrb.u16 q0,[ip],#8'
/tmp/ccSZVsM9.s:160322: Error: lo register required -- `vldrb.u16 q0,[ip,#8]'
/tmp/ccSZVsM9.s:160324: Error: bad element type for instruction -- `vstrb.u16 q2,[ip],#8'
/tmp/ccSZVsM9.s:160333: Error: lo register required -- `vldrb.u16 q0,[ip,#8]'
/tmp/ccSZVsM9.s:160334: Error: bad element type for instruction -- `vstrb.u16 q1,[ip],#8'
/tmp/ccSZVsM9.s:160453: Error: lo register required -- `vldrb.u16 q0,[ip]'
/tmp/ccSZVsM9.s:160460: Error: lo register required -- `vldrb.u16 q2,[ip,#8]'
/tmp/ccSZVsM9.s:160462: Error: invalid condition -- `vpt.u16 ne,q1,r10'
/tmp/ccSZVsM9.s:160463: Error: bad element type for instruction -- `vstrbt.u16 q0,[ip],#8'
/tmp/ccSZVsM9.s:160467: Error: lo register required -- `vldrb.u16 q0,[ip,#8]'
/tmp/ccSZVsM9.s:160469: Error: invalid condition -- `vpt.u16 ne,q1,r10'
/tmp/ccSZVsM9.s:160470: Error: bad element type for instruction -- `vstrbt.u16 q2,[ip],#8'
/tmp/ccSZVsM9.s:160479: Error: lo register required -- `vldrb.u16 q0,[ip,#8]'
/tmp/ccSZVsM9.s:160480: Error: invalid condition -- `vpt.u16 ne,q1,r10'
/tmp/ccSZVsM9.s:160481: Error: bad element type for instruction -- `vstrbt.u16 q2,[ip],#8'
/tmp/ccSZVsM9.s:160610: Error: invalid instruction shape -- `vmla.s16 q0,s12,r4'
/tmp/ccSZVsM9.s:160614: Error: bad element type for instruction -- `vstrb.u16 q0,[r2],#8'
/tmp/ccSZVsM9.s:160615: Error: invalid instruction shape -- `vmla.s16 q2,s12,r4'
/tmp/ccSZVsM9.s:160619: Error: bad element type for instruction -- `vstrb.u16 q2,[r2],#8'
/tmp/ccSZVsM9.s:160624: Error: invalid instruction shape -- `vmla.s16 q0,s12,r4'
/tmp/ccSZVsM9.s:160628: Error: bad element type for instruction -- `vstrb.u16 q2,[r2],#8'
/tmp/ccSZVsM9.s:160914: Error: garbage following instruction -- `vand q7,q4,s0'
/tmp/ccSZVsM9.s:160916: Error: garbage following instruction -- `vand q7,q5,s0'
/tmp/ccSZVsM9.s:160918: Error: garbage following instruction -- `vand q2,q2,s4'
/tmp/ccSZVsM9.s:160921: Error: garbage following instruction -- `vand q7,q7,s4'
/tmp/ccSZVsM9.s:160933: Error: garbage following instruction -- `vand q7,q4,s12'
/tmp/ccSZVsM9.s:161334: Error: garbage following instruction -- `vand q7,q4,s12'
/tmp/ccSZVsM9.s:161338: Error: garbage following instruction -- `vand q7,q2,s4'
/tmp/ccSZVsM9.s:161349: Error: garbage following instruction -- `vand q7,q4,s0'
/tmp/ccSZVsM9.s:161618: Error: garbage following instruction -- `vand q7,q4,s0'
/tmp/ccSZVsM9.s:161623: Error: garbage following instruction -- `vand q7,q5,s0'
/tmp/ccSZVsM9.s:161625: Error: garbage following instruction -- `vand q2,q2,s4'
/tmp/ccSZVsM9.s:161628: Error: garbage following instruction -- `vand q7,q7,s4'
/tmp/ccSZVsM9.s:161640: Error: garbage following instruction -- `vand q7,q4,s12'
/tmp/ccSZVsM9.s:161646: Error: garbage following instruction -- `vand q7,q4,s0'
/tmp/ccSZVsM9.s:161647: Error: invalid condition -- `vpt.u16 ne,q6,r9'
/tmp/ccSZVsM9.s:161922: Error: invalid instruction shape -- `vmov q2,s12'
/tmp/ccSZVsM9.s:162243: Error: garbage following instruction -- `vand q6,q4,s12'
/tmp/ccSZVsM9.s:162246: Error: garbage following instruction -- `vand q7,q5,s12'
/tmp/ccSZVsM9.s:162248: Error: garbage following instruction -- `vand q2,q2,s4'
/tmp/ccSZVsM9.s:162251: Error: garbage following instruction -- `vand q7,q7,s4'
[... cut ...]
/tmp/ccSZVsM9.s:187653: Error: bad element type for instruction -- `vstrb.u16 q3,[r3,s20]'
/tmp/ccSZVsM9.s:187659: Error: lo register required -- `vldrb.u16 q1,[ip,s20]'
/tmp/ccSZVsM9.s:187660: Error: MVE vector register expected -- `vmulh.u8 q1,q1,s16'
/tmp/ccSZVsM9.s:187662: Error: bad element type for instruction -- `vstrb.u16 q3,[r1,s20]'
make: *** [src/Library/Source/subdir.mk:57: src/Library/Source/arm_2d_helium.o] Error 1
"make -r -j16 all" terminated with exit code 2. Build might be incomplete.

```

FabKlein, 2024-01-24:
Thank you for the update and for the E2 Studio project.
Can you please kindly confirm that you are using GCC version >= **13** ?

GorgonMeducer, 2024-01-24:
I have run the benchmark on EK-RA8D1, here is the performance result:

[image]

GorgonMeducer, 2024-01-24:
I think I have found some performance bottlenecks in my lv_draw_sw_arm2d.h

I need to fix them.

kisvegabor, 2024-01-24:
Very impressive. I'll rerun my tests tomorrow and share the result. 

What is your `LV_DEF_REFR_PERIOD`? I think it's a good idea to set it to 33 to see the CPU usage too. 

BTW, I measure the CPU usage from the idle task.

GorgonMeducer, 2024-01-24:
> What is your LV_DEF_REFR_PERIOD? I think it's a good idea to set it to 33 to see the CPU usage too.

I am running in bare-metal env. The `LV_DEF_REFR_PERIOD` is 16. 
I am working on fixing the performance bottlenecks. Will have a try for `LV_DEF_REFR_PERIOD` `33` later.

kisvegabor, 2024-01-24:
Thank you Gabriel.

@jeremy-baker @renesasman
Can we use LLVM or AC6 in E2 Studio?

GorgonMeducer, 2024-01-24:
We are working on a workaround for GCC Helium in arm-2d library. @kisvegabor 
I will let you know the result.

GorgonMeducer, 2024-01-24:
Hi @kisvegabor, we have applied a temporary fix to arm-2d gcc support. Please get the latest version from the arm-2d repo and try again.

GorgonMeducer, 2024-01-24:
>I think it's a good idea to set it to 33 to see the CPU usage too.

@kisvegabor 
This is what you wanted (i.e. `LV_DEF_REFR_PERIOD` is 33ms):
[image]

kisvegabor, 2024-01-25:
We are getting very close. Now it fails with:
```
/home/kisvegabor/e2_studio/arm-gnu-toolchain-13.2.Rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: ./src/Library/Source/arm_2d_helium.o: in function `__arm_2d_impl_gray8_colour_filling_with_opacity':
/home/kisvegabor/e2_studio/workspace/lv_renesas/lv_ek_ra8d1/Debug/../src/Library/Source/arm_2d_helium.c:756:(.text.__arm_2d_impl_gray8_colour_filling_with_opacity+0x90): undefined reference to `__ARM_undef'
/home/kisvegabor/e2_studio/arm-gnu-toolchain-13.2.Rel1-x86_64-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/13.2.1/../../../../arm-none-eabi/bin/ld: ./src/Library/Source/arm_2d_helium.o: in function `__arm_2d_impl_cccn888_tile_copy_opacity':
/home/kisvegabor/e2_studio/workspace/lv_renesas/lv_ek_ra8d1/Debug/../src/Library/Source/arm_2d_helium.c:1412:(.text.__arm_2d_impl_cccn888_tile_copy_opacity+0x8c): undefined reference to `__ARM_undef'
collect2: error: ld returned 1 exit status
make: *** [makefile:312: lv_ek_ra8d1.elf] Error 1
```

But I can't find `__ARM_undef` anywhere.

kisvegabor, 2024-01-25:
That's crazy! The rendering time is half for you. 

I din't understand these though:
[image]

For multiple labels with 5 ms render time the CPU usage should be 5/33 = 15%
For screen sized text with 30 ms render time, the FPS should be 33. 

It seems we are still missing something from the benchmark. Probably the waiting for flushing.

I have this benchmark results (CPU usage is coming from the idle task).

SW only:
[image]

Dave2D:
[image]

GorgonMeducer, 2024-01-25:
> But I can't find `__ARM_undef` anywhere.

This is strange, I didn't introduce such thing called `__ARM_undef`... I need to check

GorgonMeducer, 2024-01-25:
Looks like Helium acceleration gets a good result. I am really happy about this.

kisvegabor, 2024-01-25:
> I am really happy about this.

It's really amazing! Congratulations!

GorgonMeducer, 2024-01-26:
The fix for `__ARM_undef` has been pushed to arm-2d repo, please try again. @kisvegabor

kisvegabor, 2024-01-29:
It's working now, thank you. However with gcc it's really just a little bit faster than software rendering. 

I've also found 2 rendering issues:
1. In the faded area of the first chart in the widgets demo
2. On the edge if the cards with opacity in the benchmark

I can add photos tomorrow

GorgonMeducer, 2024-01-30:
> However with gcc it's really just a little bit faster than software rendering.

@kisvegabor 
Yes, it depends on the pixel size, for RGB565, it is more obvious. This is because the vector size of Helium is only 128bits. For rgb565, it is capable to handle 8 pixels per vector and for RGBX888, it is only capable to handle 4 pixels per vector.

The Compiler also plays an important role. As I mentioned before, GCC has a very pool support for Helium in terms of performance, this is verified many times in different application realms. That's why I encourage you to try the LLVM which is the recommended free compiler from Arm.

kisvegabor, 2024-01-30:
I'll look into it and report back.

Here you can see the rendering issues I've mentioned:
[image]

[image]

GorgonMeducer, 2024-01-31:
> Here you can see the rendering issues I've mentioned.

With Helium (arm-2d) acceleration?

kisvegabor, 2024-02-01:
Yes, it's with Helium. Maybe it's related to GCC as well.

GorgonMeducer, 2024-02-01:
This is what I have seen:
[image]

kisvegabor, 2024-02-01:
With GCC or LLVM?

GorgonMeducer, 2024-02-01:
Arm Compiler 6. @kisvegabor

kisvegabor, 2024-02-07:
Now I can use LLVM, it builds without Arm2D but when I add it I get:
```
../src/Source/arm_2d_helium.c:655:14: error: invalid instruction, any one of the following would fix this:
  655 |             "   vldrb.u16               q1, [%[pSource]], #8         \n"
      |              ^
```      

These are the extra compiler settings:
[image]

Do you know what could be the problem?

GorgonMeducer, 2024-02-07:
@kisvegabor 
Try to define macro `USE_MVE_INTRINSICS` in your project and have a try.

GorgonMeducer, 2024-02-07:
This is the latest benchmark result (with the fix you have applied):

[image]

kisvegabor, 2024-02-07:
I'm using `LLVMEmbeddedToolchainForArm-17.0.1-Linux-x86_64` as according to Renesas this version is supported officially.

Adding `USE_MVE_INTRINSICS` solved the issue, thank you! 

There were an other error too but I could solve it by renaming `__va_list` to `va_list`.

```
Building file: ../ra/tes/dave2d/src/dave_memory.c
../src/Source/arm_2d.c:265:13: error: unknown type name '__va_list'; did you mean 'va_list'?
  265 |             __va_list ap;
      |             ^~~~~~~~~
      |             va_list
```

kisvegabor, 2024-02-07:
BTW, I wonder if Arm2D can help with Cortex-M33 too.

GorgonMeducer, 2024-02-07:
> Arm2D can help with Cortex-M33 too.

In General, Cortex-M33 has no Helium acceleration, so arm-2d brings no benefits. But since Cortex-M33 support ACI, If the target processor implements any ACI for 2D processing, arm-2d can help.

GorgonMeducer, 2024-02-11:
No idea at all…

GorgonMeducer, 2024-02-12:
@kisvegabor where is the reproducer? We need to check it. Thank you.

kisvegabor, 2024-02-12:
First of all happy new year!

Thank you for looking into it Gabriel. Here is my E2 Studio project: https://github.com/lvgl/lv_renesas/tree/llvm-with-arm2d

FabKlein, 2024-02-12:
Hello,
Would you have access to the offending compiler command line leading to the errors you reported ?

kisvegabor, 2024-02-15:
I can extract only the compiler flags from E2 Studio:
```
-std=c99 -flax-vector-conversions -fms-extensions -fshort-enums -fno-unroll-loops -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_renesas/lv_ek_ra8d1/generate" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_renesas/lv_ek_ra8d1/src" -I"." -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_renesas/lv_ek_ra8d1/ra/fsp/inc" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_renesas/lv_ek_ra8d1/ra/fsp/inc/api" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_renesas/lv_ek_ra8d1/ra/fsp/inc/instances" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_renesas/lv_ek_ra8d1/ra/fsp/src/rm_freertos_port" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_renesas/lv_ek_ra8d1/ra/aws/FreeRTOS/FreeRTOS/Source/include" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_renesas/lv_ek_ra8d1/ra/arm/CMSIS_5/CMSIS/Core/Include" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_renesas/lv_ek_ra8d1/ra_gen" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_renesas/lv_ek_ra8d1/ra_cfg/fsp_cfg/bsp" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_renesas/lv_ek_ra8d1/ra_cfg/fsp_cfg" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_renesas/lv_ek_ra8d1/ra_cfg/aws" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_renesas/lv_ek_ra8d1/ra/tes/dave2d/inc" -I"/home/kisvegabor/projects/lvgl/e2_studio-workspace/lv_renesas/lv_ek_ra8d1/src/lvgl" -D_RENESAS_RA_ -D_RA_CORE=CM85 -D_RA_ORDINAL=1
```
[image]

I could reproduce a similar issue when I enabled only `LV_USE_DRAW_SW_ASM     LV_DRAW_SW_ASM_HELIUM`. It adds some pure ASM snippets independent of Arm2D. So I it has something to do with the compiler.

GorgonMeducer, 2024-02-17:
It appears you are using c99 rather than gnu99

GorgonMeducer, 2024-02-22:
This compilation issue has been addressed and a solution is proposed in the discussion of #5702

kisvegabor, 2024-02-24:
> This compilation issue has been addressed

Was it addressed in the compiler or shall be addressed as described in #5702?

---
@GorgonMeducer could you send me the benckmark results with the latest LVGL, AC6 and Arm2D? I'm creating a presentation for the Embedded World and other conferences and I would like to show case how fast the rendering can be without GPU too.

GorgonMeducer, 2024-02-24:
@kisvegabor No problem. Will do it tomorrow.

GorgonMeducer, 2024-02-24:
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:////Users/gabriel/Library/Group%20Containers/UBF8T346G9.Office/TemporaryItems/msohtmlclip/clip.htm">
<link rel=File-List
href="file:////Users/gabriel/Library/Group%20Containers/UBF8T346G9.Office/TemporaryItems/msohtmlclip/clip_filelist.xml">
<style>

</style>
</head>

<body link="#467886" vlink="#96607D">

EK-RA8D1 Template Benchmark Summary (9.0.1   dev) |   |   |   |   |  
-- | -- | -- | -- | -- | --
Name | Avg. CPU | Avg. FPS | Avg. time | render time | flush time
Empty screen | 98% | 28 | 24 | 2 | 22
Moving wallpaper | 100% | 31 | 30 | 5 | 25
Single rectangle | 87% | 62 | 1 | 0 | 1
Multiple rectangles | 100% | 51 | 17 | 0 | 17
Multiple RGB images | 100% | 31 | 30 | 5 | 25
Multiple ARGB images | 100% | 27 | 34 | 9 | 25
Rotated ARGB images | 100% | 12 | 78 | 53 | 25
Multiple labels | 100% | 30 | 30 | 5 | 25
Screen sized text | 99% | 18 | 53 | 28 | 25
Multiple arcs | 100% | 29 | 31 | 6 | 25
Containers | 90% | 62 | 4 | 1 | 3
Containers with overlay | 99% | 27 | 34 | 9 | 25
Containers with opa | 90% | 61 | 4 | 1 | 3
Containers with opa_layer | 90% | 62 | 6 | 3 | 3
Containers with scrolling | 99% | 27 | 35 | 10 | 25
Widgets demo | 96% | 34 | 27 | 11 | 16
All scenes avg. | 96% | 37 | 27 | 9 | 18

</body>

</html>

kisvegabor, 2024-02-25:
Thank you Gabriel! Due to the long flush times I think it's better to set the limit to 30 fps. What do you think?

kisvegabor, 2024-02-25:
Oh and I just noticed some more things:
- I believe 16 bit color depth would be even faster
- if you are using an OS implementing LV_SYSMON_GET_IDLE and using lv_flush_wait_cb with a mutex could greatly lower CPU usage 
- if there is no OS, be sure to add at least 1ms delay after lv_timer_handler so that lvgl can measure the idle time better.

GorgonMeducer, 2024-02-25:
```csv
**Benchmark Summary (9.0.1 dev)
Name, Avg. CPU, Avg. FPS, Avg. time, render time, flush time
Empty screen, 89%, 24, 25, 0, 25
Moving wallpaper, 93%, 30, 30, 2, 28
Single rectangle, 19%, 30, 2, 0, 2
Multiple rectangles, 63%, 30, 20, 0, 20
Multiple RGB images, 97%, 29, 31, 3, 28
Multiple ARGB images, 98%, 26, 34, 6, 28
Rotated ARGB images, 98%, 12, 79, 50, 29
Multiple labels, 98%, 28, 32, 4, 28
Screen sized text, 98%, 19, 48, 20, 28
Multiple arcs, 98%, 27, 32, 4, 28
Containers, 39%, 30, 9, 2, 7
Containers with overlay, 91%, 25, 37, 8, 29
Containers with opa, 46%, 30, 10, 3, 7
Containers with opa_layer, 40%, 30, 11, 4, 7
Containers with scrolling, 94%, 25, 37, 7, 30
Widgets demo, 67%, 25, 26, 8, 18
All scenes avg.,76%, 26, 28, 7, 21**
```

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:////Users/gabriel/Library/Group%20Containers/UBF8T346G9.Office/TemporaryItems/msohtmlclip/clip.htm">
<link rel=File-List
href="file:////Users/gabriel/Library/Group%20Containers/UBF8T346G9.Office/TemporaryItems/msohtmlclip/clip_filelist.xml">
<style>

</style>
</head>

<body link="#467886" vlink="#96607D">

Benchmark Summary   (9.0.1 dev) |   |   |   |   |  
-- | -- | -- | -- | -- | --
Name | Avg. CPU | Avg. FPS | Avg. time | render time | flush time
Empty screen | 89% | 24 | 25 | 0 | 25
Moving wallpaper | 93% | 30 | 30 | 2 | 28
Single rectangle | 19% | 30 | 2 | 0 | 2
Multiple rectangles | 63% | 30 | 20 | 0 | 20
Multiple RGB images | 97% | 29 | 31 | 3 | 28
Multiple ARGB images | 98% | 26 | 34 | 6 | 28
Rotated ARGB images | 98% | 12 | 79 | 50 | 29
Multiple labels | 98% | 28 | 32 | 4 | 28
Screen sized text | 98% | 19 | 48 | 20 | 28
Multiple arcs | 98% | 27 | 32 | 4 | 28
Containers | 39% | 30 | 9 | 2 | 7
Containers with overlay | 91% | 25 | 37 | 8 | 29
Containers with opa | 46% | 30 | 10 | 3 | 7
Containers with opa_layer | 40% | 30 | 11 | 4 | 7
Containers with scrolling | 94% | 25 | 37 | 7 | 30
Widgets demo | 67% | 25 | 26 | 8 | 18
All scenes avg. | 76% | 26 | 28 | 7 | 21

</body>

</html>

GorgonMeducer, 2024-02-25:
> Due to the long flush times I think it's better to set the limit to 30 fps. What do you think?

The ultimate solution is using DMA to handle this flush. I have been waiting for a driver update from Renesas for using DMA to help flush.

kisvegabor, 2024-02-26:
Thank you again. These render times are just crazy! 

Can I check the project with LVGL somewhere? We are using DIRECT mode with 2 buffers so the flush is just an address swap. We need to sync the 2 buffers, but Im sure it can be speed up greatly with Arm2d too.

This way e.g. in case of the empty screen test during the flush while waiting for vsync  other taskd can run.

See here; https://github.com/lvgl/lv_renesas/blob/master/lv_ek_ra8d1/src/port/lv_port_disp.c

GorgonMeducer, 2024-02-26:
> Can I check the project with LVGL somewhere? 

The project is here which uses the LVGL cmsis-pack
https://github.com/GorgonMeducer/EK-RA8D1

GorgonMeducer, 2024-02-26:
@kisvegabor 
By the way, I think the CPU Usage should not take flush time into calculation. 

What do you think?

GorgonMeducer, 2024-02-26:
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:////Users/gabriel/Library/Group%20Containers/UBF8T346G9.Office/TemporaryItems/msohtmlclip/clip.htm">
<link rel=File-List
href="file:////Users/gabriel/Library/Group%20Containers/UBF8T346G9.Office/TemporaryItems/msohtmlclip/clip_filelist.xml">
<style>

</style>
</head>

<body link="#467886" vlink="#96607D">

Benchmark Summary   (9.0.1 dev) |   |   |   |   |  
-- | -- | -- | -- | -- | --
Name | Avg. CPU | Avg. FPS | Avg. time | render time | flush time
Empty screen | 12% | 25 | 0 | 0 | 0
Moving wallpaper | 8% | 30 | 3 | 3 | 0
Single rectangle | 1% | 30 | 0 | 0 | 0
Multiple rectangles | 0% | 30 | 0 | 0 | 0
Multiple RGB images | 7% | 30 | 2 | 2 | 0
Multiple ARGB images | 18% | 30 | 6 | 6 | 0
Rotated ARGB images | 88% | 19 | 50 | 50 | 0
Multiple labels | 20% | 30 | 4 | 4 | 0
Screen sized text | 60% | 30 | 20 | 20 | 0
Multiple arcs | 19% | 30 | 4 | 4 | 0
Containers | 8% | 30 | 2 | 2 | 0
Containers with overlay | 25% | 30 | 9 | 9 | 0
Containers with opa | 10% | 30 | 2 | 2 | 0
Containers with opa_layer | 14% | 30 | 4 | 4 | 0
Containers with scrolling | 23% | 30 | 7 | 7 | 0
Widgets demo | 28% | 30 | 8 | 8 | 0
All scenes avg. | 21% | 29 | 7 | 7 |  

</body>

</html>

kisvegabor, 2024-02-27:
> By the way, I think the CPU Usage should not take flush time into calculation.

It depends. If you have blocking wait for flushing we should indicate that the CPU really cannot be used for anything else durign flushing. If you are using an OS and `LV_SYSMON_GET_IDLE` the CPU usage will be lower.

Now you have used an empty flush_cb, right?

kisvegabor, 2024-03-03:
I've tried the SW render disabled flush_cb, and fortunately it didn't change the rendering time, but made the flush time zero. :slightly_smiling_face:  So it measures the times well.

Does you have any idea why the FSP is only 25 hear, instead of 30?
[image]

BTW, which buffering strategy have you used?

GorgonMeducer, 2024-03-03:
@kisvegabor 
> I've tried the SW render with disabled flush_cb, and fortunately it didn't change the rendering time

This should be the expected behaviour, isn't it?

> Do you have any idea why the FSP is only 25 hear, instead of 30?

To be honest, I have no idea also.

> which buffering strategy have you used?

I am using PARTIAL mode `LV_DISPLAY_RENDER_MODE_PARTIAL`

https://github.com/GorgonMeducer/EK-RA8D1/blob/main/RTE/LVGL/lv_port_disp_template.c

```c
void lv_port_disp_init(void)
{
    /*-------------------------
     * Initialize your display
     * -----------------------*/
    disp_init();

    /*------------------------------------
     * Create a display and set a flush_cb
     * -----------------------------------*/
    lv_display_t * disp = lv_display_create(MY_DISP_HOR_RES, MY_DISP_VER_RES);
    lv_display_set_flush_cb(disp, disp_flush);

    /* Example 1
     * One buffer for partial rendering*/
    static lv_color_t buf_1_1[MY_DISP_HOR_RES * MY_DISP_VER_RES / 10];          /*A buffer for 1/10 FB*/
    lv_display_set_buffers(disp, buf_1_1, NULL, sizeof(buf_1_1), LV_DISPLAY_RENDER_MODE_PARTIAL);

}
```

RenesasMan, 2024-03-25:
@kisvegabor Do you have the numbers to compare LLVM to AC6? I was told by our engineering system team that they were similar. I would like to recreate this benchmark for both compilers.

also, @GorgonMeducer you mentioned the next step was support from Renesas on using DMA to flush. Can you give me more details on what you need?

PS: I will be monitoring my github feeds from now on.

-Alex

[34 more replies]

closed 2024-06-20 by kisvegabor