lvgl/lvgl #5391: The proposal of the minimal portable free-standing core implementation

MouriNaruto, 2024-01-19:
### Introduce the problem

In the development of LVGL v9, there are lots of components added into LVGL. It's really good for lots of developers.

According the usage scenario (embedded, IoT and bare-metal) of LVGL, LVGL needs to keep the ability of running with no external dependencies. So, LVGL should not depend C standard library functions so much. And I know the efforts from LVGL, LVGL makes its standard library layer (for example, the “lv_strcmp” is a good example) to achieve that.

But I know it's not practical to make all components dependent on the LVGL standard library because not all components need to ask strictly. I think it's wise to let LVGL to have a minimal portable free-standing core exclude platform support but with demos and examples will make embedded, IoT and bare-metal developers happy.

kisvegabor, 2024-01-22:
Hi,

I think it affects only the content of the [libs](https://github.com/lvgl/lvgl/tree/master/src/libs) folder. Other code (written by us) could use LVGL's stdio wrapper without any issues.

MouriNaruto, 2024-01-22:
> Hi,
> 
> I think it affects only the content of the [libs](https://github.com/lvgl/lvgl/tree/master/src/libs) folder. Other code (written by us) could use LVGL's stdio wrapper without any issues.

@kisvegabor I agree with that. Maybe LVGL source tree can have the folder name like third-party in the future. I think it's necessary to split the libs for written by LVGL and written by third-party.

Kenji Mouri

MouriNaruto, 2024-01-22:
> Actually the `libs` folder has this purpose. We have only non-LVGL written libs there.

@kisvegabor 

But actually, there are something like the "fsdrv" and "lv_bin_decoder" folders (which are LVGL written libs) in the "libs" folder.

I also think maybe LVGL can have a united “platform” or "hal" folder for "devices", "osal" and "stdlib" etc folders.

Kenji Mouri

kisvegabor, 2024-01-23:
> But actually, there are something like the "fsdrv" and "lv_bin_decoder" folders (which are LVGL written libs) in the "libs" folder.

That's true in case of `lv_bin_decoder`. It really breaks the concept. However `fs_drv`s are the glue codes for the 3rd party libs, which can be stored in 3rd party libs' folder. 

> I also think maybe LVGL can have a united “platform” or "hal" folder for "devices", "osal" and "stdlib" etc folders.

By united do you mean having a single folder for all these?

MouriNaruto, 2024-01-23:
@kisvegabor 

> By united do you mean having a single folder for all these?

Yes, I think it will make people easy to know and modify them.

Kenji Mouri

kisvegabor, 2024-01-24:
I think these different things, and even on the same platform you can combine different stdlib-s, 3rd party libs and drivers.

MouriNaruto, 2024-01-24:
@kisvegabor

I'm agree with "on the same platform you can combine different stdlib-s, 3rd party libs and drivers".

But the current scenario, people who want to adapt LVGL to other platform need to put things to several different paths like src/draw, src/drivers, src/lib/fsdrv, src/osal, src/stdlib. I think it would be better to move these subfolders to a separate folder for making people easier to know whether is specific thing or not.

Note: In current stage src/osal only have the things related with multi-threading. It seems not like the OS Abstraction Layer means.

Kenji Mouri

MouriNaruto, 2024-01-24:
@kisvegabor 

I also have some additional notes.

I think some LVGL infrastructures in src/misc folder like lv_math may better to have a separate folder. I think the LVGL standard library is so good for bare-metal development. (It really helps me a lot, lol.)

Kenji Mouri

Kenji Mouri

kisvegabor, 2024-01-25:
Let's try to find some typical scenarios the users might facing with. E.g. 
1. Someone wants to port LVGL to Windows or Linux
2. An other guy writes an UEFI bios
3. A third ports it ESP32
4. Something else?

What they need to do to get started with LVGL?

Do you agree with this approach?

MouriNaruto, 2024-01-25:
I'm agree with the approach of finding some typical scenarios he users might facing first.

I'm agree with finding some typical scenarios the users might facing first. I think users might met the following scenarios for users who want to port LVGL to a new platform.

- Some guys want to port LVGL to a generic platform as application UI framework for their generic applications. (It will cover the first scenario you mentioned.)
- Some guys want to port LVGL to a generic platform as device simulator for improving their research and development efficiency.
- Some guys want to port LVGL to their devices as device UI framework even act as the hardware abstraction layer or operating systems and help them to create the final products. (It will cover the second scenario and the third scenario you mentioned.)

For three scenarios I mentioned, users need to implement something related to LVGL devices at least. But for the first scenario and the third scenario, users need to implement something related to osal, stdlib, draw to improve the performance.

For example, users need to implement SIMD optimized code for LVGL software render, or even replace the LVGL software render with a modern GPU friendly render based on something like Direct3D, OpenGL and Vulkan. Also, for bare-matal environments, users will need to provide their homemade optimized stdlib and osal implementations in most cases. For users who don't want to depend on other frameworks, they will need fsdrv and other libs features.

Actually in the reality, I'm a LVGL contributor, but not a LVGL user, because I have not found a proper way to utilize LVGL in my scenarios for releasing the final product. I feel sad and sorry for that. In LVGL 7.x and 8.x, many things are C macros makes I unable to produce a dynamic library to redistribute in my projects. But LVGL 9.x has lots of progress about improving the abstraction and modularized, and I'm happy for these changes.

I always think it's wise for me to add interoperability support with other Win32 infrastructures for LVGL Windows backend because I can utilize these with v9‘s improvements.

I hope LVGL can become more modularized for more scenarios in the future. (For example, if we want to make a LVGL UI designer for some IDEs like VS. The ideal way is we have the LVGL itself as a runtime library first. Only user code will compiled and linked to the runtime, and we can set the LVGL at runtime for change most settings like color depth dynamically. It will also useful for implementing the hot reload feature.)

Note: In recent days, I tried to use LVGL v9 to act as the UI subsystem for my Hyper-V optimized UEFI application maybe will make me to a LVGL user. (And I hope to see that day.)

Kenji Mouri

kisvegabor, 2024-01-29:
Thank you for sharing your thoughts!

The reason why I don't see a common point is because
- **fsdrv**: OS/framwork/User lib dependent
- **drivers**: platform/hardware dependent
- **osal**: OS dependent
- **stdlib** framework dependant (maybe hardware too.)
 

However it was came up multiple times to split `lv_conf.h` into 2 parts:
1. LVGL only part: widgets, demos, features, etc
2. Porting part: memory, OS, GPU, FS, etc settings.

Do you think it could be useful?

MouriNaruto, 2024-01-29:
I think splitting "lv_conf.h" into 2 parts is a good start, and it will be good if the porting part also has a separate folder called ports or something, like demos and examples folders in the LVGL source code tree.

Kenji Mouri

kisvegabor, 2024-02-11:
I'm sorry but I still can't wrap my head around it. :slightly_frowning_face: 
I'll keep your suggestion in mind and see if it can be a solution to requests in the future.

kisvegabor, 2024-02-27:
Let's continue in #5680

closed 2024-02-27 by kisvegabor