lvgl/lvgl #5396: Run the CI tests in more configuratios

kisvegabor, 2024-01-19:
### Introduce the problem

We should add more config option for the CI, especially for `LV_BIN_DECODER_RAM_LOAD` and parallel rendering. 

### Proposal

@XuNeo

XuNeo, 2024-01-30:
Need also include NEON acceleration to CI test. See https://github.com/lvgl/lvgl/pull/5526.

kisvegabor, 2024-01-30:
Also test with various cache settings.

kisvegabor, 2024-02-01:
Test OBJ_ID too. #5544

XuNeo, 2024-02-01:
Current CI has enabled them in lv_test_conf_full.h
```
#define LV_USE_OBJ_ID           1
#define LV_USE_OBJ_ID_BUILTIN   1
```

kisvegabor, 2024-02-19:
Test with `LV_DRAW_SW_COMPLEX 0`

XuNeo, 2024-03-07:
Not stale.

kisvegabor, 2024-09-05:
Add NuttX build to test the CI. See https://github.com/lvgl/lvgl/issues/6797

kisvegabor, 2024-09-05:
This one is also related: https://github.com/lvgl/lvgl/issues/3127

kisvegabor, 2024-09-05:
Cover this one too: https://github.com/lvgl/lvgl/issues/5152

AndreCostaaa, 2025-01-06:
After taking a look at the current Build CI i've noticed that not all test options are currently being run

https://github.com/lvgl/lvgl/blob/25232d92e7d2526dc2504b2fc0455cd2d17c6aa8/tests/CMakeLists.txt#L111-L153

https://github.com/lvgl/lvgl/blob/25232d92e7d2526dc2504b2fc0455cd2d17c6aa8/.github/workflows/ccpp.yml#L17-L21

Is there a specific reason behind this ? @kisvegabor

kisvegabor, 2025-01-07:
To be honest I don't why `OPTIONS_NORMAL_8BIT` is missing. That's why we need to review these sooner than later :slightly_smiling_face: