lvgl/lvgl #5408: Undefined Reference to lv_demo_benchmark()

jadonmmiller, 2024-01-20:
### LVGL version

v9.0.0-rc

### What happened?

Good afternoon, and thanks for your effort in designing such an awesome graphics library!

I'm trying to run the benchmark demo in V9, and I'm getting an "Undefined reference to lv_demo_benchmark" error.

It seems the demo files aren't being compiled although I think I have my config set properly. I have "#include "demos/lv_demos.h"" at the top of my main file.

If I copy the demo folder into lvgl/src, and change the two relative includes in lv_demo_benchmark.c, everything works great!

So, my question is if this is a bug but I'm especially interested in knowing if there's a way to configure PlatformIO to compile the demo files in their original location. Thanks!

### How to reproduce?

# Config File:
```c 
/**
 * @file lv_conf.h
 * Configuration file for v9.0.0-rc
 */

/*
 * Copy this file as `lv_conf.h`
 * 1. simply next to the `lvgl` folder
 * 2. or any other places and
 *    - define `LV_CONF_INCLUDE_SIMPLE`
 *    - add the path as include path
 */

/* clang-format off */
#if 1 /*Set it to "1" to enable content*/

#ifndef LV_CONF_H
#define LV_CONF_H

/*====================
   COLOR SETTINGS
 *====================*/

/*Color depth: 8 (A8), 16 (RGB565), 24 (RGB888), 32 (XRGB8888)*/
#define LV_COLOR_DEPTH 16

/*=========================
   STDLIB WRAPPER SETTINGS
 *=========================*/

/* Possible values
 * - LV_STDLIB_BUILTIN:     LVGL's built in implementation
 * - LV_STDLIB_CLIB:        Standard C functions, like malloc, strlen, etc
 * - LV_STDLIB_MICROPYTHON: MicroPython implementation
 * - LV_STDLIB_RTTHREAD:    RT-Thread implementation
 * - LV_STDLIB_CUSTOM:      Implement the functions externally
 */
#define LV_USE_STDLIB_MALLOC    LV_STDLIB_BUILTIN
#define LV_USE_STDLIB_STRING    LV_STDLIB_BUILTIN
#define LV_USE_STDLIB_SPRINTF   LV_STDLIB_BUILTIN

#if LV_USE_STDLIB_MALLOC == LV_STDLIB_BUILTIN
    /*Size of the memory available for `lv_malloc()` in bytes (>= 2kB)*/
    #define LV_MEM_SIZE (75U * 1024U)          /*[bytes]*/

    /*Size of the memory expand for `lv_malloc()` in bytes*/
    #define LV_MEM_POOL_EXPAND_SIZE 0

    /*Set an address for the memory pool instead of allocating it as a normal array. Can be in external SRAM too.*/
    #define LV_MEM_ADR 0     /*0: unused*/
[... 931 lines cut ...]
  // LVGL Tick Interface
  unsigned int tickPeriod = millis()-lastTickMillis;
  lv_tick_inc(tickPeriod);
  lastTickMillis = millis();

  // LVGL Task Handler
  lv_task_handler();
}

void initTFT()
{
  // Turn on Backlight
  pinMode(PB9, OUTPUT);
  digitalWrite(PB9, LOW);
}

void configureLVGL()
{
  // Start LVGL
  lv_init();

  // Configure the display driver
  mainDisplay = lv_tft_espi_create(HORIZONTAL_RESOLUTION, VERTICAL_RESOLUTION, drawBuffer, DRAW_BUFFER_SIZE);
}
```

W-Mai, 2024-01-22:
Hey bro, I've tried your configurations. But I don't meet your problem.

You needn't copy any other codes to any where, just keeping the original lvgl file structure is ok.

Maybe there're some problem with your build system. You can check it.

kisvegabor, 2024-01-22:
> If I copy the demo folder into lvgl/src, and change the two relative includes in lv_demo_benchmark.c, everything works great!

It's a limitation of Arduino. See [here](https://docs.lvgl.io/master/integration/framework/arduino.html#use-the-examples-and-demos).

> So, my question is if this is a bug but I'm especially interested in knowing if there's a way to configure PlatformIO to compile the demo files in their original location. Thanks!

Unfortunately I don't know if PlatformIO can solve it. Probably the guys at https://community.platformio.org/ can tell more about it

jadonmmiller, 2024-01-22:
> It's a limitation of Arduino. See [here](https://docs.lvgl.io/master/integration/framework/arduino.html#use-the-examples-and-demos).

Sounds great! I assumed it was a limitation with the Arduino IDE, but now I see that the documentation says it has to do with the build system. Thanks!

closed 2024-01-22 by jadonmmiller