lvgl/lvgl #5412: lv_obj_delete_anim_ready_cb not define,  LV_ASSERT_MALLOC(group)  error

xinsuinizhuan, 2024-01-22:
### LVGL version

v9.0.0

### What happened?

new version,vs2019 windows compile
lv_obj_delete_anim_ready_cb not define
LV_ASSERT_MALLOC(group)  error
[image]

### How to reproduce?

new version,vs2019 windows compile,
and when use the defalut lv_conf.h (default lv_conf_template.h change to lv_conf.h,open some define)
 lv_obj_allocate_spec_attr: Asserted at expression: obj->spec_attr != NULL (Out of memory) lv_obj.c:266

kisvegabor, 2024-01-22:
You need to increase LV_MEM_SIZE in lv_conf.h

xinsuinizhuan, 2024-01-23:
when i set the,
/*Enable the log module*/
#define LV_USE_LOG 1
then vs2019 compile， error as:
grammatical mistake
[image]

kisvegabor, 2024-01-23:
Can you copy past the error log, so that I can translate it to English? :slightly_smiling_face:

Ju1He1, 2024-01-23:
> Can you copy past the error log, so that I can translate it to English? 🙂

found a solution
https://github.com/lvgl/lvgl/pull/5445

xinsuinizhuan, 2024-01-24:
> Can you copy past the error log, so that I can translate it to English? 🙂

1>E:\lvgl\git\lvgl-master\src\core\lv_group.c(59,5): error C2059: 语法错误:“)”
1>E:\lvgl\git\lvgl-master\src\core\lv_group.c(148,5): error C2059: 语法错误:“)”
1>lv_obj.c
1>E:\lvgl\git\lvgl-master\lv_conf.h(355,1): warning C4005: “LV_ATTRIBUTE_EXTERN_DATA”: 宏重定义
1>E:\lvgl\git\lvgl-master\src\core\lv_obj.c : message : 参见“LV_ATTRIBUTE_EXTERN_DATA”的前一个定义
1>E:\lvgl\git\lvgl-master\src\core\lv_obj.c(266,9): error C2059: 语法错误:“)”
1>lv_obj_class.c
1>E:\lvgl\git\lvgl-master\lv_conf.h(355,1): warning C4005: “LV_ATTRIBUTE_EXTERN_DATA”: 宏重定义
1>E:\lvgl\git\lvgl-master\src\core\lv_obj_class.c : message : 参见“LV_ATTRIBUTE_EXTERN_DATA”的前一个定义
1>lv_obj_draw.c
1>E:\lvgl\git\lvgl-master\lv_conf.h(355,1): warning C4005: “LV_ATTRIBUTE_EXTERN_DATA”: 宏重定义
1>E:\lvgl\git\lvgl-master\src\core\lv_obj_draw.c : message : 参见“LV_ATTRIBUTE_EXTERN_DATA”的前一个定义
1>lv_obj_event.c
1>E:\lvgl\git\lvgl-master\lv_conf.h(355,1): warning C4005: “LV_ATTRIBUTE_EXTERN_DATA”: 宏重定义
1>E:\lvgl\git\lvgl-master\src\core\lv_obj_event.c : message : 参见“LV_ATTRIBUTE_EXTERN_DATA”的前一个定义
1>E:\lvgl\git\lvgl-master\src\core\lv_obj_event.c(106,5): error C2059: 语法错误:“)”
1>E:\lvgl\git\lvgl-master\src\core\lv_obj_event.c(113,5): error C2059: 语法错误:“)”
1>E:\lvgl\git\lvgl-master\src\core\lv_obj_event.c(120,5): error C2059: 语法错误:“)”
1>E:\lvgl\git\lvgl-master\src\core\lv_obj_event.c(127,5): error C2059: 语法错误:“)”
1>E:\lvgl\git\lvgl-master\src\core\lv_obj_event.c(144,5): error C2059: 语法错误:“)”

MOYUtianming, 2024-01-24:
for this problem: lv_obj_delete_anim_ready_cb not define
you can add
#define lv_obj_delete_anim_ready_cb         lv_obj_delete_anim_completed_cb
at the head of lv_demo_music_main.c
note: it just works, i don't know why.

kisvegabor, 2024-01-24:
I can't find any `lv_obj_delete_anim_ready_cb` calls in the latest `master`. Updating LVGL should also solve the issue.

closed 2024-02-15 by lvgl-bot