lvgl/lvgl #5422: Bar widget behaves incorrectly in RANGE mode if value set before start value

ijord, 2024-01-22:
### LVGL version

8.3.6

### What happened?

If you use an lv_bar in LV_BAR_MODE_RANGE, the behavior is generally correct where the bar is shown between the start value and the value.

However, if you set value below start value, and then update start value, the bar displays backwards. For instance, if you call:

set_range (0,5)
set_value (5)
set_start(4)
set_value(1)
set_start(0)

The bar will not show from 0-1, but instead 0-4, apparently using the last start_value as the end instead of value.

Calling value again or calling range fixes this. 

This can additionally be reproduced in Squareline by just setting the values in the above order.

### How to reproduce?

ui_IndexBar1 = lv_bar_create(ui_InfoScreen1);
    lv_bar_set_mode(ui_IndexBar1, LV_BAR_MODE_RANGE);
    lv_bar_set_range(ui_IndexBar1, 0, 5);
    lv_bar_set_value(ui_IndexBar1, 5, LV_ANIM_OFF);
    lv_bar_set_start_value(ui_IndexBar1, 4, LV_ANIM_OFF);
    lv_bar_set_value(ui_IndexBar1, 1, LV_ANIM_OFF);
    lv_bar_set_start_value(ui_IndexBar1, 0, LV_ANIM_OFF);

Swapping the last two lines fixes the issue.

kisvegabor, 2024-01-23:
It happens because you can't set the `value < start_value`. See [here](https://github.com/lvgl/lvgl/blob/master/src/widgets/bar/lv_bar.c#L99).

ijord, 2024-01-24:
I am aware that you cannot set value < start_value. I never expected that once I did that, even if I set value>start_value afterwards that it would remain broken.

The documentation should make it clear that at no point can you allow value < start_value, not just if you want a logical display. I would have thought that once value was updated to be greater than start value it would work again.

kisvegabor, 2024-01-24:
We could really save any values, just handle it during rendering. Do you have time to fix it and send a Pull request?

closed 2024-02-15 by lvgl-bot