lvgl/lvgl #5427: lottie with the new ThorVG support

shb3014, 2024-01-23:
### Introduce the problem

Since v9 is out I tried to use it with lottie. I managed to play lottie with thorvg and lvgl  on PC with some modifications. But the same code failed to work on an ESP32S3. Is playing lottie with Thorvg on esp32 viable considering its limited ram? If so, it would be nice to have an example like the one with rlottie. An additional example of playing svg would ne even better.

kisvegabor, 2024-01-23:
I'm also thinking about using ThorVG Lottie and SVG loader and having built in support in LVGL.

To exclude the our of RAM issue, can you try with a small lottie file? By small I mean a few kB of input data (JSON) and e.g. 20x20 target size.

shb3014, 2024-01-25:
> I'm also thinking about using ThorVG Lottie and SVG loader and having built in support in LVGL.
> 
> To exclude the our of RAM issue, can you try with a small lottie file? By small I mean a few kB of input data (JSON) and e.g. 20x20 target size.

We enlarged the Idle task and ISR task stack 10x times and made setup task (we use arduino as an component of IDF) stack to 30k and now it can run a 20x20 lottie animation. We'll make more evaluations and see if it can run under smaller stacks and bigger lottie files. So basically the issue we had was mostly related to RAM directly or indirectly. Nevertheless, we had to update the thorvg lib to the latest version instead of the one bundled with lvgl to make it run.

closed 2024-02-17 by lvgl-bot

wlq520501, 2024-04-25:
@kisvegabor I use this demo lv_demo_vector_graphic, the
/*Enable Vector Graphic APIs*/
#define LV_USE_VECTOR_GRAPHIC 1
/* Enable ThorVG (vector graphics library) from the src/libs folder */
#define LV_USE_THORVG_INTERNAL 1 /* Enable ThorVG (vector graphics library from the src/libs folder */)
And I've enabled these two macros, but it still shows the error. Why is this?
[image]

kisvegabor, 2024-04-29:
`lv_demo_vector_graphic` wants to allocate a huge buffer. Probably you are out of memory. I suggest changing the widget and height [here](https://github.com/lvgl/lvgl/blob/master/demos/vector_graphic/lv_demo_vector_graphic.c#L16-L17) to e.g. 50x50. as a starting point.

kisvegabor, 2024-09-30:
We are just investigating it with ThorVG. What I know so far that it allcoates at least 2 large memories:
1. move the whole Lottie file data to RAM
2. At least one "canvas" internally for rendering. Can be more for complex animations. 

Both of these are allocated with normal `malloc`.

See the discussion here: https://github.com/thorvg/thorvg/issues/2647
I hope we will find a way to optimize the memory usage.