lvgl/lvgl #5428: is v9 support label text recolor now？

NewObject404, 2024-01-23:
### LVGL version

v9.0.0

### What happened?

recoloring example not support.

C47D, 2024-01-23:
From #4599

> For v9 It can be replaced by lv_span which has much more features.

PGNetHun, 2024-01-23:
Related ticket: https://github.com/lvgl/lvgl/issues/5352

NewObject404, 2024-01-24:
> From #4599
> 
> > For v9 It can be replaced by lv_span which has much more features.

> Related ticket: #5352

Thanks, this is a great solution.

closed 2024-01-24 by NewObject404

andyrea76, 2024-05-09:
the span is all well and good but you have to kinda know in advance surely then where all the spans are in your label ?  rather than the recolouring option which it totally dynamic to the string you pass in and you only have one object to have to look after / create for the label rather than n spans

kisvegabor, 2024-05-09:
It's a very good point! 
In https://github.com/lvgl/lvgl/issues/6019 I've described an option for "Inline text processing" which could be basically the replacement of what we had in v8. What do you think about it? (Please reply in https://github.com/lvgl/lvgl/issues/6019 to keep the discussion more organized)