lvgl/lvgl #5430: Compared to lvgl8, lvgl9 is not very smooth

IAMMX, 2024-01-23:
### Introduce the problem

Compared to lvgl8, lvgl9 is not very smooth. For example, compared to the v8 version, sliding lists and charts have a significant lag. Hope the next version can be optimized.

kisvegabor, 2024-01-23:
In my tests the performance was very good. How did you test it?
- test code
- test hardware
- settings that might be interesting

IAMMX, 2024-01-23:
I tested it on MCU (STM32F4) because the performance of the chip itself is not very good. After updating v9, it is obvious that the sliding is not smooth. Use the example `lv_demo_widgets`.

IAMMX, 2024-01-23:
v8
```c
    static lv_disp_drv_t my_disp_drv;
    static lv_color_t *buf_1_1, *buf_1_2;
    disp_init();
    static lv_disp_draw_buf_t draw_buf_dsc_1;
    buf_1_1 = (lv_color_t *)lv_malloc(HOR_RES * VER_RES * sizeof(lv_color_t));
    buf_1_2 = (lv_color_t *)lv_malloc(HOR_RES * VER_RES * sizeof(lv_color_t));
    lv_disp_draw_buf_init(&draw_buf_dsc_1, buf_1_1, buf_1_2, HOR_RES * VER_RES); /*Initialize the display buffer*/
    lv_disp_drv_init(&my_disp_drv);
    /*Basic initialization*/
    my_disp_drv.hor_res = HOR_RES;
    my_disp_drv.ver_res = VER_RES;
    my_disp_drv.flush_cb = disp_flush;
    my_disp_drv.draw_buf = &draw_buf_dsc_1;
    lv_disp_drv_register(&my_disp_drv);
```
v9
```c
void lv_port_disp_init(void)
{
    uint32_t color_size;
    static uint8_t *buf_1_1, *buf_1_2;
    disp_init();

    lv_display_t *disp = lv_display_create(HOR_RES, VER_RES);
    lv_display_set_flush_cb(disp, disp_flush);

    color_size = HOR_RES *
                 VER_RES *
                 lv_color_format_get_size(LV_COLOR_FORMAT_NATIVE);
    buf_1_1 = (uint8_t *)lv_malloc(color_size);
    buf_1_2 = (uint8_t *)lv_malloc(color_size);
    lv_display_set_buffers(disp,
                            buf_1_1,
                            buf_1_2,
                            color_size,
                            LV_DISPLAY_RENDER_MODE_PARTIAL);
}
```

IAMMX, 2024-01-23:
Ok!
### v8 disp_flush
```c
static void disp_flush(lv_disp_drv_t *disp_drv, const lv_area_t *area, lv_color_t *color_p)
{
  uint32_t y = area->y1;
  lv_color_t * lcdbuf_gui = (lv_color_t *)lcd_buff;
  do
  {
    memcpy(&lcdbuf_gui[y * HOR_RES + area->x1],
                color_p, 
                (area->x2 - area->x1 + 1)*sizeof(lv_color_t));
    color_p = color_p + area->x2 - area->x1 + 1;
    y++;
  } while(y != area->y2 + 1);
  lv_disp_flush_ready(disp_drv);
}
```
### v9 disp_flush
```c
static void disp_flush(lv_display_t * disp_drv, const lv_area_t * area, uint8_t * color_p)
{
    uint32_t y = area->y1;
    uint8_t format_size=lv_color_format_get_size(LV_COLOR_FORMAT_NATIVE);
    uint8_t * lcdbuf_gui = (uint8_t *)lcd_buff;
    do
    {
        lv_memcpy(&lcdbuf_gui[(y * HOR_RES + area->x1) *format_size],
                color_p,
                (area->x2 - area->x1 + 1) *format_size);
        color_p= color_p+ (area->x2 - area->x1 + 1) * format_size;
        y++;
    } while(y != area->y2 + 1);
    lv_display_flush_ready(disp_drv);
}
```

GorgonMeducer, 2024-01-23:
Is it possible to use DMA for flushing?

IAMMX, 2024-01-23:
@GorgonMeducer
1. Since lv_color_t refers to 24 bits of data in v9, it must use uint8_t.
2. lv_memcpy and memcpy should be the same because I have configured "#define LV_USE_STDLIB_STRING LV_STDLIB_CLIB"
3. I use the arm-none-eabi-gcc compiler with the ofast optimization level

GorgonMeducer, 2024-01-23:
> lv_memcpy and memcpy should be the same because I have configured "#define LV_USE_STDLIB_STRING LV_STDLIB_CLIB"

When you use this option, the extra level of function call cuts down the information propagation path, (i.e. the (void *) in the argument list)

> Since lv_color_t refers to 24 bits of data in v9, it must use uint8_t.

it refers to 32bit in V8, so I say this isn't a fair comparison. @kisvegabor please correct me if I was wrong. 

> I use the arm-none-eabi-gcc compiler with the ofast optimization level
The latest version is 13.2 rel1.  You are right, for GCC, maybe memcpy vs lv_memcpy selection isn't that important.

IAMMX, 2024-01-23:
> Maybe you can use perf_counter to measure the CPU cycles used in disp_flush and use the cycle data as strong evidence to shut down the discussion about disp_flush efficiency.

Ok, thank you. I will use perf_counter for comparison when I get to work tomorrow. Now I am off work and there is no board in my hand, so I cannot test.

> This is true. But you know your LCD color type, so you can still use (uint16_t *) or (uint32_t *) in your disp_flush.

'static void disp_flush(lv_display_t * disp_drv, const lv_area_t * area, uint8_t * color_p)' Because in v9 the type of color_p is uint8_t, So I also use uint8_t to populate the data. But in v8 'static void disp_flush(lv_display_t * disp_drv, const lv_area_t * area, lv_color_t * color_p) 'color_p is lv_color_t

GorgonMeducer, 2024-01-23:
Good luck! @IAMMX

IAMMX, 2024-01-23:
Thank you. I'll test it tomorrow and see if there's any improvement.

kisvegabor, 2024-01-23:
The flush function looks good to me. As I see better performance with v9, I think we have left with the profiler.

IAMMX, 2024-01-23:
> The flush function looks good to me. As I see better performance with v9, I think we have left with the profiler.

OK, I will test it in profiler tomorrow.
It is also possible that lv_demo_widgets have become more complex, and it feels like a lot of changes compared to v8.I'll try writing the same interface tomorrow and see who performs well.

IAMMX, 2024-01-23:
By the way, how do I turn on the profiler in v8

kisvegabor, 2024-01-23:
The profiler is a new feature in v9 and not available in v8. 

The best would be use the chip's profiler in CubeIDE and compare v8 with v9. See [here](https://www.youtube.com/watch?v=BZKzwn5w1D8).

XuNeo, 2024-01-23:
Could you also provide the `lv_conf.h`?

IAMMX, 2024-01-24:
I tried it and found that the same interface performance is similar. I previously tested the reason for the card change and found that the load on v9's `lv_demo_widgets `had increased. Especially the second page

IAMMX, 2024-01-24:
I will close this issue，Thank you everyone

closed 2024-01-24 by IAMMX

kisvegabor, 2024-01-24:
What do you mean by card change?

IAMMX, 2024-01-24:
> What do you mean by card change?

I tried it and found that the same interface performance is similar. I previously tested the reason for the lack of fluency and found that the load on v9's lv_demo_widgets had increased. Especially the second page.