lvgl/lvgl #5442: Segmentation fault when trying to customise border color on buttonmatrix / keyboard

Johennes, 2024-01-23:
### LVGL version

master

### What happened?

I'm trying to set the border color on keyboard keys by setting the `border_color` field on the fill draw descriptor. This crashes my application with a segfault.

### How to reproduce?

```
lv_obj_add_event_cb(keyboard, keyboard_draw_task_added_cb, LV_EVENT_DRAW_TASK_ADDED, NULL);
lv_obj_add_flag(keyboard, LV_OBJ_FLAG_SEND_DRAW_TASK_EVENTS);

...

static void keyboard_draw_task_added_cb(lv_event_t *event) {
    lv_obj_t *obj = lv_event_get_target(event);
    lv_buttonmatrix_t *btnm = (lv_buttonmatrix_t *)obj;
    lv_draw_task_t *draw_task = lv_event_get_draw_task(event);
    lv_draw_dsc_base_t *dsc = draw_task->draw_dsc;

    if (dsc->part != LV_PART_ITEMS) {
        return;
    }

    ul_theme_key *key = NULL;

    if ((btnm->ctrl_bits[dsc->id1] & SQ2LV_CTRL_MOD_INACTIVE) == SQ2LV_CTRL_MOD_INACTIVE) {
        key = &(current_theme.keyboard.keys.key_mod_inact);
    } else if ((btnm->ctrl_bits[dsc->id1] & SQ2LV_CTRL_MOD_ACTIVE) == SQ2LV_CTRL_MOD_ACTIVE) {
        key = &(current_theme.keyboard.keys.key_mod_act);
    } else if ((btnm->ctrl_bits[dsc->id1] & SQ2LV_CTRL_NON_CHAR) == SQ2LV_CTRL_NON_CHAR) {
        key = &(current_theme.keyboard.keys.key_non_char);
    } else {
        key = &(current_theme.keyboard.keys.key_char);
    }

    bool pressed = lv_btnmatrix_get_selected_btn(obj) == dsc->id1 && lv_obj_has_state(obj, LV_STATE_PRESSED);

    if (draw_task->type == LV_DRAW_TASK_TYPE_LABEL) {
        ((lv_draw_label_dsc_t *)dsc)->color = lv_color_hex((pressed ? key->pressed : key->normal).fg_color);
    }

    if (draw_task->type == LV_DRAW_TASK_TYPE_FILL) {
        ((lv_draw_rect_dsc_t *)dsc)->bg_color = lv_color_hex((pressed ? key->pressed : key->normal).bg_color);
        ((lv_draw_rect_dsc_t *)dsc)->border_color = lv_color_hex((pressed ? key->pressed : key->normal).border_color); <<< segfaults
    }
}
```

I tried to analyse the issue with Valgrind which points at an invalid write.

```
==35521== Invalid write of size 8
==35521==    at 0x4412CF: remove_free_block (lv_tlsf.c:592)
==35521==    by 0x4417DB: block_remove (lv_tlsf.c:636)
==35521==    by 0x441922: block_merge_next (lv_tlsf.c:706)
==35521==    by 0x4423DF: lv_tlsf_free (lv_tlsf.c:1170)
==35521==    by 0x4427BE: lv_free_core (lv_mem_core_builtin.c:178)
==35521==    by 0x442D59: lv_free (lv_mem.c:125)
==35521==    by 0x4264FE: lv_draw_dispatch_layer (lv_draw.c:224)
==35521==    by 0x4266C0: lv_draw_dispatch (lv_draw.c:161)
==35521==    by 0x426774: lv_draw_finalize_task_creation (lv_draw.c:138)
==35521==    by 0x425F2C: lv_draw_rect (lv_draw_rect.c:252)
==35521==    by 0x44CBC2: draw_main (lv_buttonmatrix.c:768)
==35521==    by 0x44DCF2: lv_buttonmatrix_event (lv_buttonmatrix.c:657)
==35521==  Address 0x8c8d000000490110 is not stack'd, malloc'd or (recently) free'd
==35521== 
==35521== 
==35521== Process terminating with default action of signal 11 (SIGSEGV)
==35521==  General Protection Fault
==35521==    at 0x4412CF: remove_free_block (lv_tlsf.c:592)
==35521==    by 0x4417DB: block_remove (lv_tlsf.c:636)
==35521==    by 0x441922: block_merge_next (lv_tlsf.c:706)
==35521==    by 0x4423DF: lv_tlsf_free (lv_tlsf.c:1170)
==35521==    by 0x4427BE: lv_free_core (lv_mem_core_builtin.c:178)
==35521==    by 0x442D59: lv_free (lv_mem.c:125)
==35521==    by 0x4264FE: lv_draw_dispatch_layer (lv_draw.c:224)
==35521==    by 0x4266C0: lv_draw_dispatch (lv_draw.c:161)
==35521==    by 0x426774: lv_draw_finalize_task_creation (lv_draw.c:138)
==35521==    by 0x425F2C: lv_draw_rect (lv_draw_rect.c:252)
==35521==    by 0x44CBC2: draw_main (lv_buttonmatrix.c:768)
==35521==    by 0x44DCF2: lv_buttonmatrix_event (lv_buttonmatrix.c:657)
```

I'm unable to see where exactly things go wrong there memory-wise. :think

C47D, 2024-01-24:
Hi,

I tried to replicate the issue but there's some code missing (keyboard creation and setup and current theme). Also I wanted to ask if you did configured something specially or it's the default configuration that you used.

kisvegabor, 2024-01-24:
By having fully reproducible code I could also test it.

to narrow the potential issues, I suggest trying something simpler first. E.g.
```c
        ((lv_draw_label_dsc_t *)dsc)->color = lv_color_hex(0xff0000);
```

Johennes, 2024-01-24:
Sorry, it's scrolled offscreen in the snippet above but the segfault happens here:

```
    if (draw_task->type == LV_DRAW_TASK_TYPE_FILL) {
        ((lv_draw_rect_dsc_t *)dsc)->bg_color = lv_color_hex((pressed ? key->pressed : key->normal).bg_color);
>>      ((lv_draw_rect_dsc_t *)dsc)->border_color = lv_color_hex((pressed ? key->pressed : key->normal).border_color);
    }
```

So setting `bg_color` works but trying to set `border_color` crashes. Setting `border_width` also works but setting `border_side` crashes.

The keyboard is created like this:

```
    /* Keyboard (after textarea / label so that key popovers are not drawn over) */
    keyboard = lv_keyboard_create(lv_scr_act());
    lv_keyboard_set_mode(keyboard, LV_KEYBOARD_MODE_TEXT_LOWER);
    lv_keyboard_set_textarea(keyboard, textarea);
    uint32_t num_keyboard_events = lv_obj_get_event_count(keyboard);
    for(uint32_t i = 0; i < num_keyboard_events; ++i) {
        if(lv_event_dsc_get_cb(lv_obj_get_event_dsc(keyboard, i)) == lv_keyboard_def_event_cb) {
            lv_obj_remove_event(keyboard, i);
            break;
        }
    }
    lv_obj_add_event_cb(keyboard, keyboard_value_changed_cb, LV_EVENT_VALUE_CHANGED, NULL);
    lv_obj_add_event_cb(keyboard, keyboard_ready_cb, LV_EVENT_READY, NULL);
    lv_obj_set_pos(keyboard, 0, is_keyboard_hidden ? keyboard_height : 0);
    lv_obj_set_size(keyboard, hor_res, keyboard_height);
    ul_theme_prepare_keyboard(keyboard);
```

The last line registers the draw event handler.

```
void ul_theme_prepare_keyboard(lv_obj_t *keyboard) {
    lv_obj_add_event_cb(keyboard, keyboard_draw_task_added_cb, LV_EVENT_DRAW_TASK_ADDED, NULL);
    lv_obj_add_flag(keyboard, LV_OBJ_FLAG_SEND_DRAW_TASK_EVENTS);
}
```

I'm using [this](https://gitlab.com/postmarketOS/buffybox/-/blob/a882de795708748596707162eb850a70b0432afb/unl0kr/themes.c) custom theme where there is a custom style assigned for the keys.

```
    reset_style(&(styles.key));
    lv_style_set_bg_opa(&(styles.key), LV_OPA_COVER);
    lv_style_set_border_side(&(styles.key), LV_BORDER_SIDE_FULL);
    lv_style_set_border_width(&(styles.key), ...);
    lv_style_set_radius(&(styles.key), ...);
```

I narrowed it down some more in that commenting out `lv_style_set_bg_opa(&(styles.key), LV_OPA_COVER);` avoids the segfault (but also then the background is obviously just transparent).

XuNeo, 2024-01-24:
For LV_DRAW_TASK_TYPE_FILL, the dsc should have type `lv_draw_fill_dsc_t` instead of lv_draw_rect_dsc_t.

I also met similar problem today. I would suggest lvgl avoid converting data type from void* pointer. It bypasses all possible compiler checks.
 
Try:
```c
    if (draw_task->type == LV_DRAW_TASK_TYPE_FILL) {
        ((lv_draw_fill_dsc_t *)dsc)->color = lv_color_hex((pressed ? key->pressed : key->normal).bg_color);
/*border_color ?*/
        ((lv_draw_fill_dsc_t *)dsc)->border_color = lv_color_hex((pressed ? key->pressed : key->normal).border_color);
```

To debug these issues in future, you can search the LV_DRAW_TASK_TYPE_XXX in souce code and check where this task is created, thus get which dsc is actually used.

```c
        lv_draw_fill_dsc_t * bg_dsc = lv_malloc(sizeof(lv_draw_fill_dsc_t));
        lv_draw_fill_dsc_init(bg_dsc);
        t->draw_dsc = bg_dsc;
        bg_dsc->base = dsc->base;
        bg_dsc->base.dsc_size = sizeof(lv_draw_fill_dsc_t);
        bg_dsc->radius = dsc->radius;
        bg_dsc->color = dsc->bg_color;
        bg_dsc->grad = dsc->bg_grad;
        bg_dsc->opa = dsc->bg_opa;
        t->type = LV_DRAW_TASK_TYPE_FILL;

        lv_draw_finalize_task_creation(layer, t);
```

Johennes, 2024-01-24:
Thanks a ton for the pointer! You're right I was using the wrong DSC type. But also, it turns out there's a separate draw task type and DSC type for the border. This now works fine:

```
    if (draw_task->type == LV_DRAW_TASK_TYPE_FILL) {
        ((lv_draw_fill_dsc_t *)dsc)->color = lv_color_hex((pressed ? key->pressed : key->normal).bg_color);
    }

    if (draw_task->type == LV_DRAW_TASK_TYPE_BORDER) {
        ((lv_draw_border_dsc_t *)dsc)->color = lv_color_hex((pressed ? key->pressed : key->normal).border_color);
    }
```

Thanks all!

closed 2024-01-24 by Johennes

kisvegabor, 2024-01-25:
I winder if we should have functions to get the the draw dsc from the draw task with a non void return value. E.g.
```
//Return NULL for borders
lv_draw_fill_dsc_t * lv_draw_task_get_fill_dsc(lv_draw_task_t * t);
```

kisvegabor, 2024-01-26:
Great! Would be interested in filing a PR?

Johennes, 2024-01-26:
I gave it a shot in https://github.com/lvgl/lvgl/pull/5505.