lvgl/lvgl #5443: lv_disp_get_inactive_time starts has very high value right after app launch

Johennes, 2024-01-23:
### LVGL version

master

### What happened?

I'm using a separate thread to generate ticks and `lv_disp_get_inactive_time` in the main loop to shutdown the device on inactivity.

```
static void *tick_thread (void *args) {
    while (1) {
        usleep(5 * 1000); /* Sleep for 5 millisecond */
        lv_tick_inc(5); /* Tell LVGL that 5 milliseconds have elapsed */
    }
    return NULL;
}

int main(int argc, char *argv[]) {
    ...

    /* Start the tick thread */
    pthread_t ticker;
    pthread_create(&ticker, NULL, tick_thread, NULL);

    ...

    uint32_t timeout = ...; /* ms */

    while(1) {
        if (!timeout || lv_disp_get_inactive_time(NULL) < timeout) {
            lv_timer_periodic_handler();
        } else if (timeout) {
            shutdown();
        }
    }
```

For some reason `lv_disp_get_inactive_time` produces very large values such as 935217835 right after launching the application which then triggers the shutdown directly. I added logging in the loop to output the values and when I press a key, the value of `lv_disp_get_inactive_time` goes down and starts counting up normally again. I don't understand why it doesn't start at zero on launch though.

I only started seeing this when updating from 8.3 to current `master`.

### How to reproduce?

See above.

kisvegabor, 2024-01-24:
Are you sure that the display is already created when you call `lv_disp_get_inactive_time`? If it's created in an other thread it might happen that this thread runs first.

To better understand what happens you can print the values in [lv_display_get_inactive_time](https://github.com/lvgl/lvgl/blob/master/src/display/lv_display.c#L823).

Johennes, 2024-01-24:
Actually, I think the problem is that `disp->last_activity_time` is `0`.

Johennes, 2024-01-24:
So I guess I could work around this by calling `lv_display_trigger_activity(disp)` before entering the main loop. But should `last_activity_time` not be initialized to the current time when the display is created?

kisvegabor, 2024-01-24:
> But should last_activity_time not be initialized to the current time when the display is created?

It's a good idea! Could you try if it really solves your issue?

Johennes, 2024-01-24:
```
diff --git a/src/display/lv_display.c b/src/display/lv_display.c
index d67a9a44b..cd9c586aa 100644
--- a/src/display/lv_display.c
+++ b/src/display/lv_display.c
@@ -84,6 +84,7 @@ lv_display_t * lv_display_create(int32_t hor_res, int32_t ver_res)
     disp->layer_head->color_format = disp->color_format;
 
     disp->inv_en_cnt = 1;
+    disp->last_activity_time = lv_tick_get();
 
     _lv_ll_init(&disp->sync_areas, sizeof(lv_area_t));
```

I tried with the above patch and it seems to work. `lv_display_trigger_activity` starts counting up from 0 in the main loop now.

kisvegabor, 2024-01-24:
Great! I've created #5463

closed 2024-01-24 by kisvegabor

Johennes, 2024-01-24:
Thank you! :slightly_smiling_face:

PR #5463 (2024-01-25): fix(display): set last_activity_time to the current time in lv_display

### Description of the feature or fix

fixes #5443

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/m [... cut]