lvgl/lvgl #5447: Use the `lv_obj_set_style_transform_scale` function to scale, and there will still be residual borders when zoomed out.

lhdjply, 2024-01-24:
### LVGL version

v9.0.0

### What happened?

Use the `lv_obj_set_style_transform_scale` function to scale, and there will still be residual borders when zoomed out.

https://github.com/lvgl/lvgl/assets/18370433/5fa0469b-1846-4527-9bb2-171f46333a7f

### How to reproduce?

```c
static void anim_cb(void * obj, int32_t value)
{
  lv_obj_set_style_transform_scale(obj, value, LV_PART_MAIN);
}

void test_demo(void)
{
  lv_obj_t *obj=lv_obj_create(lv_screen_active());
  lv_obj_set_size(obj,200,200);
  lv_obj_set_style_bg_color(obj,lv_palette_main(LV_PALETTE_BLUE),LV_PART_MAIN);
  lv_anim_t aa;
  lv_anim_init(&aa);
  lv_anim_set_var(&aa, obj);
  lv_anim_set_values(&aa, 0, 256);
  lv_anim_set_time(&aa, 500);
  lv_anim_set_playback_time(&aa, 500);
  lv_anim_set_repeat_count(&aa, LV_ANIM_REPEAT_INFINITE);
  lv_anim_set_path_cb(&aa, lv_anim_path_linear);
  lv_anim_set_exec_cb(&aa, anim_cb);
  lv_anim_start(&aa);
}
```

kisvegabor, 2024-01-24:
Thank you, I'll test it tomorrow.

kisvegabor, 2024-01-25:
I'm working on it. It seems like LVGL zooms a little bit more than it should. Therefore when the size of the zoomed area is calculated a smaller area will be invalidated. 

Investigating...

lhdjply, 2024-01-25:
I think it's the same problem with #5237.

kisvegabor, 2024-01-26:
I'm also suspecting it.

lhdjply, 2024-01-29:
Is there any progress?

kisvegabor, 2024-01-31:
Finally I found the issue and opened #5548. Please try it out.

PR #5548 (2024-03-04): fix(draw): fix the invalidation of scaled areas
### Description of the feature or fix

fix #5447

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scr [... cut]
closed 2024-02-15 by lhdjply