lvgl/lvgl #5452: about  real-time refreshing

pan1024, 2024-01-24:
### Introduce the problem

How to refresh controls in real-time during events, such as in button events, refreshing a label or img multiple times before the event ends

kisvegabor, 2024-01-24:
You can ask LVGL to refresh the invalidated areas by calling `lv_refr_now(NULL)`

pan1024, 2024-01-24:
> You can ask LVGL to refresh the invalidated areas by calling `lv_refr_now(NULL)`

Thank you for your prompt reply. The solution you proposed is effective. If possible, you can mention this in the v9 version of the document, which is very useful and makes the program more flexible

kisvegabor, 2024-01-25:
Sure, I've opened this PR: #5480

Please review it.

PR #5480 (2024-01-25): docs(disp): document lv_refr_now
### Description of the feature or fix

fixes #5452

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-for [... cut]