lvgl/lvgl #5458: lv_image_set_align does not correctly place the image

s-hadinger, 2024-01-24:
### LVGL version

v9.0.0

### What happened?

Here is the Berry code
``` berry
  var tas_logo = lv.image(bg)
  # load image from Flash (not relevant here)
  tas_logo.set_scale(150)
  tas_logo.set_align(lv.ALIGN_LEFT_MID)
```

The C equivalent would be:
```C
  lv_image_set_align(tas_logo, LV_ALIGN_LEFT_MID);
```

However this call has no effect and the image is not moved within its parent.

However calling the following works:
```C
  lv_obj_set_align(tas_logo, LV_ALIGN_LEFT_MID);
```

Additional question: there is now `LV_IMAGE_ALIGN_LEFT_MID` but I don't find anywhere in the source where this is used. And I understand it's for a different behavior: placement of the image in the object, not placement in the parent.

kisvegabor, 2024-01-24:
Hi,

With this code It's working like this:
```c
    LV_IMAGE_DECLARE(img_cogwheel_argb);
    lv_obj_t * img1 = lv_image_create(lv_screen_active());
    lv_image_set_src(img1, &img_cogwheel_argb);
    lv_obj_align(img1, LV_ALIGN_CENTER, 0, 0);
    lv_obj_set_size(img1, 200, 200);
    lv_obj_set_style_bg_opa(img1, LV_OPA_50, 0);
    lv_image_set_align(img1, LV_IMAGE_ALIGN_LEFT_MID);
```
[image]

If I add scale too
```c
    lv_image_set_scale(img1, 128); //Half size
```
[image]

It's not on the edge because the pivot is at the image center.

---

> there is now LV_IMAGE_ALIGN_LEFT_MID but I don't find anywhere in the source where this is used.

It's handled [here](https://github.com/lvgl/lvgl/blob/master/src/widgets/image/lv_image.c#L730)

s-hadinger, 2024-01-24:
Thanks, but this confirms my concerns that `set_align()` is not consistent between `lv_image_set_align` and `lv_obj_set_align`. As Berry is object oriented, both are mapped to `set_align` method name, but with different behaviors depending on whether you use is on the `image` or `obj` class.

I suppose that `lv_image_set_align` should have a different name to preserve the behavior of `set_align`.

Additionally, `align()` on an image retains the behavior of `obj` while `set_align()` has a different behavior.

I'm wondering if MicroPython has the same issue.

kisvegabor, 2024-01-25:
Oh, you are right. It's a bad choice of name. What about `lv_image_set_mode`?

s-hadinger, 2024-01-25:
`mode` is very generic. What about `lv_image_set_inner_align` ?

kisvegabor, 2024-01-26:
Sounds good! In CSS it's called `object-fit`. 

cc @W-Mai

W-Mai, 2024-01-26:
> Sounds good! In CSS it's called `object-fit`.
> 
> cc @W-Mai

Yes.

Now that this matter is mentioned, let me express my thoughts.

When the `image` is set to `size content`, the scale of the image will affect the size of obj.

When the `image` is set to a specific `width` and `height`, you can use `align` and `obj-fit` to adjust the actual size or position of the image.

kisvegabor, 2024-01-30:
> When the image is set to size content, the scale of the image will affect the size of obj.

True, with `LV_SIZE_CONTENT` all these have no effect.

> When the image is set to a specific width and height, you can use align and obj-fit to adjust the actual size or position of the image.

Yes, but how shall we name our image align? I think `object-fit` sounds not that intuitive. I like the `inner_align` proposal of @s-hadinger. What do you think?

s-hadinger, 2024-01-31:
I'm temporarily renaming to `lv_image_set_inner_align` in my Berry branch to solve the conflict. I will adapt to the final naming if it changes.

kisvegabor, 2024-02-01:
I've opened #5560 to rename it

s-hadinger, 2024-02-02:
Btw I will move Tasmota to LVGL 9.0.0 this weekend.

closed 2024-02-02 by kisvegabor

kisvegabor, 2024-02-05:
Amazing! :heart_eyes:

PR #5560 (2024-02-02): refactor(image): rename align to inner_align
### Description of the feature or fix

fixes #5458

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scri [... cut]