lvgl/lvgl #5461: masking behaviour

davide-faletti, 2024-01-24:
### LVGL version

v9.0.0

### What happened?

Is it still possible to use custom masking from the application side?
Or now masking is just an internal feature?

I need to apply an angular mask to an image at runtime, with lvgl v8.x.x was possible in this way:
```C
   il_draw_mask_img_t *self = IL_DRAW_MASK_IMG(lv_event_get_user_data(e));

    il_return_if_fail(_il_draw_mask_img_apply_mask(self));

    int16_t start = _il_draw_mask_img_get_angle(IL_MIN(self->start_mask_angle, self->stop_mask_angle));
    int16_t end   = _il_draw_mask_img_get_angle(IL_MAX(self->start_mask_angle, self->stop_mask_angle));
    int16_t x_mask = IL_DRAW_WIDGET(self)->x + self->x_mask;
    int16_t y_mask = IL_DRAW_WIDGET(self)->y + self->y_mask;

    /* setup the mask */
    self->mask = il_mem_malloc(sizeof(lv_draw_mask_angle_param_t));
    lv_draw_mask_angle_init(self->mask, x_mask, y_mask, start, end);
    self->mask_id = lv_draw_mask_add(self->mask, NULL);
```

How can I change my code to do the same thing in lvgl 9.0.0?

Thanks.

kisvegabor, 2024-01-24:
As now rendering is more generic we have removed the software render related masking from the application level. 

Can you load the image to RAM? Or have have only a bitmap mask for the whole image in RAM (1 byte per pixel)?

davide-faletti, 2024-01-24:
I just add the mask to the lv_image in the event LV_EVENT_DRAW_MAIN_BEGIN and remove it at the and of the object drawing

kisvegabor, 2024-01-24:
I understand that it worked like this in v8, but I'm trying to find a solution for v9. That's why I'm asking if storing the image or at least a mask in RAM is an option.

We were already talking about introducing a generic vector based mask system in v9 too, but its development hasn't started yet. 

cc @XuNeo

davide-faletti, 2024-01-25:
I can't load the image to ram(no enought memory). i'll wait some news for this new feature in lvgl. 
thanks

kisvegabor, 2024-01-25:
I see and thank you. It's a good feedback for us that app level masking is required.

XuNeo, 2024-01-25:
Hi Gabor,

Do you see an easy way to add the mask back? I'm asking because our application code also uses mask and we have seen no other way to workaround it but the vector API.

kisvegabor, 2024-01-26:
Removing the mask hugely simplified the SW rendering process but I understand the need for it.

Have you used it on arbitrary widgets or images only?

davide-faletti, 2024-01-26:
Unfortunately I also used it on a basic widget in another cases

XuNeo, 2024-01-27:
I used it only in image widget and only with single A8 mask.

kisvegabor, 2024-01-29:
The difficult part will be to find a description for masks which works well with SW render and GPUs too, and also have some roots in CSS one we mostly follow it.

kisvegabor, 2024-02-13:
Should be partly resolved by https://github.com/lvgl/lvgl/pull/5545

closed 2024-03-09 by lvgl-bot