lvgl/lvgl #5462: Error compiling lv_observer disabling LV_USE_BUTTON

davide-faletti, 2024-01-24:
### LVGL version

9.0,0

### What happened?

Disabling the macro define LV_USE_BUTTON in lv_conf.h causes the following error:
```
../../libraries/lvgl/lvgl-9.0.0/src/others/observer/lv_observer.c: In function ‘lv_subject_remove_all_obj’:
../../libraries/lvgl/lvgl-9.0.0/src/others/observer/lv_observer.c:362:39: error: ‘btn_value_changed_event_cb’ undeclared (first use in this function); did you mean ‘arc_value_changed_event_cb’?
  362 |     while(lv_obj_remove_event_cb(obj, btn_value_changed_event_cb));
      |                                       ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                       arc_value_changed_event_cb
../../libraries/lvgl/lvgl-9.0.0/src/others/observer/lv_observer.c:362:39: note: each undeclared identifier is reported only once for each function it appears in
make: *** [../../libraries/lvgl/lvgl.mak:68: ../../build/lvgl/objs/others/observer/lv_observer.o] Error 1

```

The code is to be fixed as follow:
```
#if LV_USE_BUTTON
    while(lv_obj_remove_event_cb(obj, btn_value_changed_event_cb));
#endif
```

C47D, 2024-02-06:
@kisvegabor Should we close this? The fix was merged 2 weeks ago

closed 2024-02-06 by XuNeo

kisvegabor, 2024-02-06:
Yes, thanks!